sandeepmistry/arduino-LoRa #349: Lora sx 1278 initialization of the lora sx1278 failed

feras33, 2020-03-20:
Hello 
Dear all, 
I have been trying to connect lora sx1278 on the arduino mega 2560. I followed all the steps on github but still failing in transmitting.Can any one review the steps with me?

IoTThinks, 2020-03-21:
Show the photos of your hardware setup, codes and pin connection.

Noone would be able to do a blind guess.

feras33, 2020-03-21:
Thanks for your assistance. I have too many wiring conections in my project but i will explain what i have connected for the lora.
 LORA------------MEGA

MOSI------------51
SCK-------------52
MISO-----------50
NSS-------------10
RESET-----------9
DIO0-------------2
3.3 V from the arduino to the lora sx1278 
And the ground connection to the ground 

I am using SPI.h and LoRa.h Library. I have used the code from the library just to  make sure that it is operating well. The one that sends hellow word and increment the number of counts each it transmits this sentence. But it keep telling me to lora failed.

Karanc141, 2020-03-21:
if you add a Serial.println(version) here: https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L87 - what is the value?

Karanc141, 2020-03-21:
if you add a return Serial.println(version) here: https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L120 - what is the value?

feras33, 2020-03-21:
there is an error
class 'LoRaClass' does not have any field named '_onTxDone'

IoTThinks, 2020-03-22:
@feras33 TxDone is only available in latest zip clone of this lib.

If you use the lib via Arduino IDE, then it is old version 0.7.0. There is no txDone.

IoTThinks, 2020-03-22:
@morganrallen Please help to push Release v0.7.1 to Arduino Library.
Otherwise, the examples will not match with the v0.7.0 in Arduino library.

IoTThinks, 2020-03-22:
[image]

SS shoud be pin 53 instead.
If LoRa can not set SS low then init LoRa failed.

I believe in Arduino, you need to follow the default pinout.

In esp32, can change the SS pin to any pin.

morganrallen, 2020-03-22:
> @morganrallen Please help to push Release v0.7.1 to Arduino Library.
> Otherwise, the examples will not match with the v0.7.0 in Arduino library.

Doh! I pushed the commit but forgot to tag the release. I've done and pushed that now, it can take some time to show up in the Library Manager but is on it's way. Good catch @IoTThinks , thanks!

morganrallen, 2020-03-22:
Well, it might have already been published? Everyone sure it's up to date? Were the examples pull directly from this repo?

IoTThinks, 2020-03-22:
@morganrallen Awesome. I will check the Arduino IDE on Monday for new release.

Normally, I copy the examples from github but install the library from Arduino IDE.

Thanks for your effort.

feras33, 2020-03-22:
i have updated the latest version 0.7.1 .I am getting a new error.
Error compiling for board Arduino Mega or Mega 2560.

morganrallen, 2020-03-22:
Two points here
First, this error is incomplete and doesn't help us debug this issue.
Second, we run build tests against this device and they are passing, so this is likely something in your code.

feras33, 2020-03-23:
I have been trying to do all what is mention on github. The lora chip still not working can you provide a picture of setup or coding to check on my arduino, Because the library code it is compiled but still not working.

feras33, 2020-03-23:
By the way nss pin is the same as ss pin? Because there is no nss pin on arduino mega.

Karanc141, 2020-03-24:
Yes, it is same. SS is Slave Select and NSS is new Slave Select.

Karanc141, 2020-03-24:
https://github.com/sandeepmistry/arduino-LoRa/issues/349#issuecomment-602058094

IoTThinks, 2020-03-26:
Try to change to SS to pin 53.

IoTThinks, 2020-03-26:
Try to change to SS to pin 53. 
I believe the pins for SPI in Arduino may be fixed.

feras33, 2020-03-26:
I tried it, but still no use.

IoTThinks, 2020-03-26:
You can learn from this friend. https://github.com/sandeepmistry/arduino-LoRa/issues/351
At least he can send something with Mega 2650. :D

IoTThinks, 2020-04-15:
I made Mega + sx1278 works at https://github.com/sandeepmistry/arduino-LoRa/issues/205#issuecomment-613848638

closed 2021-05-25 by IoTThinks