sandeepmistry/arduino-LoRa #352: Starting LoRa failed

MariaDan, 2020-03-25:
I am receiving `Starting LoRa failed` while using the demo code. 
**Hardware** used: Adafruit ItsyBitsy 32u4 - 3V 8MHz and LoRa SX1278 433MHz. 
**Library** used: https://github.com/sandeepmistry/arduino-LoRa
**Connections**: https://www.electroschematics.com/rf-radio-frequency/?fbclid=IwAR2SMn3RftYkTGwjH4TqrT4lA3NPWiSz8Rj8xSYt2S5UswWhf2ruW11zGtk

I verified everything and seems to be fine, I don't understand why I get this error. 
Would you mind helping me, please? 

Thank you so much!

IoTThinks, 2020-03-28:
No one can do a blind guess with insufficient information.
Likely a pin mapping error to wrong SPI pin.
And wrong SS pin.

MariaDan, 2020-03-28:
Please tell me what kind of information is missing, so I can give it to you. I left the connections link above, a screenshot from the links above is here: 
[image]

IoTThinks, 2020-03-28:
Do this to make sure the pins are right.
Of course you need to replace the correct pins to those LORA_...
And make sure you choose the correct board.

And normally Arduino Uno will not have enough current to send LoRa packets far.

99.99% of "Starting LoRa failed" is wrong pin mapping.
```
  SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI);
  LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO012);
  
  while (!LoRa.begin(433E6)) {
   Serial.println("Starting LoRa failed");
    delay(1000);
  }
```

MariaDan, 2020-03-28:
Thank you for your answer. 
When I compile, I het this error: **no matching function for call to 'SPIClass::begin(int, int, int)'**
The entire error is here: 
"C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_150590\LoRaSender.ino: In function 'void setup()':

LoRaSender:8:23: error: no matching function for call to 'SPIClass::begin(int, int, int)'

   SPI.begin(13, 12, 11);

                       ^

In file included from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_150590\LoRaSender.ino:1:0:

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\libraries\SPI\src/SPI.h:159:15: note: candidate: static void SPIClass::begin()

   static void begin();

               ^~~~~

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\libraries\SPI\src/SPI.h:159:15: note:   candidate expects 0 arguments, 3 provided

exit status 1
no matching function for call to 'SPIClass::begin(int, int, int)'"

IoTThinks, 2020-03-28:
My bad,  SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI); is for arduino-esp32 only.
You can remove the above line for Arduino as the SPI pins are fixed.

MariaDan, 2020-03-28:
I get the same error. Is is possible to be a problem the fact that I use **Adafruit ItsyBitsy 32u4 - 3V 8MHz** to replace Arduino Uno? 

[image]

IoTThinks, 2020-03-28:
Last try.
In setup()
pinMode(LoRa_SS, OUTPUT);
digitalWrite(LoRa_SS, HIGH);

If no luck, then Arduino Uno is a better choice than Arduino Nano.

IoTThinks, 2020-03-28:
Or you set the SPI frequency down to 8Mhz.
LoRa.setSPIFrequency(8E6);
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-spi-frequency

MariaDan, 2020-03-28:
Unfortunately, it still doesn't work. 
I tried on **Arduino Uno** and it works. What can be the difference? What should I modify on my Itsy Bitsy to make it work as well?

IoTThinks, 2020-03-28:
I suspect the frequency of your board is too low (8Mhz),

Or your board's pin definition is different from Nano or Uno.

Or the power input of your board is insufficient.
You can try to provide external power to LoRa.

MariaDan, 2020-03-28:
Thank you so much! You have all my respect. 
I declare this issue closed.

closed 2020-03-28 by MariaDan

MariaDan, 2020-03-31:
I found and solved the issue, maybe this answer will help someone: 

1. **Itsy Bitsy has different pinout, so you should make the following connections:** 
    LoRa module -> Itsy Bitsy 
    GND               ->   GND
    Vcc                 ->   3.3V
    MISO             ->   MISO 
    MOSI             ->   MOSI     
    SLCK              ->   SCK
    Nss                ->   D10
    Di00              ->   D7
    RST               ->   D9   

2. **Add the code below in void setup:** 
   LoRa.setSPIFrequency(8E6);  //set the SPI frequency down to 8Mhz
   LoRa.setPins(10, 9, 7); //override the default NSS, NRESET, and DIO0 pins used by the library 

3. **Don't forget to change the frequency, write the one your LoRa module is using**. Mine has 433 MHz:  LoRa.begin(433E6)

IoTThinks, 2020-03-31:
Love to hear the working update. 

Remember to update the lib to 0.7.2.
0.7.1 has issue.