sandeepmistry/arduino-LoRa #355: Arduino Nano 33 IoT

skuppens, 2020-04-02:
Hi,

I was wondering if your code also works with the Arduino Nano 33 IoT.
It doesn't give any errors when compiling. Output seems OK 
`11:07:12.328 -> LoRa init succeeded.
11:07:12.328 -> 
11:07:12.328 -> LoRa Simple Node
11:07:12.328 -> Only receive messages from gateways
11:07:12.328 -> Tx: invertIQ disable
11:07:12.328 -> Rx: invertIQ enable
11:07:12.328 -> 
11:07:12.328 -> Send Message!
11:07:12.396 -> TxDone
11:07:13.324 -> Send Message!
11:07:13.392 -> TxDone
11:07:14.349 -> Send Message!
`
But when I check the Gateway I have (One from TTN, indoor model) I don't see any message arriving). Also not sure If I should see them, but I would like them to be visible somewhere.
Trying to get this to work with TTN. 
The LIMC libraries give errors on the IRQ pins or something.

Any thoughts on how to tackle this one?
Kindest regards,

IoTThinks, 2020-04-02:
Your gateway may be a LoRaWAN gateway.
This library is LoRa.

They are not compatible.

skuppens, 2020-04-02:
That’s true…
My mistake, looked over that.

Thank you

Stijn

closed 2020-11-12 by IoTThinks