sandeepmistry/arduino-LoRa #361: possible to group with inveriq?

halukmy, 2020-05-03:
i mean there are 4 different farming group

water, cows, sun panels and something more

make communicate only water group with water, others not see or read, like invertiq?

anyone tried?

IoTThinks, 2020-05-04:
Set each group with different frequencies.

slavino, 2020-05-04:
Or simply use more advanced implementation to steer your implementation from central point(s) but keep the LoRa client devices in same "domain".

I'd suggest you to go for MQTT on lightweight Debian with RaspberryPi to keep it low cost and verify if it suits your needs. (writing simple Gateway code for LoRa-WiFi-MQTT isn't huge overhead).

halukmy, 2020-05-04:
> Or simply use more advanced implementation to steer your implementation from central point(s) but keep the LoRa client devices in same "domain".
>
> I'd suggest you to go for MQTT on lightweight Debian [...]

hows possible with arduino?

@IoTThinks  thanks for advice! 

without change freqs can we hack it with invertiq methods?

premmable maybe?

https://github.com/xreef/LoRa_E32_Series_Library this buddy has some technics about it
https://www.mischianti.org/wp-content/uploads/2019/10/fixedMessageToASpecifiedDevice-768x417.jpg

IoTThinks, 2020-05-04:
invertiq= true or invertiq = false creates 2 groups only.
Nodes with different Preample settings still interfere each other.

Originally, Sync Word fits your need to separate nodes to different networks.
However, it can not provide 100% isolation between groups.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#sync-word

IoTThinks, 2020-05-04:
> https://github.com/xreef/LoRa_E32_Series_Library this buddy has some technics about it
> https://www.mischianti.org/wp-content/uploads/2019/10/fixedMessageToASpecifiedDevice-768x417.jpg

This one is applicable to UART LoRa only.
UART Lora is SPI LoRa + STM8 (+firmware)

halukmy, 2020-05-06:
@IoTThinks i respect your knowledge bro, respect so much!

so your primary advice is change freq, and second is use  sync-word?

is there any example?

IoTThinks, 2020-05-06:
Im only a one-year newbie here.
To change frequency and sync word, you can read from the API here.

You can change frequency/sync word before you send a packet to a group and revert back on default frequency/sync word

https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-spi-frequency

halukmy, 2020-05-06:
whats about sync-word? never use it before https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#sync-word is it use case for groupping?

IoTThinks, 2020-05-06:
By right, syncWord is for grouping of networks.
It is what you want.

The LoRa chip will filter packets by syncWord.
However, it is not 100% isolated. So choose the sync word wisely aka more difference in Sync word in bimary is better.

Can have up to 255 SyncWord. In examples folder should have examples.

0x12 for LoRa, 0x34 If I remember correctly.

However, in the same frequency, signals with different syncwords still interfere each other.

halukmy, 2020-05-08:
@IoTThinks  thanks bro,

so i set it on setup 

 int SyncWord = 0x22;  LoRa.setSyncWord(SyncWord);          

and on dublex sample  
 if (SyncWord == 0x22) 
{
go n
} else dont listen

if i'm right please share it,

and if i'm right

how can i add syncblock on message and send via sender message it?

IoTThinks, 2020-05-08:
SyncWord is a kind of parameters of the LoRa chip and stored in register. It is not in the transmitted message.

Just set SyncWord to be the same for two nodes in the same group.

Nodes with the same SyncWord will receive the packet.

halukmy, 2020-05-08:
@IoTThinks i was doing it like message header (222.... ... ) if message starts with (222) go on if not nvm,

for mqtt gateway, how can send seperate syncwords ?

and is it same method my first line word?

IoTThinks, 2020-05-08:
I do like your way too.
Put a simple dst_id in front to know which node to pick up or discard.

However, they are not the same.
SyncWord is at Chip level.
Simple dst_id is code level.

halukmy, 2020-05-08:
@IoTThinks can you share me a publisher/sender sample code for syncword?

i got two groups, syncword like (1) and (2)

how can i send only (2) or maybe 3is group is that possible?

halukmy, 2020-05-08:
group #1
set  LoRa.setSyncWord(0xF3);
    sendMessage(message);

group #2
set  LoRa.setSyncWord(0xFF);
    sendMessage(message); 

on real time with codding?

IoTThinks, 2020-05-08:
Semtech advises to use dst_id in messages or set frequency instead of Sync word.

halukmy, 2020-05-08:
@IoTThinks  thanks dude, is there any code sample of dst_id? cant find it here

IoTThinks, 2020-05-08:
In examples folder, look for senders.
Just append “123” in front of sending messages.

neo7530, 2020-05-27:
I have from - to addressing on my nodes. It works like an ip address. Add 2 32-bit words in front of your message and you're done.

closed 2020-11-12 by IoTThinks