sandeepmistry/arduino-LoRa #363: LoRa.endPacket() hangs

mikaelprag, 2020-05-14:
So I see a few mentions of this already, but the suggestions do not help in my case.

The issue is that the first package is always sent correctly and received. On the second run execution stops on LoRa.endPacket() on the sending device.

To make it really simple my test devices are 2 Arduino Unos and 2 RF95W modules hooked up to the standard pins using the standard sketches LoRaSender and LoRaReceiver at 868MHz.

I have tried to lower the SPI frequency to 1E6 and tried with 8E6 as well but no change.

Now here comes the twist. When I add a capacitor (220uF) over the + and - of the sender, it all works perfect. So this would suggest some power issue at the sender. I tried changing the power to a lab power supply (@3.3V) and even a battery without any change.

Another twist is if I switch the roles of these devices, it works without any issues!

Do I just have a bad RF95W or do I really need a capacitor on the sender or is there anything else I could try?

IoTThinks, 2020-05-14:
Sx1278 needs 80+mA to TX.
3.3v of Arduino may provide up to 50mA.

My Mega said up to 50mA but I use ok.

You can try to measure the 3.3v when doing TX to see if the current drops?

Or can try to reduce TxPower via API.

mikaelprag, 2020-05-14:
Thanks for your reply!

I don't have an oscilloscope to see the spike well, but I did measure with my multimeter.

I don't know how this happens, maybe someone more into electrics can enlighten but when I have no capacitor on and even though I'm feeding the module 3.3v it measures 4.49v when it's working and sending correctly. When I don't have a capacitor on and it gets stuck, the voltage drops to 3.88v. How can the voltage be higher than what I'm feeding it? In any case, the drop in voltage is there as you said. 

I guess it would make sense to add a capacitor in this scenario? If I use async transfer the module doesn't hang, but it doesn't send anything either. Maybe it's better though since it doesn't hang and if the proper power was supplied the following cycle it would send.

mikaelprag, 2020-05-14:
Setting the TX power to 2 seems to have no effect.

IoTThinks, 2020-05-15:
May be you are providing 5v instead?
Voltage drops during TX may imply power issue.
My esp32 may hit up to 100mA for TX.

Adding a capacitor may imply the cause of the power issue, too.
It is my experience. Im not electronics.

May be the Uno sender has power issue.
Uno receiver uses less power.
That why when you swap the two boards, things can run again?

mikaelprag, 2020-05-15:
Nope, it's really 3.3v going in and then 4.49v when operating. Very odd. The same is true on the RX unit, also when they are idle, not just when TX/RX.

I did switch to an ATTiny84 and hooked the RFM95 up to that and it has the same issue, with battery, lab PSU and power from an Arduino. With cap it's fine.

Well, I did switch the roles around but they are both RFM95s on Unos, so I don't see why one chip would require more current than the other?

I have more units of RFM95 coming in from another supplier soon to test with.

IoTThinks, 2020-05-15:
I used Ai-thinker Ra02 433Mhz.
So far Ok for Esp32 and Arduino Mega.

cynicer, 2020-09-15:
Thanks, 220uF does it for me.

closed 2020-11-12 by IoTThinks

c4explosive, 2021-06-22:
Yes, the 220uF capacitor, works. Module: SX1278, 433MHz; arduino mega.

IoTThinks, 2021-06-23:
> Yes, the 220uF capacitor, works. Module: SX1278, 433MHz; arduino mega.
So it is the power issue?

NguyenDTheVinh, 2023-02-03:
Hi @c4explosive, can you tell me where should I add the 220uF capacitor?

cynicer, 2023-02-03:
Just add a decoupling capacitor between Vcc and GND.

GeneralLyric5, 2025-04-04:
Can you add a image of were, we have to put the capacitor? just like in the photo? 

[image]