sandeepmistry/arduino-LoRa #364: LoPy4 starting LoRa failed

paidforby, 2020-05-14:
Hey there! I am receiving `Starting LoRa failed` while using the demo code. I've seen [FAQ #1](https://github.com/sandeepmistry/arduino-LoRa#faq) about using [setPins](https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins), but am still not able to solve my problem,

I've been trying to get the Pycom LoPy4 working with this library. I'm using the most simple example, LoRaSender, and I've set the pins according to the API docs and Pycom's documentation, but LoRa is still failing to start.

My test code looks like this,
```
 #include <SPI.h>
 #include <LoRa.h>
 
 int counter = 0;
 
 void setup() {
   Serial.begin(115200);
   while (!Serial);
 
   Serial.println("LoRa Sender");
 
   LoRa.setPins(LORA_CS, LORA_RST, LORA_IRQ);
   
   if (!LoRa.begin(915E6)) {
     Serial.println("Starting LoRa failed!");
     while (1);
   } 
 } 
 
 void loop() {
   Serial.print("Sending packet: ");
   Serial.println(counter);
   
   // send packet
   LoRa.beginPacket();
   LoRa.print("hello ");
   LoRa.print(counter);
   LoRa.endPacket();
   
   counter++;
   
   delay(5000);
 } 
```
where `LORA_CS`, `LORA_RST`, `LORA_IRQ` are defined as `18`, `-1`, and `23` in the arduino-esp32 variants, here https://github.com/espressif/arduino-esp32/blob/master/variants/lopy4/pins_arduino.h#L14. Note: I've also tried passing the raw integers to `setPins()`.

The LoPy4's pinout can be found here, https://docs.pycom.io/datasheets/development/lopy/#app

I am thinking that the problem has something to do with the SPI settings, `LORA_MISO` is `19`, `LORA_MOSI` is `27`, and `LORA_CS` is `18`, but there is another SPI interface on pins `37`, `22`, and `13`. Maybe the library is having trouble choosing the correct SPI? Or maybe I am missing something more obvious. I've tried setting the SPI frequency to `10E6` and `8E6`. I've also tried setting the SPI interface to something else, like `SPI0` or `SPI1`, but those don't exist. Wonder if anyone has had any success with the LoPy4 and this library. The only mentions of Pycom or LoPy4 in other issues were not helpful.

Any advice would be greatly appreciated!

IoTThinks, 2020-05-14:
Double check the board pinout.
And try to put this before LoRa.begin

SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_SS);
  LoRa.setPins(LORA_SS, LORA_RESET, LORA_DIO0);

paidforby, 2020-05-14:
Very nice!

@IoTThinks your suggestion worked. I knew it was something with the SPI settings. 

To clarify, I just added,

`SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS);`

before

`LoRa.setPins(LORA_CS, LORA_RST, LORA_IRQ);`

in my original example. Either, `LORA_IRQ` or `LORA_IO0` will work since  they are both aliased to `23`.

Thanks!

closed 2020-05-14 by paidforby