sandeepmistry/arduino-LoRa #365: Low power exeample

maxholgasson, 2020-05-16:
Hi, do you have a sample code to the video you show an youtube with rfm95 in sleep mode and CAD mode?

IoTThinks, 2020-05-19:
I believe only need to run LoRa.end() before setting the MCU to deep sleep.

maxholgasson, 2020-05-19:
I guess probably not or how do you make sure to wake up the whole setting when there's a packet sent?

IoTThinks, 2020-05-19:
You want wake up event? 
May be you can play around with CAD in DIO1.
I never try yet, btw.

Now, I prefer doing like Class A of LoRaWAN.
Node to wake up periodically, send a packet to gateway and wait a while for downloading packet.
Then back to sleep.

maxholgasson, 2020-05-19:
I want what Sandeep showed in the video. A low power receiver that wakes up with CAD eg

sandeepmistry, 2020-05-19:
I never made a video :)

Do you have a link?

maxholgasson, 2020-05-19:
Hi,
ah I thought that's your video because this github is linked there https://www.youtube.com/watch?v=_Lw62DjQsZ8

chocotov1, 2020-05-21:
I made that video, including (pt. 1: https://www.youtube.com/watch?v=fP5RO6sIQGk). 

In the video you can see an implementation of CAD mode that's described in the Semtech SX1276 datasheet on page 42. I never published the code as it's pretty experimental and in this setup with the watch crystal nobody would probably want to try to implement it.

I'll tidy up the code that's used in the video and make an example that uses the watchdog instead of the watch crystal for the sleep intervals. Should be ready in a few days in my fork.

Would love to see CAD mode in the original arduino-LoRa library!

IoTThinks, 2020-05-21:
@chocotov1 : So eager to wait for your code :-)

chocotov1, 2020-05-21:
I created the branch cadmode in my fork. It's a working showcase of CAD mode, nothing more! I don't really encourage anyone to use it because there's probably a better way to implement it in the library. I commented out a couple of attachInterrupt() lines, so I definitely broke something in the progress.

I included the example LoRaReceiverCadDemo328p. Here the watchdog timer of the 328p (UNO) is configured with a 16 ms sleep time. Power consumption: 1.17 mA.

Setting the watchdog to 30 ms had a consumption of 680 uA. You can change the sleep time by changing these lines:

```
WDTCSR = WDTO_30MS;
//WDTCSR = WDTO_15MS;
```
Be sure to also check that the preamble is set appropriately in the sender. It must be at least as long as the sleep time:

```
LoRa.setPreambleLength(35);
```

maxholgasson, 2020-05-22:
Thanks for your work. I'm getting my RFM95 modules ready and am curious about testing your code. Which sketch would you recommend for the sender unit?

chocotov1, 2020-05-22:
LoRaSender should work. It's probably best to define the longer preamble there too, for instance like this:

```
LoRa.setPreambleLength(30);
```

chocotov1, 2020-05-23:
I uploaded a new demonstration video that uses the code of my cadmode branch:
https://www.youtube.com/watch?v=b3Wj6lMpMNk

If nobody else is interested i'd be happy to help to add CAD mode to the upstream repository. Im hesitant because i'm not familiar with how the other interrupt based code works.

morganrallen, 2020-05-23:
If you check the PRs there are multiple attempts to include CAD detection, that part of the code is fairly straight forward. I'd suggest looking over #334 

Would be awesome to finally land a PR for CAD detection.

chocotov1, 2020-05-23:
Thanks for pointing that out. I'll take a look over there!

maxholgasson, 2020-05-23:
Great! 
btw you wrote "I'll tidy up the code that's used in the video and make an example that uses the watchdog instead of the watch crystal for the sleep intervals. "
I read that the internal watchdog of the 328P is quite power hungry. Someone suggested to use a watchdog IC instead like this https://github.com/TomWS1/tplWDT
Would that be also an option to use such a hardware watchdog?

morganrallen, 2020-05-24:
Having this particular software library require an external piece of hardware would be completely prohibitive for multi-device compatibility. Instead establishing a MACRO or extern function to use the best available method for the MCU is use is probably a better approach.

IoTThinks, 2020-05-24:
@morganrallen The CAD is still pending for a release, right?

In LoRaWAN, a CAD is useful to detect a packet from any SF of a frequency.

maxholgasson, 2020-05-24:
@morganrallenit was just a question as this timer IC is obiously more power efficeint than the internal watchdog. There are even breakoutboard available for those who don't use selfmade pcbs. To have this at least as an option for low power devices would be great.

chocotov1, 2020-05-24:
@maxholgasson 

The 8 uA power consumption of the watchdog of the 328p (or any other Atmel chip) is so low that trying to optimize isn't worth it.

About parsing the data my latest video: The data of the nodes is packed into a compact format, roughly using the principles that are explained here: https://www.thethingsnetwork.org/docs/devices/bytes.html

> I would like to blink a led on pin7 when the first node is received and blink pin8 when second node..as an example

Easy solution:
- first node sends one byte with value 1
- second node sends one byte with value 2

neo7530, 2020-05-24:
Cad mode is working excellent on a receiving only node. But sadly it can't mixed with transmitting code.  Have modified the gateway code to act as repeater. So if I were able to have cad mode in addition, (for sleep mode) this would give a nice solar powered repeater...

chocotov1, 2020-05-25:
> I uploaded a new demonstration video that uses the code of my cadmode branch:
> https://www.youtube.com/watch?v=b3Wj6lMpMNk

I discovered that in my example sketch and the sketch of my video that the ADC was turned on. With it turned off the power consumption dropped to 590 uA (display turned off).

```
ADCSRA = 0;
```

maxholgasson, 2020-05-25:
Wow ok, where did you put the `ADCSRA = 0;` int the go_to_sleep() function?

neo7530, 2020-05-25:
It is in setup()

neo7530, 2020-05-27:
@chocotov1 Can you transmit with your CAD-Code? Didn't got it to work yet.

chocotov1, 2020-05-27:
@neo7530 

I didn't try before, but now i've tested it and it worked for me. I modified my example. The new code is commented out, so you need to uncomment it first to send a message like I did.

maxholgasson, 2020-05-27:
Nice! btw Do both Dio0 and DIO1 need interrupt capable pins? I would need at least one interrupt pin on the 328P for something else

neo7530, 2020-05-28:
Yea, you need both interrupt pins. Dio1 is for cad detect, so the avr will wake up if a cad occurs.

IoTThinks, 2020-05-28:
@neo7530 I only have dio0 connected to MCU.

Can I short circuit DIO0 and DIO1 to use CAD too?
So basically, dio0 and dio1 connect to the same GPIO.

neo7530, 2020-05-28:
@IoTThinks you can separate the Dios with a diode, but this wouldn't work with this code, because it have 2 separate int handlers and flags.maybe you can modify the code. Don't have a clue if this works as expected then.

neo7530, 2020-05-28:
@chocotov1 This works, bud sadly enableInvertIQ and disableInvertIQ are not working :(

morganrallen, 2020-05-28:
You do not need both DIO0 & DIO1 for CAD detection. It is possible with just DIO0. 
[image]
As you can see DIO0 has `CADDone`, and DIO1 has `CADDetected`. The way this runs is...
* User sets CADMode
* Channel Activity *IS* detected
* `CADDetectedMask` interrupt flag is set
* `CADDetected` triggered on `DIO1`
* `CADDone` triggers on `DIO0`

Even without `DIO1` the `CADDetectedMask` can be check after `DIO0` is triggered.

neo7530, 2020-05-28:
Have tested CAD Mode again. I am not going to get it below 4mA power consumption while sleeping. Have ripped off LDO from Arduino, Power-LED, Crystal and flashed 8MHz internal Bootloader. Then i tested again, and got 4mA while CAD, 11mA while receiving and 120mA while Transmitting. (RFM98W Module). Do i overlook something? How do you get 0.6mA?

chocotov1, 2020-05-28:
> You do not need both DIO0 & DIO1 for CAD detection. It is possible with just DIO0.

I implemented the idea of @morganrallen with the interrupt flag / mask in my cadmode branch. The accompanying example sketch is LoRaReceiverCadDemo328pDIO0. For receiving with the interrupt pins you'd need one for RxDone and RxTimeout so I modified that to receiving without the DIO pins.

@maxholgasson: If you run out of pins again check out the pin change interrupt that's available on all pins.

@neo7530: great!

maxholgasson, 2020-06-04:
That's awesome!

neo7530, 2020-06-04:
I have added some features in my repo. (Full working Gateway with invertIQ and a periodical beacon)

neo7530, 2020-06-05:
Have done some power measurements:

//WDTCSR = WDTO_1S; //SF11 (receiving not every packet) 50uA / SF12 100uA (receiving not every packet) 
//WDTCSR = WDTO_500MS; //SF11 430uA / SF12 1.5mA              recommended for SF12
//WDTCSR = WDTO_250MS; //SF10 600uA / SF11 1.2mA / SF12 2.1mA recommended for SF11
//WDTCSR = WDTO_120MS; //SF9 460uA / SF10 900uA / SF11 1,8mA  recommended for SF10
WDTCSR = WDTO_60MS; //SF8 480uA / SF9 870uA / SF10 1.57mA     recommended for SF9
//WDTCSR = WDTO_30MS; //SF7 540uA / SF8 920uA / SF9 1.59mA    recommended for SF7 & SF8

maxholgasson, 2020-06-16:
@neo7530 and @chocotov1 will your code also work with a SAMD21 used in an Arduino Zero? A Im running out of space, ram, pins and I'm looking for an alternative to an 328P

chocotov1, 2020-06-18:
@maxholgasson I don't have a Arduino Zero so i'm not 100% certain. My code specifically sets up the watchdog registers of the 328p and also uses the sleep functions of avr/wdt.h.

A similar approach is probably also possible on the SAMD21. You just need to be able to put the chip in a low power state for say 15 or more ms. The SAMD21 might offer alternative ways other than the watchdog to achieve that.

chocotov1, 2020-06-19:
@maxholgasson SAMD21 sleeping info:
https://www.youtube.com/watch?v=wmWqkJ97Zsc
https://www.arduino.cc/en/Reference/ArduinoLowPower

maxholgasson, 2020-06-19:
OK, yes as far as I remind avr/wdt.h won't work an a SAMD21 but as you shown there are other libraries or commands for this purpose.
So I would have to change just the sleeping part to use it on a SAMD21?
e.g change the content of go_to_sleep() with the according sleepfunction for 15ms sleep of the SAMD21?

chocotov1, 2020-06-19:
Yeah, that should work. You also need make sure the interrupt for DIO0 is setup correctly.

I saw the in the video I linked that the power consumption of the mkr1000 in sleep was 580 uA. That's a lot more than the 8 uA I see using the wdt on the 328p. The 328p cadmode demo I presented here uses 590 uA while actively receiving messages.

maxholgasson, 2020-06-20:
Yeah 590µA is amazing. However I need a bigger MCU for menu, display etc. I'm gonna try it with the SAMD21 to get your code running as sone as I got my prototyping board done. Other option is to connect a 328P to a SAMD21 and exchange data over I2C or SPI or so

IoTThinks, 2020-06-25:
@chocotov1 @maxholgasson : The following code is for Arduino only, right?
I'm using ESP32, What should I do for ESP32?
Thanks a lot.

```
 // wdt sleep settings:
  // 10.8 Watchdog Timer
  WDTCSR |= 1<<WDCE | 1<<WDE;

  // Table 10-3. Watchdog Timer Prescale Select
  //WDTCSR = WDTO_8S; // 1<<WDP3 | 1<<WDP0;           // 8 seconds
  //WDTCSR = WDTO_4S; // 1<<WDP3;                     // 4 seconds
  //WDTCSR = WDTO_2S; // 1<<WDP2 | 1<<WDP1 | 1<<WDP0; // 2 seconds
  //WDTCSR = WDTO_1S; // 1<<WDP2 | 1<<WDP1;           // 1 seconds
```

chocotov1, 2020-09-27:
@IoTThinks 
These settings are only for the 328p. I don't have experience with the ESP32.

chocotov1, 2020-09-27:
> Yeah 590µA is amazing. However I need a bigger MCU for menu, display etc. I'm gonna try it with the SAMD21 to get your code running as sone as I got my prototyping board done. Other option is to connect a 328P to a SAMD21 and exchange data over I2C or SPI or so

I've tried a SPI relay setup with 2 MCUs (Pro Minis running 8 MHz). I had to look up this technique, but with a little effort I can confirm it can be done. I made a little demo video:

https://www.youtube.com/watch?v=vsuVu33MASg

As for the alternative with SAMD21, i've bought a couple of SAMD21 boards and looked into that too.

The challenge there is not only the Lora part (with cad mode) but more so getting down with the details of the SAMD21. The SAMD21 has some advanced features like the builtin RTC with 32 kHz crystal. Another huge benefit is that it's possible to debug your program with a debugger.

The problem is that SAMD21 is more complex, the datasheet is bigger and sadly only a few people (compared to AVR the Arduinos) are using the more advanced features.

In the meantime I've tackled the challenge of getting the SAMD21 to sleep for a short time (32 ms for instance). One problem that arises is that you then shouldn't try to use to USB cable anymore. It becomes unusable. Uploading code is then only possible via the SWD pads (all boards seems to have these exposed). You need a programmer for this. I have the Atmel ICE.

Some general SAMD21 boards with Lora tips i've gathered so far (note that I still haven't made a CAD MODE solution for SAMD21 yet):
- MKRWAN1300 isn't suited for low power CAD MODE: The enclosed SX1272 Lora chip is controlled by another MCU
- MKRZERO or Adalogger M0 with seperate RFM95 seems to work ok
- Adafruit Feather M0 LoRa is probably the best alternative for a sleek SAMD21 Lora solution
- Sparkfun Pro RF is also ok, downside: The blue activity LED doesn't light up at lower voltage, form factor not nice for breadboard

With some tweaks in the SAMD21 Arduino core, I made some board variants with lower clock speeds. This way the SAMD21 boards also work at lower voltages (like with 2x NiMH battery setup). At low voltage the power consumption during sleep is around 250 uA. I suspect it can be lowered a bit with some hardware mods. Since these boards are so expensive i won't mess with them for now.

IoTThinks, 2020-09-28:
My esp32 now can reach 60uA deepsleep with LoRa.
When I switch off power for LoRa, I can reach 30uA.

You can check how much your LDO consumes to go under 60uA.

Btw, I check CAD mode. It costs like a RX so I guess we can not use it during deep sleep.
Unless someone has greater idea.

chocotov1, 2020-09-30:
CAD mode with SAMD21 / MKRZERO:

https://youtu.be/tunuQ5vRMbY

I'll make a similar SAMD21 sketch with serial output instead and put add it to my fork this week.

maxholgasson, 2020-10-18:
@chocotov1
Thanks for this awesome demonstration.

>builtin RTC with 32 kHz crystal. Another huge benefit is that it's possible to debug your program with a debugger.
That's true, however I could get to run either one of these^^ Tried using a debugger last year with eclipse I think but it was too complicated.

>during sleep is around 250 uA. 
Mh yes the SAMD21 is quite power hungry in comparison toa 328P. I think if the SPI communication between 328P and SAMD21 works -like you proofed- I would add a mosfet to the circuit and cut off the SAMD21 power.

maxholgasson, 2020-10-27:
@chocotov1 
In your sketch _LoRaReceiverCadDemo328pDIO0_  you using `    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }` to print out the packets. i would like to transmit at least two different vales: first packet should contain the sensor's ID and the second packet the battery voltage.

Does anyone have a clue how to transmit and process the data so I can use those two values?

chocotov1, 2020-10-27:
@maxholgasson 
The while loop printing LoRa.read() in the example prints all bytes in the message. Notice that this works well with arbitrary / unknown message lengths.

For your sensor data you can define a custom message format. For instance the first byte could be the sensor id from 0 till 255. The second byte could be the temperature from 0 till 255.

```
byte sensor_id   = LoRa.read();
byte temperature = LoRa.read();
```
These values are not so practical, so you are free to organize it in your own format. It's good practice to not waste bytes because then the radio transmissions takes longer. For instance if you have 8 sensors, you'd ony need 3 bits for the sensor id field.

See also my reply of 24 May and look here: https://www.thethingsnetwork.org/docs/devices/bytes.html

To efficiently organize your sensor data into different bytes you'll have to understand the bit operations: https://playground.arduino.cc/Code/BitMath/

maxholgasson, 2020-11-01:
@chocotov1  Thanks a lot, it's working now. Nice to play around with those RFM95 boards.

>you'll have to understand the bit operations

It took quite some time till I understood what you mean but now I got it. Thanks for the hint!

maxholgasson, 2020-11-11:
@chocotov1 
> I'll make a similar SAMD21 sketch with serial output instead and put add it to my fork this week.
HI,
 I looked carefully at your forks to understand how you managed to get that working with a SAMD21. 
Did I understand it right that you modified the fork _ArduinoCore-samd_ to wake up the SAMD21 every few ms with a watchdog/internal timer? 
Basically the MCU wakes up and then executes the commands you already did in the _LoRaReceiverCadDemo328pDIO0_ sketch but you need to modify there the adresses of the register so that it fits to the new chip SAMD21?
>The challenge there is not only the Lora part (with cad mode) but more so getting down with the details of the SAMD21.

I hope to understand that someday and do my very best to follow the ideas.

chocotov1, 2020-11-12:
@maxholgasson 

I actually didn't add a SAMD21 example sketch yet, until today! See my fork of arduino-LoRa in the cadmode branch: examples/LoRaReceiverCadDemoZeroDIO0/LoRaReceiverCadDemoZeroDIO0.ino. In this setup I observed a power consumption around 1.2 mA @ 3.3 V during CAD mode.

I am hesitant to publish this sketch. I enjoyed the challenge of making it work. But for someone like you, probably just needing something that gets the job done, I think you'd be better off using something less experimental. The SAMD21 is a niche in it's own right, combine that with wanting to use CAD mode of the SX1276 and you'll notice there only 2 people that are interested in that.

Before you try to upload that sketch, **I need to emphasize that you really need to be using the SWD pads and a programmer like the Atmel ICE to upload the sketch!** Putting the MCU to sleep like this sketch does messes up the USB connection. It's better to just **don't** connect the USB cable at all.

In my fork of ArduinoCore-samd I added some board definitions with lower clock speeds and without the bootloader. This way the board still works at low voltages around 2 V. The no bootloader variants also save the hassle of having to separately upload the bootloader each time. Of course, without a bootloader you'll also won't be able upload sketches over the USB cable anymore.

This is how I deployed my fork of ArduinoCore-samd on my computer: I had first installed the normal Arduino SAMD21 core using the library manager. Afterwards I cloned my fork in the same folder:
```chocotov1@host:~/.arduino15/packages/arduino/hardware/samd/1.8.8$ git config --get remote.origin.url
git@github.com:chocotov1/ArduinoCore-samd.git
```
Notice that Serial1 is used in the sketch. You can read from it by using a serial adapter and pin 14 (TX) of the MKRZERO.

About page 38 of the datasheet: Yeah, that's where learned the details of CAD mode. The places where it says STANDBY are actually points, where Lora module automatically puts itself in standby mode.

maxholgasson, 2020-11-12:
Hi, ah cool, I'll have a look and see if I can get it to work.

> I am hesitant to publish this sketch. I enjoyed the challenge of making it work. But for someone like you, probably just needing something that gets the job done, I think you'd be better off using [...]

Actually both is true ;) Since a couple of years I made again and again efforts to have proper low power solution for different ideas I had. I'd like to build small environment sensors which can work outside and are only battery powered or other remote controls where both sender and receiver are battery powered for monitoring. Eg ultrasonic detectors connected to a sound recorder for monitoring bats.🦇 Sometimes there appeared some hopefull approaches but on a closer look there was some kind of disillusion, either too high power consumption or too high latency due to duty-cycle. I even bought a programmer for a rfm219SW, a very simple receiver I came across tearing down various cheap chinese battery powered door bells and stuff that worked for month with battery and took it as a challenge to control this module but so far didn't succeed. I really would like to understand how that works and your sucesfull working RFM95 CAD code is a great thing. I wish I would understand more and maybe I could contribute some day to the code as soon as I understand more. During my research I also found [LowPower Lab](https://lowpowerlab.com/forum/index.php) First I was hopefull as they where looking for a similar solution: a battery powered receiver but none of them found a reliable working way to do it. I could imaging they would also appreciate your demo code. So for me it's both a challenge and to have something I can use for further ideas like my sensors or remotes.

>I need to emphasize that you really need to be using the SWD pads and a programmer like the Atmel ICE to upload the sketch!

Thanks for the important advice, in my case I designed a own pcb with a bare SAMD21 and only a SWD connector for programming - there is no USB. I'm using a modified Arduino Zero as a programmer as I couldn't afford a proper ICE^^ It used to function as a debugger as well but it was to confusing to get it to run - a debugger would probably very handy.

chocotov1, 2020-11-13:
In that case, i'll gladly keep helping you with the Lora and SAMD21 experiments. Maybe you can make a repository of your project with some background info of the hardware. We can then look at same code and I can maybe verify your problems.

That's really advanced with your PCB and SAMD21. Did you get the my example working? You might be seeing lower power consumption values, since the Arduino SAMD21 boards have some extra hardware like the vreg.

I checked out my example again, uploading it the normal Arduino way (without programmer). I was surprised to see notice that the code doesnt introduce the the massive USB problems anymore. The problem must have been solved, since in the published version I took some measures to disable USB. Double press the reset button if you want to upload again.

For anyone else interested in trying, I still do advice to only upload the SAMD21 sketch if you also have programmer. After you've uploaded code that breaks the USB connection, the programmer is needed so you can upload other code again.

maxholgasson, 2020-11-13:
Ok I'm trying to make repository(I need to look it up first what I have to include and hwo a repository is built) 
I made two different test bords for the SAMD21. Haven't test your code yet on the SAMD21 but will on coming weekend.
>You might be seeing lower power consumption values, since the Arduino SAMD21 boards have some extra hardware like the vreg.
Yes I'm very curious. It's a while ago I made the pcbs but put them away as the consumption was too high (during that time I used RX mode without CAD, so all time alive)
I then follwed the idea to connect a 328P to the SAMD21 and use the SAMD21 only for OLED, buttons, etc. So I'm open for everything. I recently came across a MCU [STM32L152RE](https://www.mouser.de/ProductDetail/STMicroelectronics/NUCLEO-L152RE?qs=fK8dlpkaUMv3JswoDBtTmg==) which also sounds decent and maybe more close to the low power 328P than a SAMD21. It's supported by this LoRa-lib and can be programmed with the regular Arduino IDE
>I was surprised to notice that the code doesn't introduce the massive USB problems anymore.
I just remembered that I got in trouble first time using the sleep and wakeup by interrupt with the SAMD21. I first had to cut the power after uploading otherwise it didn't work properly. This error was hard to find.

maxholgasson, 2020-11-15:
I figured out that I couldn't use my full equipent pcb as there where to many components that also drawing some power. I now switched for power consumption testing to my 1:1 breakoutboard and can confirm 1.2mA with a multimeter. I couldn't find the original schematic of the adapter board as I modified the original schematic but it's basically just a 1:1 pinput with a crystal attached.
interestingly my cheap oszi displays an average of around 900µA. I also tested the SAMD21 in sleep mode without the RFM95 attached and I can't get it to a lower consumption of 500µA (even with all pins set to input).  https://github.com/maxholgasson/universalLoRaReceiver/wiki

IoTThinks, 2020-11-15:
If unable to reach under 500uA, check the LDO first.

Some LDOs have high quiescent current.

maxholgasson, 2020-11-15:
Hi, I don't have a LDO. It's directly connected to a linear power supply

chocotov1, 2020-11-18:
Awesome!

I also found some improvements for my SAMD21 power consumption experiments.

The default setup of the pins in wiring.c turns out to be less efficient. By defining VERY_LOW_POWER in boards.txt the power consumption on my Lora CAD mode test dropped from 1.25 mA to 0.75 mA. I checked out the Moteino boads.txt and they do the same thing:
```
mkrzero.build.extra_flags=-DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ {build.usb_flags} -DVERY_LOW_POWER
```
I read about it over here: https://github.com/adafruit/Adafruit_SleepyDog/issues/17

By commenting out the following line I saved around 10 uA in a test sketch with RTC interrupt every 30 ms, EIC (pin change interrupt) and a blinking led:
```
 //NVMCTRL->CTRLB.reg |= NVMCTRL_CTRLB_SLEEPPRM_DISABLED;           // Disable auto power reduction during sleep - SAMD21 Errata 1.14.2
```
I was encouraged to remove it after reading https://community.atmel.com/comment/1851566#comment-1851566. 
The workaround is described in SAM-D21-Family-Silicon-Errata-and-DataSheet-Clarification-DS80000760D.pdf. I checked all my SAMD21 Arduino boards and all are SAMD21 silicon revision D and therefore unaffected.

One more thing I noticed, is that my MKRZERO has around 100 uA higher power consumption when the voltage is lower than 3.3 V. I suspect it could be caused by the something like the vreg or the battery charging circuit.

Summary:
* Lora CAD mode test with oled display, 32 ms intervals,  at 3.3 V:
  * MKRZERO 3.3 V: 0.75 mA
  * Pro Mini: 0.60 mA
* Lowest consumption I got with MKRZERO, RTC, EIC and blinking led experiment: 66 uA

> I had to upload the bootloader after the sketch first in order to get it working every time. Don't know why but that way it's working

You can change that by editing boards.txt. Change _flash_with_bootloader.ld_ into _flash_without_bootloader.ld_ example:
```
mkrzero.build.ldscript=linker_scripts/gcc/flash_with_bootloader.ld

mkrzero_no_bl.build.ldscript=linker_scripts/gcc/flash_without_bootloader.ld
```

[9 more replies]