sandeepmistry/arduino-LoRa #368: Specify SPI bus

beowulff, 2020-05-26:
I am having problems using TFT_eSPI and arduino-LoRa at the same time. I have an SPI LCD display with no CS line, so it needs to be driven off of a separate bus. The TFT_eSPI driver is supposed to be able to specify which bus to use, but no matter which one I select, as soon as I call init_loRa(), the LCD starts to act wonky. The odd thing is that it doesn't fail outright - but some functions (like tft.fillScreen()) don't work.

So, I would like to force arduino-LoRa to use a specific bus - either HSPI or VSPI. How can I do this?

(FWIW - I'm reasonably sure that I have all the device's pins specified correctly, and none are shared between the two devices (LCD and LoRa module)).

IoTThinks, 2020-05-27:
```
LoRaClass LoRa2;
SPI.begin(LORA2_SCK, LORA2_MISO, LORA2_MOSI);
  LoRa2.setPins(LORA2_SS, LORA2_RST, LORA2_DIO012);
  
  while (!LoRa2.begin(LORA2_FREQ_BASE)) {
    log("[LoRa 2] Starting LoRa failed!");    
    LORA2_Status="FAILED";
    delay(1000);
  }
```

IoTThinks, 2020-05-27:
Can try to usw VSPI insteas of SPI.
The above code for the 2nd LoRa.
The first one just uses “LoRa”

beowulff, 2020-05-27:
That gets me a 2nd instance of the LoRa driver - how do I tell "SPI.begin" to use VSPI or HSPI?

I'm not a C++ programmer, so I have some difficulties with this stuff...

IoTThinks, 2020-05-27:
SPIClass spiRFID(HSPI);
Then in your setup call:
spiRFID.begin(14,12,13,15); //CLK,MISO,MOIS,SS

https://github.com/espressif/arduino-esp32/issues/1219

beowulff, 2020-05-27:
Thanks, I figured it out:

SPIClass spiLoRA(HSPI);
	
	LoRaClass LoRa2;
	
	LoRa2.setPins(SS, RST, DI0);
	
	spiLoRA.begin(SCK,MISO,MOSI,SS);
	
	while (!LoRa2.begin(915E6)) {
		ESP_LOGI(TAG,"[LoRa 2] Starting LoRa failed!");
		delay(1000);
	}

But - it still doesn't work! 
I need to get my logic analyzer hooked up to see if I can figure out what's going on. No matter how I have the two SPI busses configured, as soon as I call SPI.begin, the TFT_eSpi driver stops working correctly.

beowulff, 2020-05-27:
I think I figured it out:

SPIClass spiLoRA(HSPI);
	
	spiLoRA.begin(SCK,MISO,MOSI,SS);
	
	LoRa.setSPI(spiLoRA);

Portia-Lin, 2020-05-31:
> I think I figured it out:

Did you connect the display and LoRa via SPI? If so, could you help me with this problem?

beowulff, 2020-05-31:
Yes, I am using a NiceRF LoRa Module on one SPI bus and my LCD (some cheap 1.8” 240x240 SPI panel, with no CD pin) on the other SPI bus. I am using the following pins for the LoRa Module:
MOSI - 23
MISO - 19
SCK - 18
NSS - 5
RESET - 17
DIO - 22

And the following pins for the LCD:
MOSI - 4
SCK - 32
RESET - 2
DC - 27

The LCD is on the VSPI bus, and the LoRa is on the HSPI bus.

Portia-Lin, 2020-05-31:
Ohh, i have arduino nano... I don`t have this pins

IoTThinks, 2020-05-31:
@Portia-Lin Arduino may have 1 SPI only.
You need to share mosi, miso, sck to both spi lora and lcd.

And separate cs/ss for lora and lcd.
Spi will take turn to poll lora and lcd when needed but not concurrently.

closed 2020-11-12 by IoTThinks