sandeepmistry/arduino-LoRa #369: Not showing anything

OhhPian, 2020-05-29:
Hi, need some help here. Just recently bought this Lora to test for new project.
[https://shopee.com.my/product/110869073/5615064681](url)

Im using Arduino IDE, wanna testing the Lora first, im using example coding Lora Sender and Lora Receiver, changing the frequency to 433 on both board. After that im open serial monitor and change baud to 9600 but it shows nothing, an i missing something?

IoTThinks, 2020-05-30:
You may need to use Tera Term to view the console.
Console of Arduino IDE is not very good.

OhhPian, 2020-06-01:
*update*

The code is fine, maybe my board or arduino IDE have problem. So i just upload again and again in order to ensure the LoRa is working.

closed 2020-11-12 by IoTThinks