sandeepmistry/arduino-LoRa #373: What is the length of thepreamble?

Portia-Lin, 2020-06-07:
Hello. I want to calculate data rate in byte bps.

I used the following lora settings
Spread Factor: 12
Bandwidth: 62.5
The table shows that the data rate will be 146 bytes per seconds.

[image]

The structure of the package is shown in the figure:

[image]

Now i want to calculate how many seconds my message will be transmitted. I need to know the programmed length of the preamble. What is it for your library? And maybe you can tell me how to calculate the length of the header.
Then I will be able to know the length of the whole packet and calculate the transmission time

Portia-Lin, 2020-06-07:
In the library, I found tapes that are responsible for the length of the preamble:
#define REG_PREAMBLE_MSB         0x20
#define REG_PREAMBLE_LSB         0x21
But I still don't understand the length ...

IoTThinks, 2020-06-08:
Check here https://www.loratools.nl/#/airtime

Portia-Lin, 2020-06-08:
Thank you, there is a standard length 8 symbols. 0x20 and 0x21 are listed in the library. What is the length of 0x20? I don`t understand this

morganrallen, 2020-06-08:
Those are the address location for the preamble length setting. Unfortunately we do not provide a way to read that back yet, but if you have not changed it, it will stay at the default of 0.

Section 4.1.1.7 explains how to calculate airtime.

Portia-Lin, 2020-06-08:
Therefore, the length of the preamble is 0?

Portia-Lin, 2020-06-08:
I tried to calculate it and I don't understand what max(numeric, 0) means. And I don't know if I'm doing it right.

[image]

IoTThinks, 2020-06-09:
The preamble should be > 0 as it is used to detect if a packet is a LoRa packet.
You can set it here: https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#preamble-length.

The wording says: Supported values are between 6 and 65535.

I believe transferring a preamble bit or a normal data bit take the same period of time.

Portia-Lin, 2020-06-09:
But how to calculate the function max ()

Portia-Lin, 2020-06-11:
Now i want to use the center frequency of channel 4 of LPD433 - 433.150. I inserted the following code:
```
LoRa.begin(433.150E6)
```
Everything works, but does it really work at this frequency?

IoTThinks, 2020-06-13:
I checked with my SDR.
This library really set the frequency right with LoRa AI-Thinker.

Portia-Lin, 2020-06-14:
> I checked with my SDR.

Thank you very much!

closed 2020-06-18 by Portia-Lin