sandeepmistry/arduino-LoRa #375: Confirmation of data transfer from transmitter to receiver?

Portia-Lin, 2020-06-18:
I am developing a system with many sensors (10 pieces). After sending a signal to the sensors, they must transmit information to the receiver. The signal is sent to all sensors at the same time, so immediately after that they will start sending data. But because they send a signal at the same time, the receiver will not receive all the signals. How to get out of this situation? Or in the absence of confirmation, the sensors send the signal again, how to do it?

Arismane, 2020-07-04:
Why not programming each sensor to have a different small delay before starting to send the data from the time it received the signal so the receiver doesn't have a problem with the reception?

brainelectronics, 2020-11-04:
@Portia-Lin you could request an ACK from each sensor. This ACK could be sent by each sensor after recieving the signal from the master, and the master could send an ACK to the sensor after receiving their sensor value.
This works of course only if you do not send a broadcast message.
Check my PR #406

closed 2020-11-12 by IoTThinks