sandeepmistry/arduino-LoRa #376: Does the SyncWord actually do anything?

beowulff, 2020-06-18:
No matter what I set the SyncWord to on the transmitter and receiver, the receiver gets correct packets from the transmitter. I want to be able to have separate networks that ignore each other if the SyncWords don't match, but as far as I can tell, they seem to be ignored. I am testing at very close range (5'). I am using continuous receive, and I would like to not be notified of a packet if the SyncWords don't match.

beowulff, 2020-06-18:
OK, I'll amend my post - there does seem to be some discrimination, but it's poor. If I set the transmitter to be 0x4E and the receiver to be 0x12, most of the time (but not all), the receiver won't get the packet. If the two SyncWords are close, like 0x34 and 0x35, the packets are received every time.
Not very useful...

IoTThinks, 2020-06-19:
SyncWord wont do the job well.
Similar SyncWord will let the messages to be received.
Even Semtech recommend to put an dst id in front of the message.

closed 2020-11-12 by IoTThinks