sandeepmistry/arduino-LoRa #377: Sleeping the radio

BrendonHills, 2020-06-18:
I want to put the radio to sleep when I put the processor to sleep & I can see that there are three possible ways to save radio power consumption:

LoRa.idle();
LoRa.sleep(); and
LoRa.end()

Which will save most power whilst the main processor is sleeping?
Is it quick, relaible and easy to 'wake-up' from these states?
How do I 'wake-up' or re-start from any of these states - must I run 'LoRa.begin(frequency);' in the ISR or main loop, or some other command?

I am using the Adafruit Feather 32u4 RFM95 HOPE RF boards.

Thank you for any assistance.

IoTThinks, 2020-06-19:
LoRa.end then LoRa.begin
After deep sleep, the MCU runs setup() again anyway.

BrendonHills, 2020-06-19:
> Reply to: sandeepmistry/arduino-LoRa <reply@reply.github.com>

Thanks! I didn’t realise setup() was run again – how does that fit in with the ISR?

If the hardware interrupt triggers the ISR, does it run setup after the ISR or before?

IoTThinks, 2020-06-19:
Try it yourself.

closed 2020-06-19 by BrendonHills