sandeepmistry/arduino-LoRa #379: Guru meditation error while using onTxDone /onRxDone on ESP32

neo7530, 2020-06-22:
I get crashes while using onrxdone or ontxdone on ESP32 module:

``Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4008b356  PS      : 0x00060234  A0      : 0x8008a513  A1      : 0x3ffbe660  
A2      : 0x3ffb8984  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060223  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074  
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060221  A15     : 0x00000000  SAR     : 0x0000000e  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
Core 1 was running in ISR context:
EPC1    : 0x400899ae  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008b356

Backtrace: 0x4008b356:0x3ffbe660 0x4008a510:0x3ffbe680 0x400887ef:0x3ffbe6a0 0x400d454d:0x3ffbe6e0 0x400d1c79:0x3ffbe700 0x400d1d6c:0x3ffbe730 0x400d1da0:0x3ffbe750 0x400d24dd:0x3ffbe770 0x40080eb6:0x3ffbe790 0x40080f01:0x3ffbe7b0 0x40084dc1:0x3ffbe7d0 0x4000bfed:0x3ffb1e90 0x40089b19:0x3ffb1ea0 0x4008879b:0x3ffb1ec0 0x400d454d:0x3ffb1f00 0x400d1c79:0x3ffb1f20 0x400d1d6c:0x3ffb1f50 0x400d229d:0x3ffb1f70 0x400d16a0:0x3ffb1f90 0x400d5265:0x3ffb1fb0 0x400889e9:0x3ffb1fd0

Core 0 register dump:
PC      : 0x400899b0  PS      : 0x00060034  A0      : 0x8008a1c1  A1      : 0x3ffbe150  
A2      : 0x3ffbee78  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060021  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0x3ffc090c  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000001  
A14     : 0x00060520  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400899b0:0x3ffbe150 0x4008a1be:0x3ffbe180 0x40088ad7:0x3ffbe1a0 0x4008b1cd:0x3ffbe1c0 0x40084dca:0x3ffbe1d0 0x400ed757:0x3ffbbff0 0x400d7dcb:0x3ffbc010 0x4008a18a:0x3ffbc030 0x400889e9:0x3ffbc050

Rebooting...
``

IoTThinks, 2020-06-23:
Install espexceptiondecoder into Arduino IDE to see the exception details.

neo7530, 2020-06-23:
This ist with onReceive:

PC: 0x400d504e: String::String(float, unsigned char) at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\WString.cpp line 111
EXCVADDR: 0x00000000

Decoding stack results
0x400d504e: String::String(float, unsigned char) at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\WString.cpp line 111
0x400d1aa6: onReceive(int) at C:\Users\mw\Documents\Arduino\OLED_LoRa_Receiver_ESP32/OLED_LoRa_Receiver_ESP32.ino line 683
0x400d246f: LoRaClass::handleDio0Rise() at C:\Users\mw\Documents\Arduino\libraries\arduino-LoRa-master\src\LoRa.cpp line 676
0x40080eb6: LoRaClass::onDio0Rise() at C:\Users\mw\Documents\Arduino\libraries\arduino-LoRa-master\src\LoRa.cpp line 715
0x40080f01: __onPinInterrupt at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-gpio.c line 220
0x40089b19: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x40088937: uxQueueMessagesWaiting at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1753
0x400d5809: uartAvailable at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-uart.c line 269
0x400d4731: HardwareSerial::available() at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\HardwareSerial.cpp line 111
0x400d1391: loop() at C:\Users\mw\Documents\Arduino\OLED_LoRa_Receiver_ESP32/OLED_LoRa_Receiver_ESP32.ino line 163
0x400d51b1: loopTask(void*) at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x400889e9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

neo7530, 2020-06-23:
And this with onTxDone:

PC: 0x4008b358: vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c line 188
EXCVADDR: 0x00000000

Decoding stack results
0x4008b358: vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c line 188
0x4008a510: vTaskPlaceOnEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 2901
0x400887ef: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1590
0x400d467d: spiTransaction at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-spi.c line 722
0x400d1da9: SPIClass::beginTransaction(SPISettings) at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI\src\SPI.cpp line 135
0x400d1e9c: LoRaClass::singleTransfer(unsigned char, unsigned char) at C:\Users\mw\Documents\Arduino\libraries\arduino-LoRa-master\src\LoRa.cpp line 703
0x400d1ed0: LoRaClass::readRegister(unsigned char) at C:\Users\mw\Documents\Arduino\libraries\arduino-LoRa-master\src\LoRa.cpp line 689
0x400d260d: LoRaClass::handleDio0Rise() at C:\Users\mw\Documents\Arduino\libraries\arduino-LoRa-master\src\LoRa.cpp line 658
0x40080eb6: LoRaClass::onDio0Rise() at C:\Users\mw\Documents\Arduino\libraries\arduino-LoRa-master\src\LoRa.cpp line 715
0x40080f01: __onPinInterrupt at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-gpio.c line 220
0x400d4768: spiTransferByteNL at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-spi.c line 785
0x400d1dda: SPIClass::transfer(unsigned char) at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI\src\SPI.cpp line 158
0x400d1ea3: LoRaClass::singleTransfer(unsigned char, unsigned char) at C:\Users\mw\Documents\Arduino\libraries\arduino-LoRa-master\src\LoRa.cpp line 704
0x400d2377: LoRaClass::explicitHeaderMode() at C:\Users\mw\Documents\Arduino\libraries\arduino-LoRa-master\src\LoRa.cpp line 694
0x400d23c1: LoRaClass::parsePacket(int) at C:\Users\mw\Documents\Arduino\libraries\arduino-LoRa-master\src\LoRa.cpp line 224
0x400d17d8: loop() at C:\Users\mw\Documents\Arduino\OLED_LoRa_Receiver_ESP32/OLED_LoRa_Receiver_ESP32.ino line 173
0x400d53a5: loopTask(void*) at C:\Users\mw\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x400889e9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

IoTThinks, 2020-06-23:
Source code for this part?
Are you using FreeRTOS?

The bellow error may indicate you use too much MCU in core 1.  Hence, esp32 decided to kill your process and reboot.
Check any process with too much things in onReceive....
Guru Meditation Error: Core 1 panic'ed (Interrupt wdt timeout on CPU1)

neo7530, 2020-06-23:
This is the sample Code from here… 

OnTxDone does disableinvertiq() only and crashes… No, i use plain Arduino.

IoTThinks, 2020-06-24:
> OnTxDone does disableinvertiq() only and crashes… No, i use plain Arduino.
In onTXDone, can try to set a bool flag only.
Then in main loop, set disableInvertIQ based on the flag.

neo7530, 2020-06-25:
Doesn't help. Crashs. :(

IoTThinks, 2020-06-26:
Show your code.

neo7530, 2020-06-26:
Not Working:
https://github.com/neo7530/arduino-LoRa/tree/master/examples/OLED_LoRa_Receiver_ESP32_notworking

Working:
https://github.com/neo7530/arduino-LoRa/tree/master/examples/OLED_LoRa_Receiver_ESP32_working

IoTThinks, 2020-06-26:
Both of your codes are likely not working properly and will miss a lot of incoming packets.
"`Guru Meditation Error: Core 1 panic'ed (Interrupt wdt timeout on CPU1)"
=> Mean you have a lot of blocking or long running functions in main loop

From my experience, a 200-line function will unlikely to work.
Refactor your code or test again with the simplest LoRa receiver in example.

neo7530, 2020-06-26:
The main-loop does nothing except waiting for a incoming message via Lora or serial. If a serial command is send, the main loop parses the characters and finally send them via Lora. I get all the packets I sent.

Jose2059, 2020-06-29:
[image]

Check if your prepossessed file has ICACHE_RAM_ATTR . ESP32 needs this attribute to run in ram instead of flash.
You may want to create a copy of the library, and modifiy the ISR call back:

[image]

neo7530, 2020-06-30:
> Check if your prepossessed file has ICACHE_RAM_ATTR . ESP32 needs this attribute to run in ram instead of flash.
You may want to create a copy of the library, and modifiy the ISR call back:

This is included in LoRa.c.

JRAlvarez2017, 2020-07-01:
The ESP32 (and the ESP8266) interrupt callbacks need to ne in RAM, see here:
https://arduino-esp8266.readthedocs.io/en/latest/reference.html
[image]

This LoRa library uses a conditional #define: if the preprocesor finds "#define ESP32" then line 713 of the LoRa.c is compiled replacing ISR_PREFIX with ICACHE_RAM_ATTR. However, when you compile with the ArduinoIDE, it does not find the define and the  interrupt function does not get the attribute. Writing "#define ESP32" in the ino sketch does not work. That is why I suggested you to copy the library and rename it. Or easier, you can just modify line 713 to make it like this:
[image]
I think I read once in a ExpressIff doc/blog that if you have functions inside the interrupt callback, you have to also add the attribute to them. So I use to modify this line (and more) as well:
[image]
But it may not be necessary, I never checked.
Nowdays, I use Visual Studio Code, and with it you can add your preprocessor defines.

One way to check if you have the attribute ICACHE_RAM_ATTR added to the callback functions is to load this sketch (in the examples) to one module (ESP32 + LoRa): 
[image]

and this other example to the other module:

[image]

If everything works fine, then the problem should be located in the code managing the interrupts, but I think you should verify with the examples first.

JRAlvarez2017, 2020-07-01:
[image]

neo7530, 2020-07-01:
Ahhh, thank you for sharing. I will try this @home and report back.

morgan-wild, 2020-08-14:
Same here with an ESP32, I tried ICACHE_RAM_ATTR and limited the code of my interrupt callback to:

```
void loRaOnReceiveInterrupt(int packetLength) {

    Serial.println("LoRa: Receive interrupt");

    String packet = LoRa.readString(); // Takes 2-4 ms
}
```

Because 2-4ms is may be too long for an interrupt callback, I also tried to use a task:

```
void loRaOnReceiveInterrupt(int packetLength) {

    // It is not possible to execute long running (>1ms) in a function called
    // by interrupt without fire the watch dog. So a  task is used for that.
    xTaskCreate(
        loRaOnReceiveTask,      // Function that should be called
        "loRaOnReceiveTask",    // Name of the task (for debugging)
        1000,                   // Stack size (bytes)
        NULL,                   // Parameter to pass
        4,                      // Task priority
        NULL                    // Task handle
    );
}

void loRaOnReceiveTask(void * parameter) {

    Serial.println("LoRa: Receive interrupt + task");

    String packet = LoRa.readString();

    vTaskDelete(NULL);
}
```

But no, I have still the same issue randomly after some receive:

```
LoRa: Receive interrupt + task
LoRa: Receive interrupt + task
LoRa: Receive interrupt + task
LoRa: Receive interrupt + task
LoRa: Receive interrupt + task
LoRa: Receive interrupt + task
LoRa: Receive interrupt + task
LoRa: Receive interrupt + task
LoRa: Receive interrupt + task
LoRa: Receive interrupt + task
LoRa: Receive interrupt + task
LoRa: Receive interrupt + task
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4008c07e  PS      : 0x00060e34  A0      : 0x8008b23b  A1      : 0x3ffbe660  
A2      : 0x3ffb8480  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060e23  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074  
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060e21  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400dbb90  LEND    : 0x400dbbf2  LCOUNT  : 0x00000002
Core 1 was running in ISR context:
EPC1    : 0x4008a6c2  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008c07e

Backtrace: 0x400d396c:0x3ffb8600 0x400896bd:0x3ffb8630

Rebooting...
```

Someone may save us?

dhorner, 2020-08-14:
Have you tried just setting a global to the packet length in the interrupt callback and in your main look check the variable to see if you need to read from Lora.  That way the interrupt returns almost immediately.  And you have the packet length as well.

int loraPacketLength = 0;
void loRaOnReceiveInterrupt(int packetLength) {
  loraPacketLength = packetLength;
}

void loop {
  if (loraPacketLength > 0) {
    Serial.println("LoRa: Receive packet");
    String packet = LoRa.readString(); // Takes 2-4 ms
    loraPacketLength = 0; // be sure to reset this to 0
  }
}

> Same here with an ESP32, I tried ICACHE_RAM_ATTR and limited the code of my interrupt callback to:
>
> ```
> void loRaOnReceiveInterrupt(int packetLength) {
>
>     Serial.println("LoRa: Receive [...]

GIPdA, 2020-08-15:
@morgan-wild Since you're using FreeRTOS, you should use a flag/semaphore to notify for an incoming packet from ISR, and wait for this flag in your task (or main loop), instead of creating a task from an ISR (not even sure that's supposed to work!).

Also, ```readString()``` seems way to long to me, it's juste supposed to read from registers. Unless it takes this long only when called from an ISR, can't really say what happens in that case, but I would advise against that, not safe.

morgan-wild, 2020-08-16:
> @morgan-wild Since you're using FreeRTOS, you should use a flag/semaphore to notify for an incoming packet from ISR, and wait for this flag in your task (or main loop), instead of creating a task [...]

GIPdA thank you for your answer. According to it, I tried:

```
int _loRaPacketLength = 0;

void setup() {

    [...]

    // Initialize the interrupt
    LoRa.onReceive(loRaOnReceiveInterrupt);
    LoRa.receive();

    // Initialize the reading task
    xTaskCreate(
        loRaMeshProcessTask,    // Function that should be called
        "loRaMeshProcessTask",  // Name of the task (for debugging)
        2000,                   // Stack size (bytes)
        NULL,                   // Parameter to pass
        3,                      // Task priority
        NULL                    // Task handle
    );

    [...]

}

void loRaOnReceiveInterrupt(int packetLength) {

    // Just store the length of the received packet
    _loRaPacketLength = packetLength;

}

void loRaMeshProcessTask(void * parameter) {

    while(true) {

        if (_loRaPacketLength > 0) {

            String packet = LoRa.readString();

            Serial.println("RX: " + packet);

            // Reset the packet length
            _loRaPacketLength = 0;
        }

        vTaskDelay(100 / portTICK_RATE_MS);
    }
}
```

And the result is (with 80 to 83 the number of the packet send by another loRa device):

```
RX: 80 {"state":{"time":419607,"type":"ground module","battery":100,"ram":0}}
RX: 81 {"state":{"time":424703,"type":"ground module","battery":100,"ram":0}}
RX: 82 {"state":{"time":429943,"type":"ground module","battery":100,"ram":0}}
RX: 83 {"state":{"time":435120,"type":"ground module","battery":100,"ram":0}}
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4008c084  PS      : 0x00060534  A0      : 0x8008b23b  A1      : 0x3ffbe660        
A2      : 0x3ffb8480  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001        
A6      : 0x00060523  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074        
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001        
A14     : 0x00060521  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400dbde4  LEND    : 0x400dbe46  LCOUNT  : 0x00000002        
Core 1 was running in ISR context:
EPC1    : 0x4008a6c2  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008c084        

Backtrace: 0x4008c084:0x3ffbe660 0x4008b238:0x3ffbe680 0x400894c3:0x3ffbe6a0 0x400dbb65:0x3ffbe6e0 0x400d43f1:0x3ffbe700 0x400d4580:0x3ffbe730 0x400d45b4:0x3ffbe750 0x400d48f5:0x3ffbe770 0x40080f95:0x3ffbe790 0x400810b5:0x3ffbe7b0 0x400852b1:0x3ffbe7d0 0x400d826d:0x3ffb1e30 0x400d84f5:0x3ffb1e90 0x401699d5:0x3ffb1ec0 0x400d9405:0x3ffb1ee0 0x400d943d:0x3ffb1f00 0x400d2b9a:0x3ffb1f20 0x400d2eaf:0x3ffb1f70 0x400d34de:0x3ffb1f90 0x400dc925:0x3ffb1fb0 0x400896bd:0x3ffb1fd0

Core 0 register dump:
PC      : 0x4008a6c2  PS      : 0x00060034  A0      : 0x8008aee9  A1      : 0x3ffbe150        
A2      : 0x3ffbf3f8  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001        
A6      : 0x00060021  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffc2324        
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000001        
A14     : 0x00060320  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000006        
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000        

Backtrace: 0x4008a6c2:0x3ffbe150 0x4008aee6:0x3ffbe180 0x400897ab:0x3ffbe1a0 0x4008bef5:0x3ffbe1c0 0x400852ba:0x3ffbe1d0 0x40169abf:0x3ffbc820 0x400df8ce:0x3ffbc840 0x4008aeb1:0x3ffbc860 0x400896bd:0x3ffbc880

Rebooting...
```

No packets are lost but the crash is still here, even with an extremely short interrupt callback.

It is very strange.

GIPdA, 2020-08-16:
Check where it crashes exactly by decoding the backtrace, it's not very useful otherwise. I suspect there's a function somewhere that sometimes waits too long.
The lib still does SPI transactions while in ISR before calling your callback, so maybe that's why.

You could also try to modify the lib to call ```LoRaClass::handleDio0Rise()``` outside of ISR context to see if that resolves the issue. I would modify ```LoRaClass::onReceive``` to replace the ```LoRaClass::onDio0Rise``` callback to another one that sets a flag, then manually call ```handleDio0Rise()``` in a separate task when the flag is set.

Jose2059, 2020-08-16:
> 
> 
> > @morgan-wild Since you're using FreeRTOS, you should use a flag/semaphore to notify for an incoming packet from ISR, and wait for this flag in your task (or main loop), instead of creating a task from an ISR (not even sure that's supposed to work!).
> > Also, `readString()` seems way to long to me, it's juste supposed to read from registers. Unless it takes this long only when called from an ISR, can't really say what happens in that case, but I would advise against that, not safe.
> 
> GIPdA thank you for your answer. According to it, I tried:
> 
> ```
> int _loRaPacketLength = 0;
> 
> void setup() {
> 
>     [...]
> 
>     // Initialize the interrupt
>     LoRa.onReceive(loRaOnReceiveInterrupt);
>     LoRa.receive();
> 
>     // Initialize the reading task
>     xTaskCreate(
>         loRaMeshProcessTask,    // Function that should be called
>         "loRaMeshProcessTask",  // Name of the task (for debugging)
>         2000,                   // Stack size (bytes)
>         NULL,                   // Parameter to pass
>         3,                      // Task priority
>         NULL                    // Task handle
>     );
> 
>     [...]
> 
> }
> 
> void loRaOnReceiveInterrupt(int packetLength) {
> 
>     // Just store the length of the received packet
>     _loRaPacketLength = packetLength;
> 
> }
> 
> void loRaMeshProcessTask(void * parameter) {
> 
>     while(true) {
> 
>         if (_loRaPacketLength > 0) {
> 
>             String packet = LoRa.readString();
> 
>             Serial.println("RX: " + packet);
> 
>             // Reset the packet length
>             _loRaPacketLength = 0;
>         }
> 
>         vTaskDelay(100 / portTICK_RATE_MS);
>     }
> }
> ```
> 
> And the result is (with 80 to 83 the number of the packet sent by another loRa device):
> 
> ```
> RX: 80 {"state":{"time":419607,"type":"ground module","battery":100,"ram":0}}
> RX: 81 {"state":{"time":424703,"type":"ground module","battery":100,"ram":0}}
> RX: 82 {"state":{"time":429943,"type":"ground module","battery":100,"ram":0}}
> RX: 83 {"state":{"time":435120,"type":"ground module","battery":100,"ram":0}}
> Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
> Core 1 register dump:
> PC      : 0x4008c084  PS      : 0x00060534  A0      : 0x8008b23b  A1      : 0x3ffbe660        
> A2      : 0x3ffb8480  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001        
> A6      : 0x00060523  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074        
[... 77 lines cut ...]
	LoRa.setCodingRate4(6);
	LoRa.setTxPower(20);
	LoRa.setOCP(140);
	LoRa.setSyncWord(0xF3);
	LoRa.onReceive(onReceive);

// starting of the Task (core 0 "loop")
    for(;;)
    {
     // code to run in core 0 related to the hardware configured in the start (loop core0Loop "setup")
		if (receivedLoRaP)
		{
			LoRa.sleep();
			receivedLoRaP = false;
			processLoRaPacket();
			LoRa.receive();
		}
		vTaskDelay(100);
       }
}

I am not showing the details of the function processLoRaPacket(), because it is customized for your application.

Another important factor to take into account, if you add more tasks, is to manage the priority of the task: do not use high numbers (the higher the number, the higher the priority).
If adapting this code to yours, still does not work, it would help to troubleshot seeing the full code and hardware design.

morgan-wild, 2020-08-17:
Thank you so much @Jose2059 for your detailed response. After reading it, I've spend some hours to go deeper, working arround your approach.

@GIPdA GIPdA, thank you too, "merci beaucoup même", I kept your quote in mind during my tests: 
I suspect there's a function somewhere that sometimes waits too long.

The last night, I've found data I asked to send when I tried to read, it is another thing to know.

So! I've have something which seems to work, even in bad conditions and making send and receive using interrupts without any crash.

In some words:

Declare a volatile flag, initialize lora, create a xTask for reading and configure the interrupt to set the flag to true, only that. In the xTask, make a loop with a read only if the flag is true then set it to false. It is the solution given by @Jose2059 and it works for pure receiving.

But if you want to send too, be careful and take in consideration the flag! If you try to send something when it is true, you will make something bad in a buffer and when the task will try to read, the Guru crash may appears.

In this code, I tried to reproduce real conditions, enabling Wifi, sending and a periodic long process. There is no crash.

I just noticed after some minutes (over 10), the receive stops to work. But it is another problem I will try to resolve.

```
#include <SPI.h>
#include <Wifi.h>
#include <LoRa.h>

#define PIN_SCK 18
#define PIN_MISO 19
#define PIN_MOSI 23
#define PIN_SS 5
#define PIN_RST 26
#define PIN_DI0 36
#define FREQUENCY 433E6

volatile bool _receivedLoRaP{false};

unsigned long _sendingPacketNumber = 0;
unsigned long _sendingTimer = 0;
unsigned long _processTimer = 0;

void setup() {

    delay(1000);

    // -------------------------------------------------------------------
    // Initialize Serial, Wire (I2C) and SPI

    Serial.begin(115200);

    // Make it heavier, turn on Wifi
    WiFi.softAP("WIFI_HOTSPOT_SSID", "WIFI_HOTSPOT_PASS");

    SPI.begin(PIN_SCK, PIN_MISO, PIN_MOSI);

    LoRa.setPins(PIN_SS, PIN_RST, PIN_DI0);

    if(!LoRa.begin(FREQUENCY)) {
        Serial.println("Failed to start LoRa");
        while (true);
    }

    Serial.println("LoRa started");

    xTaskCreatePinnedToCore(
        core0Task,              // Function that should be called
        "core0Task",            // Name of the task (for debugging)
        5000,                   // Stack size (bytes)
        NULL,                   // Parameter to pass
        3,                      // Task priority                                    <--- increased to 3
        NULL,                   // Task handle
        1                       // Core you want to run the task on (0 or 1)        <-- seems work better on core 1
    );
[... 70 lines cut ...]

        vTaskDelay(10);
    }
}

```

The trace looks to this, with other module sending packet each 5 seconds:

```
RX: 1 77777777 167 704 0 0 {"state":{"time":3639083,"type":"ground module","battery":25,"ram":0}}
Sending packet 132
Emulating a long process........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................      
Emulating a long process........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................      
RX: 1 77777777 168 704 0 0 {"state":{"time":3639083,"type":"ground module","battery":25,"ram":0}}
Sending packet 133
Emulating a long process............................................................................................................................................................................................................................................................................................................................ [... cut ...] ..................................................................................................................................................................................................................................................................................................................................................      
RX: 1 77777777 170 704 0 0 {"state":{"time":3639083,"type":"ground module","battery":25,"ram":0}}
Sending packet 135
Emulating a long process..........
```

JRAlvarez2017, 2020-08-17:
You are very welcome morgan-wild. 
You quoted a why? to put the line of code LoRa.sleep();
.
.
if (_receivedLoRaP) {

            // **LoRa.sleep(); //< why?** ----------------------------------------

            Serial.print("RX: ");
            Serial.println(LoRa.readString());

            _receivedLoRaP = false;
            // LoRa.receive();
        }
.
.
There are two reasons why I put the LoRa board to sleep while processing the packet. 
One, is because my applications have a lot of packets traffic, and there are times (it is random, but eventually happens) that another packet B arrives while the code is processing the first packet A. This will trigger the interrupt, stopping the packet processing, and most likely will corrupt the data or will crash the ESP32. 
The second is because the LoRa.sleep() is the only way to clear the LoRa chip buffer. If not, packets will write one after the other in the 255 bytes buffer of the LoRa device. Clearing the buffer between packets will ensure the library does not load a fraction of the packet.

I did not added the part of the code to send, because I though you were focused on troubleshooting the crashes.
Here you have an example of what I use. The packet is in a struct (mccOperVars), I assign the packet Ids, the packet type, and write the struct to the stream with an custom function located in a auxiliary file. You can use any stream write function of your choice as in the library examples. This is just to give you an idea.
Remember, when your code executes LoRa.beginPacket(), the radio (RFM95) is not receiving anymore (it has been switched to send mode), so no worries about the received packet flag.
If you want to go deeply into the details of how the LoRa RFM95 works, you can read its datasheet (https://cdn.sparkfun.com/assets/a/e/7/e/b/RFM95_96_97_98W.pdf).

void sendPacket(unsigned int packet_type)
{
	// wait until the radio is ready to send a packet
	while (LoRa.beginPacket() == 0)
	{
		delay(200);
	}
	mccOperVars.DeviceId = THISBOARD;
	mccOperVars.DestinationId = OFFICE;
	mccOperVars.PacketType = packet_type;
	writeLoRaStruct(mccOperVars);
	LoRa.endPacket(); // true = async / non-blocking mode
	LoRa.receive();
}

// code in auxiliary.h file
template <class T>
inline void writeLoRaStruct(T &var)
{
	LoRa.write((byte *)&var, sizeof(var));
}

GIPdA, 2020-08-18:
Glad you made it working. But remember that you're working with interrupts and tasks that can preempt each other at any moment, and you have a race condition: you could still receive a packet just before sending, between your test ```if (_receivedLoRaP) return; ``` and ```LoRa.beginPacket()```, then receive and send could be mixed and that would be bad ;)

Fortunately, you can use a mutex to solve that. Here some (untested ;) ) code that should do the trick, as well as another way to wait for the receive flag, more "RTOS-friendly".
I assume that with Arduino, the FreeRTOS headers are already included, and static semaphores are enabled.

```
// Global variables
// Mutex to protect for concurrent access to LoRa object
StaticSemaphore_t _loraMutexBuffer;
SemaphoreHandle_t _loraMutex = xSemaphoreCreateMutexStatic(&_loraSendMutexBuffer);

// Flag to notify received packets
StaticSemaphore_t _loraReceiveSemBuffer;
SemaphoreHandle_t _loraReceiveSem = xSemaphoreCreateMutexStatic(&_loraReceiveSemBuffer);

// ## SEND ##

// Send a packet
unsigned const maxWaitTime_ms = 1000;
if (xSemaphoreTake(_loraMutex, (TickType_t)(maxWaitTime_ms/portTICK_PERIOD_MS) ) == pdTRUE) {
    // We were able to obtain the semaphore and can now access the shared resource.

    // Send packet
    ...

    xSemaphoreTake(_loraReceiveSem, 0); // Clear the receive flag if it were to be set just before sending (we lost a packet then)

    LoRa.receive();

    // We have finished accessing the shared resource.  Release the semaphore.
    xSemaphoreGive(_loraMutex);
}

// ## RECEIVE ##

void loRaOnReceiveInterrupt(int packetLength) {
    BaseType_t higherPriorityTaskWoken = pdFALSE;
    xSemaphoreGiveFromISR(_loraReceiveSem, &higherPriorityTaskWoken);
    portYIELD_FROM_ISR(higherPriorityTaskWoken);
}

// Receive a packet (in your receive task)
if (xSemaphoreTake(_loraReceiveSem, portMAX_DELAY) == pdTRUE) { // Infinite wait (so no need for a delay() afterwards), but you can set a timeout if needed
    // Got a packet, get mutex
    unsigned const maxWaitTime_ms = 1000;
    if (xSemaphoreTake(_loraMutex, (TickType_t)(maxWaitTime_ms/portTICK_PERIOD_MS) ) == pdTRUE) {
        // Read packet
        ....

        xSemaphoreGive(_loraMutex);
    } else {
        // Couldn't get the mutex!
        Serial.println("Failed to read packet (mutex timeout)");
    }
}
```

https://www.freertos.org/xSemaphoreCreateBinaryStatic.html
https://www.freertos.org/xSemaphoreCreateMutexStatic.html

morgan-wild, 2020-08-18:
Thank you very much!
I will try it as soon as possible!

closed 2021-01-13 by IoTThinks