sandeepmistry/arduino-LoRa #381: Problem with Ra02 restart

PrzemyslawMotyl, 2020-07-02:
Hi,

I,m using LoRa Ra02 with ESP8266.
There is such issue for the LoRa receiver.
If there was a problem on SPI then LoRa stops receiving messages.

It can reproduced:

1. Connection is established:
  - LoRa "sender" sends messages every 2 seconds.
  - LoRa "receiver" receives that messages

2. Disconnect SPI SCK for one second

After that LoRa won't receive any message.
Is it possible to restart it somehow?

Now I'm restarting whole ESP if no data are received after fixed time as a workaround.

IoTThinks, 2020-07-02:
Why you disconnected SCK pin?

Establish SPI again?
```
SPI.end();
SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_SS);
```

PrzemyslawMotyl, 2020-07-02:
Disconnection of SCK pin is used here only to show the problem that we can't reset LoRa in case of problems on SPI.
SPI reset didn't helped. After that LoRa sometimes receives messages sometimes not.

IoTThinks, 2020-07-02:
I never have any problem with SPI.
```
LoRa.end()
LoRa.begin()
```

PrzemyslawMotyl, 2020-07-02:
This also didn't helped.
After that 50%-90% packages are lost.
I know this is seldom case. In my case I've just observed that LoRa can't receive packages after several hours.
It seems LoRa.parsePacket() returns 0.
I tired to reproduce it in a simpliest way and I did it after SCK disconnection.
It would be very good to know how to restart LoRa without restarting the whole ESP.

PrzemyslawMotyl, 2020-07-03:
I had to execute the same procedure as in setup + "LoRa.end()" at the beginning.
Thank you for great support.

      LoRa.end();
      LoRa.setPins(ss, rst, dio0);

      if (!LoRa.begin(433E6)) {
        Serial.println("Resting LoRa failed!");
      } else {
        Serial.println("Resting LoRa ok");
        LoRa.setSyncWord(0xF3);
      }

Arismane, 2020-07-04:
Dear PrzemyslawMotyl, I have the same problem. My LoRa receiver stops receiving after some hours although my arduino's other procedures continue to work. Exactly when do you execute the above procedure inside your program that restarts the LoRa interface? Periodically?

PrzemyslawMotyl, 2020-07-05:
Hi,

Here is simplified code.
My lora sends packages every 3 seconds.
If no data is received during 10 seconds then LORA is restarted.

[code.txt](https://github.com/sandeepmistry/arduino-LoRa/files/4874966/code.txt)

Arismane, 2020-07-05:
PrzemyslawMotyl thanks for your reply. I will try your solution, although I am already trying another solution to our problem which seems to work better till now. 
I added loRa.receive() before parse.packet to my code that I hadn't before. Have you tried it?

PrzemyslawMotyl, 2020-07-05:
I will try your solution. Thx

closed 2020-11-12 by IoTThinks

Wilfred-Fortuin, 2023-10-31:
hi need help my gateway shows an unknown receiver. im using a lora ra-02 and a d1