sandeepmistry/arduino-LoRa #382: Implicit vs Explicit header mode

beowulff, 2020-07-02:
I am trying to polish some (mostly working) code.
I decided to try using "Implicit Header" mode, to save a few mS per transmission. All of my transmit and receive parameters are fixed, so I had no reason to doubt it wouldn't work. 
But - it didn't.
I changed LoRa.beginPacket() to LoRa.beginPacket(TRUE) and LoRa.receive() to LoRa.receive(TRUE)...
And, everything stopped working. I could occasionally get a packet through, but it was rare. I'm testing in my office, so the distances were only a few feet. Changing both calls back to () fixed the issue immediately.

So, what did I do wrong?

dhorner, 2020-07-02:
The arguments for those calls are NOT boolean.  They expect an int that is the actual packet size. So instead of true or false they want your actual packet size and packet size of 0 (the default) = explicit header mode.

beowulff, 2020-07-02:
Ah...
Thank you.
Well, that's not going to work, because my packets vary in size.
OK, back to Explicit header mode.

closed 2020-11-12 by IoTThinks

matburnham, 2021-03-25:
> The arguments for those calls are NOT boolean. They expect an int that is the actual packet size. So instead of true or false they want your actual packet size and packet size of 0 (the default) = explicit header mode.

Why?

While it is indeed an int, the code doesn't seem to actually use the value in the argument other than for a trueness check:

[LoRa.cpp#L303-L321](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L303-L321):
```cpp
int LoRaClass::beginPacket(int implicitHeader)
{
  if (isTransmitting()) {
    return 0;
  }

  // put in standby mode
  idle();

  if (implicitHeader) {
    implicitHeaderMode();
  } else {
    explicitHeaderMode();
  }

  // reset FIFO address and paload length
  writeRegister(REG_FIFO_ADDR_PTR, 0);
  writeRegister(REG_PAYLOAD_LENGTH, 0);

  return 1;
}
```

The actual packet length seems to be derived from `LoRaClass::write(const uint8_t *buffer, size_t size)`

IoTThinks, 2021-03-25:
I use implicit mode once and it was working for me.
Pardon me if my memory fails me.

In implicit mode, you NEED to know the size of the packet. The chip won't tell you the packet size as the header is missing in this mode.

The correct usage is 
- beginPacket(true)
- LoRa.parsePacket(KEY-IN-THE-PACKET-SIZE) or LoRa.receive(KEY-IN-THE-PACKET-SIZE)

matburnham, 2021-03-25:
Ah, okay, I'd read the original comment as suggesting `LoRa.beginPacket` as also needing the length. Of course, `parsePacket` and `receive` will need to know the packet length.