sandeepmistry/arduino-LoRa #383: No packets received using example

MasterPuffin, 2020-07-03:
Hello,
I'm using two Arduino Nano clones with Dragino Lora Bee Modules (https://wiki.dragino.com/index.php?title=Lora_BEE).
The wiring is [this](https://playground.boxtec.ch/lib/exe/detail.php/wireless/lora_bee_pinout.png?id=wireless:lorabee) which is equal to the one provided in this repo.

I'm running the example "Sender" and "Receiver" sketches without any modifications except changing the frequency to `868e6`.
The sender is reporting that it's sending packages, however the receiver is not picking up any. However the receiver is picking up empty packages with a RSSI around -150 from time to time, which I guess are signals from other nodes.

I also tried overwriting the pins manually, setting SPI Frequency and TX power as well as using an external power supply for the Lora Modules all without any effect.

IoTThinks, 2020-07-03:
Try to use ReceiveCallback Example. It's continuous RX.
For Receiver with parsePacket, it's normal to have empty packet. It's Single RX. When no packet is received, the packet size is 0.

MasterPuffin, 2020-07-03:
Thanks for your reply, we are getting somewhere!
First of all an addition to my last post: The modules are right next to each other so range is not the problem.

@IoTThinks I tried the ReceiveCallback example. Here is the log of the sender
```
13:32:00.359 -> LoRa Sender
13:32:00.359 -> Sending packet: 0
13:32:05.395 -> Sending packet: 1
13:32:10.426 -> Sending packet: 2
13:32:15.430 -> Sending packet: 3
13:32:20.461 -> Sending packet: 4
13:32:25.490 -> Sending packet: 5
13:32:30.542 -> Sending packet: 6
13:32:35.541 -> Sending packet: 7
13:32:40.588 -> Sending packet: 8
13:32:45.615 -> Sending packet: 9
```
and here is the log from the receiver (sorry for not posting it as text, however GitHub apparently doesn't allow the pasting of these characters)

[image]

So there is clearly data transmitted, however right now it's unreadable garbage.

MasterPuffin, 2020-07-03:
To add to my previous comment:
I changed the frequency slightly to limit confounding factors

Here are the two logs:

```
13:49:27.604 -> LoRa Sender
13:49:27.604 -> Sending packet: 0
13:49:32.629 -> Sending packet: 1
13:49:37.645 -> Sending packet: 2
13:49:42.642 -> Sending packet: 3
13:49:47.671 -> Sending packet: 4
13:49:52.657 -> Sending packet: 5
```

```
13:49:28.222 -> LoRa Receiver Callback
13:49:32.669 -> Received packet 'ardude 1ardude 3ardud' with RSSI -30
13:49:32.703 -> Received packet 'e 4ardud⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮' with RSSI -30
13:49:32.842 -> Received packet 'e 5ardud⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮' with RSSI -30
13:49:32.980 -> Received packet 'e 6ardud⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮e 7ardude ' with RSSI -30
```` 
I also noticed, that the receiving of any packets stops after a few seconds on the receiver.

IoTThinks, 2020-07-03:
You have interference with nodes of other people.

Can try to change to and odd frequency like 895 Mhz and Sf7.

abtzrn, 2020-07-03:
I am having a similar problem. Besides the LoRa Receiver Callback example not working at all, the LoRaReceiver example outputs this:
```
LoRa Receiver
Packet size: 133 Received packet '' with RSSI 44
Packet size: 133 Received packet '' with RSSI 44
Packet size: 133 Received packet '' with RSSI 44
Packet size: 8 Received packet 'hello 47' with RSSI -82
Packet size: 133 Received packet 'hello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 4' with RSSI -82
Packet size: 133 Received packet 'hello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 476⸮⸮⸮
⸮⸮⸮9⸮⸮⸮;H⸮ȇ⸮t⸮⸮⸮t⸮	6âix ESHp G⸮̀5⸮f,+!⸮⸮[⸮㘈v.Q⸮⸮⸮⸮L⸮ ⸮⸮⸮f⸮j⸮<Q⸮d⸮⸮&⸮⸮⸮ӥC⸮⸮x⸮(⸮⸮P:⸮⸮T=⸮⸮⸮g\i⸮⸮⸮
F⸮憬K⸮⸮⸮_LO=>b⸮⸮{o⸮%⸮v"⸮⸮⸮⸮{`⸮u⸮⸮LFM~⸮4]⸮*?⸮d⸮8sg⸮m!H{⸮Y|⸮]⸮K9VXg_ڟW⸮⸮қ⸮⸮⸮⸮M⸮o6⸮⸮⸮k⸮⸮y6⸮⸮>\.⸮k⸮⸮<⸮⸮y⸮SdKhello 4' with RSSI -82
```
Is this as well an interference with some other node?
What can I do to avoid interference? (I already tried to change my frequency and I believe the spreading factory is 7 by default)

IoTThinks, 2020-07-04:
You send 1,2,3...and in your receiver I see “hello...”
Are they your messages?

For callback you need to connect DIO0 properly.
It is sure working.

And be sure to connect your antennas.

abtzrn, 2020-07-04:
Ahh thanks! I forgot to connect DIO0 in my setup.
Antennas are connected (Im using the AI-Thinker Ra-01 LoRa module which came with a spring antenna)
Yes these are my messages, Im sending "hello 1", "hello 2", "hello 3", ...

I believe I fixed it now. I changed the the if-condition to `if (packetSize && LoRa.packetRssi() < 0)`. With this condition I made sure to not print messages from the sender with an RSSI of 44, who is constantly sending (I have no clue what or who this is). Still, there sometimes are weird messages like the ones above (but much less).
Thank you!

IoTThinks, 2020-07-04:
Dio0 is only needed when you use onReceive and onTxDone callback.  This is Continous RX mode.

If you use parsePacket, Dio0 is not required. But you need to ignore those with packetSize=0.
This is Single RX mode.

MasterPuffin, 2020-07-08:
So what I did now is to set the radio to a very odd frequency and use a Sync Word. This mostly works now, however from time to time the receiver is going crazy. I also added SNR and PFE (Packet Frequency Error) for debugging purposes.

```
19:00:48.132 -> LoRa Sender
19:00:48.166 -> Starting LoRa success!
19:00:48.166 -> Sending packet: 0
19:00:53.170 -> Sending packet: 1
19:00:58.186 -> Sending packet: 2
19:01:03.190 -> Sending packet: 3
19:01:08.188 -> Sending packet: 4
19:01:13.194 -> Sending packet: 5
19:01:18.196 -> Sending packet: 6
19:01:23.231 -> Sending packet: 7
19:01:28.231 -> Sending packet: 8
19:01:33.222 -> Sending packet: 9
19:01:38.245 -> Sending packet: 10
19:01:43.277 -> Sending packet: 11
19:01:48.339 -> Sending packet: 12
19:01:53.390 -> Sending packet: 13
```

```
19:00:47.632 -> LoRa Receiver Callback
19:00:47.632 -> Starting LoRa success!
19:00:48.211 -> Received packet 'ardude 0' with RSSI -23' and SNR 9.50' and PFE 3279
19:00:53.230 -> Received packet 'ardude 1' with RSSI -23' and SNR 9.50' and PFE 3263
19:00:58.220 -> Received packet 'ardude 2' with RSSI -24' and SNR 9.50' and PFE 3246
19:01:03.225 -> Received packet 'ardude 3' with RSSI -23' and SNR 9.50' and PFE -65473
19:01:03.295 -> Received packet 'ardude 3' with RSSI -23' and SNR 9.50' and PFE 3246
19:01:08.258 -> Received packet 'ardude 4' with RSSI -24' and SNR 9.75' and PFE 3246
19:01:13.264 -> Received packet 'ardude 5' with RSSI -23' and SNR 9.50' and PFE 3246
19:01:13.332 -> Received packet '⸮ArQ' with RSSI -23' and SNR 9.50' and PFE 3246
19:01:13.367 -> z⸮Q⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮' with RSSI -23' and SNR 9.50' and PFE -65469
19:01:13.538 -> Received packet 'm⸮⸮⸮⸮⸮r' with RSSI -23' and SNR 9.50' and PFE 3246
19:01:13.608 -> Received packet '⸮⸮C⸮a' with RSSI -23' and SNR 9.50' and PFE -65486
19:01:13.677 -> Received packet '⸮_&b⸮
19:01:13.710 -> ' with RSSI -23' and SNR 9.50' and PFE 3222
19:01:13.779 -> Received packet '⸮
19:01:13.848 -> Received packet 'g⸮m⸮⸮⸮' with RSSI -23' and SNR 9.50' and PFE 3242
19:01:13.918 -> Received packet '⸮⸮⸮⸮⸮<' with RSSI -23' and SNR 9.50' and PFE 3246
19:01:13.987 -> Received packet '⸮M_H-Pm'' with RSSI -23' and SNR 9.50' and PFE 3246
19:01:14.054 -> Received packet 'ĉ⸮i' with RSSI -23' and SNR 9.50' and PFE -65465
19:01:14.158 -> Received packet 'CM⸮4⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮' with RSSI -23' and SNR 9.50' and PFE 3246
19:01:14.433 -> Received packet 'ardude 5' with RSSI -23' and SNR 9.50' and PFE 3246
19:01:18.270 -> Received packet 'ardude 6' with RSSI -23' and SNR 9.50' and PFE -65488
19:01:18.337 -> Received packet '	
19:01:18.337 -> z⸮Qm⸮⸮' with RSSI -23' and SNR 9.50' and PFE 3221
19:01:18.407 -> Received packet 'ardude 6' with RSSI -23' and SNR 9.50' and PFE 3229
19:01:23.269 -> Received packet 'ardude 7' with RSSI -24' and SNR 9.50' and PFE 3221
19:01:23.338 -> Received packet 'ardude 7' with RSSI -24' and SNR 9.50' and PFE 3229
19:01:28.264 -> Received packet 'ardude 8' with RSSI -24' and SNR 9.25' and PFE -65495
19:01:28.333 -> Received packet 'C⸮a⸮_&' with RSSI -24' and SNR 9.25' and PFE 3229
19:01:28.402 -> Received packet 'b⸮
19:01:28.437 -> ⸮
19:01:28.576 -> Received packet '⸮⸮3g⸮m' with RSSI -24' and SNR 9.25' and PFE 3229
19:01:33.273 -> Received packet '⸮⸮⸮⸮⸮⸮' with RSSI -34' and SNR 9.00' and PFE 3252
19:01:33.343 -> Received packet '⸮⸮<⸮M_⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮' with RSSI -34' and SNR 9.00' and PFE 3240
19:01:33.548 -> Received packet 'H-Pm'ĉ' with RSSI -34' and SNR 9.00' and PFE 3235
19:01:33.617 -> Received packet '⸮iCM' with RSSI -34' and SNR 9.00' and PFE 3238
19:01:33.685 -> Received packet '⸮4⸮0m' with RSSI -34' and SNR 9.00' and PFE 3254
19:01:33.755 -> Received packet '~9;⸮w;⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮' with RSSI -34' and SNR 9.00' and PFE 3229
19:01:34.064 -> Received packet '⸮%⸮n⸮}⸮M' with RSSI -34' and SNR 9.00' and PFE -65490
19:01:34.133 -> Received packet '5⸮⸮⸮⸮⸮l⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮' with RSSI -34' and SNR 9.00' and PFE 3253
19:01:34.439 -> Received packet '⸮⸮삂⸮⸮m⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮' with RSSI -34' and SNR 9.00' and PFE -65489
19:01:34.716 -> Received packet '⸮ ⸮P=⸮⸮' with RSSI -34' and SNR 9.00' and PFE -65470
19:01:34.819 -> Received packet 'ardude 9' with RSSI -34' and SNR 9.00' and PFE 3229
19:01:38.282 -> Received packet 'ardude 10' with RSSI -23' and SNR 9.25' and PFE -65488
19:01:38.350 -> Received packet '⸮3g⸮m⸮⸮' with RSSI -23' and SNR 9.25' and PFE 3229
[... cut ...]
19:01:43.345 -> Received packet 'ardude 11' with RSSI -23' and SNR 9.25' and PFE 3229
19:01:43.415 -> Received packet '⸮⸮⸮⸮⸮⸮' with RSSI -23' and SNR 9.25' and PFE 3229
19:01:48.381 -> Received packet '<⸮M_H-Pm' with RSSI -34' and SNR 9.25' and PFE 3231
19:01:48.451 -> Received packet 'ardude 12' with RSSI -34' and SNR 9.25' and PFE 3229
19:01:53.422 -> Received packet 'ardude 13' with RSSI -23' and SNR 9.25' and PFE 3250
19:01:53.491 -> Received packet 'ardude 13' with RSSI -23' and SNR 9.25' and PFE 3229
```

MasterPuffin, 2020-07-31:
I got it working now. I wired the Lora Modules to an external 3,3V power supply. This includes also the receiver as this could also be the reason packets are not displayed (see #385 ).

Down below is also the modified code for continuous receiving:

``` 
#include <SPI.h>
#include <LoRa.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver Callback");
  // LoRa.setSpreadingFactor(7);
  
  if (!LoRa.begin(897.85E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  LoRa.setSyncWord(0xF3);
  Serial.println("Starting LoRa success!");

  // register the receive callback
  LoRa.onReceive(onReceive);

  // put the radio into receive mode
  LoRa.receive();
}

void loop() {
  // do nothing
}

void onReceive(int packetSize) {  
  if (packetSize == 0) return;
  //if (abs(LoRa.packetFrequencyError()) > 5000) return;
  // received a packet
  Serial.print("Received packet ");

  // read packet
  //for (int i = 0; i < packetSize; i++) {
  //  Serial.print((char)LoRa.read());
  //}

  String incoming = "";
 
  for (int i = 0; i < packetSize; i++) {
    //incoming += (char)LoRa.read();
    Serial.print((char)LoRa.read());
  }
 
  //Serial.print("Message: '" + incoming);

  // print RSSI of packet
  Serial.print("' with RSSI ");
  Serial.print(LoRa.packetRssi());
  Serial.print(" and SNR ");
  Serial.print(LoRa.packetSnr());
  Serial.print(" and PFE ");
  Serial.println(LoRa.packetFrequencyError());
}
```

closed 2020-07-31 by MasterPuffin

IoTThinks, 2020-08-01:
Arduino Nano is known to provide insufficient 3.3v power to LoRa module.
Glad you solved the issue.