sandeepmistry/arduino-LoRa #390: MKR1310 no communication in 433 MHz band while 869 and 915 MHz work

engeen-nl, 2020-08-16:
Thank you first of all for your hard work. The datasheet is dazzling for the SX1276 so amazing work!

We're building an industrial remote using the MKR1310. In Europe the 869 MHz band is limited to 250 kHz which gives us only the possibility of 5 Hz messages while maintaining a duty cycle of 10%. (I'm measuring the time Lora.endPacket takes, around 20 ms).

So I was hoping to be able to use the 433 MHz band where 500 kHz is available. I tested both 869 and 915 MHz and I have communication. But when I toggle to 433 MHz on both sender and receiver, the receiver does not receive any packets.

I thought it had to do with the `LowFrequencyModeOn` bit of the `RegOpMode` register, so I updated your library to set this bit high whenever RegOpMode is written. But no success.

I noticed earlier you were able to measure a 433 MHz signal using  your SRD (Airspy) as mentioned here: https://github.com/sandeepmistry/arduino-LoRa/issues/205#issuecomment-613849836

But I don't have that luxury. Do you have any ideas what could be wrong?

IoTThinks, 2020-08-16:
Check the datasheet.of your module.

Some modules only support either low frequency 433/470Mhz or high frequency 868/915/923 Mhz.

engeen-nl, 2020-08-16:
Datasheet of Semtech says the SX1276 has a frequency range of 137 - 1020 MHz. The datasheet of Murata sais an SX1276 is on board but only mentions 868 and 915 to be supported frequencies. But I understand with this library we directly talk with the SX1276 right?

IoTThinks, 2020-08-17:
By right, sx1276 from Semtech will support 137 - 1020 MHz.
However, some LoRa modules which use SX1276 as internal chip only do external wiring for either low or high frequency.
May be to save the cost.

wero1414, 2020-09-07:
The chip sx1276 is able to manage those frequencies and this library support both frequencies Low and High, but as far as the datasheet says this module only supports only high frequencies i mean 868 or 915 Mhz that is cause the way that murata build the module they only connect the RF output for high frequency and also do all the antenna matching to those frequencies.

closed 2020-09-07 by wero1414