sandeepmistry/arduino-LoRa #391: Is it available

Rita-GE, 2020-08-17:
Are you receiving `Starting LoRa failed` while using the demo code?

PLEASE see the [FAQ #1](https://github.com/sandeepmistry/arduino-LoRa#faq) about using [setPins](https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins) **BEFORE** submitting an issue.

IoTThinks, 2020-08-17:
Newer Chinese LoRa modules have DIO2.
https://resource.heltec.cn/download/Wireless_Stick_Lite/Wireless_Stick_Lite.pdf

But what is FHSS used for?

Rita-GE, 2020-08-18:
FHSS (frequency hopping spread spectrum) is a method of transmitting radio signals by shifting carriers across numerous channels with a pseudorandom sequence which is already known to the sender and receiver.
For more detail, please see [here](https://www.techopedia.com/definition/25709/frequency-hopping-spread-spectrum-fhss).
The module I used is [Ra-02 AI-thinker](http://wiki.ai-thinker.com/_media/lora/docs/c048ps01a1_ra-02_product_specification_v1.1.pdf).
As SX1278's [datasheet](https://cdn-shop.adafruit.com/product-files/3179/sx1276_77_78_79.pdf) shows, it has that function on page 32.
[image]
I'm so curious why don't you know that.

wero1414, 2020-09-07:
This library does not support FHSS now we're open to pull requests to be implemented, thanks!

closed 2020-09-07 by wero1414