sandeepmistry/arduino-LoRa #392: Modifying spread factor reduces RSSI for RFM95W

KnightOfNih, 2020-08-18:
Hello All,

I'm using the RFM95W LoRa transceiver and am trying to send/receive signals from within a multi-story concrete building.  So far I can only receive signals +/- 1 floor.

[According to this study](https://smartmakers.io/en/lorawan-range-part-2-range-and-coverage-of-lorawan-in-practice/), LoRa was able to transmit 1.5km and through 14 buildings.  That's some real range!

Why am I having trouble getting a signal through 2 floors of concrete?

Code is below for the sender and receiver.  I've tried various settings, but what's odd is that modifying the spread factor to 9 or 12 actually reduces the signal quality significantly.  

This was "scientifically" tested by holding the sender & receiver about 2 feet apart in my hands or on a desk.  Antennas attached [look like these](https://www.amazon.com/DIYmall-Connector-Antenna-915MHz-Lora32u4/dp/B084KVYBH5/ref=sr_1_20?dchild=1&keywords=lora+antenna&qid=1597760719&sr=8-20).

**Default Settings**

`
  // LoRa.setPreambleLength(8);
  // LoRa.setCodingRate4(5);
  // LoRa.enableCrc();
  // LoRa.setSpreadingFactor(12);
  // LoRa.setSignalBandwidth(125E3);
  // LoRa.setTxPower(20);
  LoRa.setSyncWord(SYNC_WORD);
  Received packet 'LoRa: 0' with RSSI -48
Received packet 'LoRa: 1' with RSSI -22
Received packet 'LoRa: 2' with RSSI -21
Received packet 'LoRa: 3' with RSSI -21
Received packet 'LoRa: 4' with RSSI -21
Received packet 'LoRa: 5' with RSSI -22
Received packet 'LoRa: 6' with RSSI -22
Received packet 'LoRa: 7' with RSSI -23
Received packet 'LoRa: 8' with RSSI -22
Received packet 'LoRa: 9' with RSSI -22
Received packet 'LoRa: 10' with RSSI -28
Received packet 'LoRa: 11' with RSSI -52
`

**Spread Factor 12
Bandwidth: 125hz
Tx Power: 20**

` 
// code settings
 LoRa.setPreambleLength(8);
  LoRa.setCodingRate4(5);
  LoRa.enableCrc();
  LoRa.setSpreadingFactor(12);
  LoRa.setSignalBandwidth(125E3);
  LoRa.setTxPower(20);
  LoRa.setSyncWord(SYNC_WORD);

// output
  Received packet 'LoRa: 0' with RSSI -50
Received packet 'LoRa: 1' with RSSI -24
Received packet 'LoRa: 2' with RSSI -17
Received packet 'LoRa: 3' with RSSI -16
Received packet 'LoRa: 4' with RSSI -41
Received packet 'LoRa: 5' with RSSI -52
Received packet 'LoRa: 6' with RSSI -67
Received packet 'LoRa: 7' with RSSI -62
Received packet 'LoRa: 8' with RSSI -55
Received packet 'LoRa: 9' with RSSI -54
Received packet 'LoRa: 10' with RSSI -44
Received packet 'LoRa: 11' with RSSI -63
Received packet 'LoRa: 12' with RSSI -56
Received packet 'LoRa: 13' with RSSI -49
Received packet 'LoRa: 14' with RSSI -48
Received packet 'LoRa: 15' with RSSI -51
Received packet 'LoRa: 0' with RSSI -52
`

**Spread Factor 9**

`
  LoRa.setPreambleLength(8);
  LoRa.setCodingRate4(5);
  // LoRa.enableCrc();
  LoRa.setSpreadingFactor(9);
  // LoRa.setSignalBandwidth(125E3);
  LoRa.setTxPower(20);
  LoRa.setSyncWord(SYNC_WORD);

Received packet 'LoRa: 0' with RSSI -49
Received packet 'LoRa: 1' with RSSI -38
Received packet 'LoRa: 2' with RSSI -20
Received packet 'LoRa: 3' with RSSI -21
Received packet 'LoRa: 4' with RSSI -21
Received packet 'LoRa: 5' with RSSI -51
Received packet 'LoRa: 6' with RSSI -50
Received packet 'LoRa: 7' with RSSI -50
Received packet 'LoRa: 8' with RSSI -52
Received packet 'LoRa: 9' with RSSI -46
Received packet 'LoRa: 10' with RSSI -44
Received packet 'LoRa: 11' with RSSI -44
`

**Arduino Code**

`
#include <Arduino.h>
#include <LoRa.h>

#define SCK  14
#define MISO 12
#define MOSI 13
#define SS  16
#define DIO0 26
#define RST -1
#define SPISPEED 8000000
#define BAUD_RATE 115200
#define SYNC_WORD 0x34

#define IS_SENDING true

int counter = 0;
SPIClass * hspi = NULL;

void setup() {

  hspi = new SPIClass(HSPI);
  hspi->begin(SCK, MISO, MOSI, SS);

  //initialize Serial Monitor
  Serial.begin(BAUD_RATE);
  while (!Serial);

  //setup LoRa transceiver module
  LoRa.setSPI(*hspi);
  LoRa.setSPIFrequency(SPISPEED);
  LoRa.setPins(SS, RST, DIO0);
  LoRa.setPreambleLength(8);
  LoRa.setCodingRate4(5);
  LoRa.enableCrc();
  LoRa.setSpreadingFactor(12);
  LoRa.setSignalBandwidth(125E3);
  LoRa.setTxPower(20);
  LoRa.setSyncWord(SYNC_WORD);

  
  //replace the LoRa.begin(---E-) argument with your location's frequency 
  //433E6 for Asia
  //866E6 for Europe
  //915E6 for North America
  while (!LoRa.begin(915E6)) {
    Serial.println(".");
    delay(500);
  }

  Serial.println("LoRa Initializing OK!");
}

void loop(){
  if(IS_SENDING){

    String msg = "LoRa: " + String(counter);

    Serial.print("Sending packet: ");
    Serial.println(msg);

    LoRa.beginPacket();
    LoRa.print(msg);
    LoRa.endPacket(true);
    Serial.println("Sent.");
    counter++;

    Serial.println("Waiting 10 sec");
    delay(1000 * 10);

    
  } else {
    // try to parse packet
    int packetSize = LoRa.parsePacket();
    if (packetSize) {
      // received a packet
      Serial.print("Received packet '");

      // read packet
      while (LoRa.available()) {
        String LoRaData = LoRa.readString();
        Serial.print(LoRaData); 
      }

      // print RSSI of packet
      Serial.print("' with RSSI ");
      Serial.println(LoRa.packetRssi());
    }
  }
}

`

Are you receiving `Starting LoRa failed` while using the demo code?

PLEASE see the [FAQ #1](https://github.com/sandeepmistry/arduino-LoRa#faq) about using [setPins](https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins) **BEFORE** submitting an issue.

KnightOfNih, 2020-08-18:
Looks like the RSSI is random, or is following a pattern I don't understand.  Set spread factor to 9 and go the following:

Received packet 'LoRa: 0' with RSSI -51
Received packet 'LoRa: 1' with RSSI -49
Received packet 'LoRa: 2' with RSSI -48
Received packet 'LoRa: 3' with RSSI -49
Received packet 'LoRa: 4' with RSSI -46
Received packet 'LoRa: 5' with RSSI -20
Received packet 'LoRa: 6' with RSSI -19
Received packet 'LoRa: 7' with RSSI -19
Received packet 'LoRa: 8' with RSSI -18
Received packet 'LoRa: 9' with RSSI -18

Who knows ...

KnightOfNih, 2020-08-18:
Looking into a possible PCBA fabrication error as I shouldn't be getting such random RSSI values.

wero1414, 2020-09-07:
You should be able to get the same RSSI in a controled enviroment i did face like the same "problem" before but it wasnt a problem cuase i did get long distance on the range of Km

tabascojoe, 2020-12-02:
Just noticed this thread.
The study you refer to isn't through buildings but rather around and over buildings.
(The authors don't understand RF transmission.)
In your case you are trying to transmit through floors of a building.
You just aren't going to penetrate more than a floor or two at power levels and frequencies available with these devices.

KnightOfNih, 2020-12-02:
Thanks tabascojoe.  I eventually talked to some RF experts and they told me the same thing.  I was pretty bummed.  Still pretty cool tech, just not for my use case.

closed 2020-12-02 by KnightOfNih