sandeepmistry/arduino-LoRa #393: Sending while in receive callback

markusschweitzer, 2020-08-20:
Hi,
Im using the latest version from git and this example: https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaReceiverCallback/LoRaReceiverCallback.ino

When I try to send a packet in the receive callback, (beginPacket, write, endPacket) the receiver sends the packet, but doesnt receive anymore.

Calling Lora.idle() and LoRa.receive() after the sending helps for a few packets, but eventually it stops receiving.

Any idea how I can make this work?

best regards,
markus

IoTThinks, 2020-08-20:
Did you connect MCU to LoRa DIO0?
By right, in the library, the sending will be skipped if a packet is receiving.

markusschweitzer, 2020-08-20:
Yes, the ISR gets called correctly (DIO0 is connected), I read all the data and send a response packet :/

IoTThinks, 2020-08-20:
By right, you can not send a message during a receive.
This is chip limitation.

Send after the receiving is complete.
And sending outside the ISR callback.

markusschweitzer, 2020-08-20:
Ok, I wasnt aware of that. Is there any way to do this without a periodic call in the loop() function?

IoTThinks, 2020-08-20:
By right, you should not do any processing inside the ISR also.
ISR should set a flag only.
Then you can process sending or receiving in the main loop or in a FreeRTOS task.

```
void onReceive(int packetSize) {
  // received a packet
  Serial.print("Received packet '");

  // read packet
  for (int i = 0; i < packetSize; i++) {
    Serial.print((char)LoRa.read());
  }

  // print RSSI of packet
  Serial.print("' with RSSI ");
  Serial.println(LoRa.packetRssi());
}
```

IoTThinks, 2020-08-21:
100ms is more than enough.

IoTThinks, 2020-11-12:
Seems solved.

closed 2020-11-12 by IoTThinks