sandeepmistry/arduino-LoRa #397: high current sleep issue with RFM95 on ESP8266

steenl, 2020-09-09:
Using library version 0.7.2 I am seeing significant current drawn after code below after which the ESP8266 is put into a deep sleep.  Without the RFM95, I see about 300uA in deep sleep.  With RFM95, there is a 200ms cycle that peaks for about 30ms at 1.1mA from a 0.8mA base.  
```
    delay(1);
    LoRa.sleep();
    Serial.println("Done LoRa sleep");
    LoRa.end();
    delay(100);
```
I have looked at library code and do not see anything obvious wrong relative to the RFM95 spec, but is there a way to verify the RFM95 is in sleep mode, or recommended delay or other common environment issue blocking RFM95 entering sleep?

IoTThinks, 2020-09-10:
Try to set LoRa_RST into INPUT mode after LoRa.end()
And feedback the result.

steenl, 2020-09-10:
Thanks for response.  My issue was a missed BME280 that was not properly put to sleep.  Correcting that, and reducing supply voltage to near-minimum, results in expected range of 40uA for complete design.

closed 2020-09-10 by steenl