sandeepmistry/arduino-LoRa #398: Sender working but receiver not working

Eduardix2020, 2020-09-13:
Hello guys,

I got a RA-02 AI thinker Lora module,

The sender is sending the packets but the receiver is stucked with the message:

Lora receiver
(and nothing more),

I am using the library LoRaSender and LoRaReceiver, I just changed to 433 hz I am in Canada.

I read that is required to use a level shifter (3.3-5.5) bidirectional but people seems to have it worked with no need of this.

Also I have pluged the receiver LoRa to a 3.3V external supply but no results

I have 2 arduino nanos with the old bootloader.

Below the code

Sender
-----------------------
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}

Receiver 
------------------------------------
#include <SPI.h>
#include <LoRa.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}

[image]

IoTThinks, 2020-09-13:
Are you sure your chip is 433Mhz?

Eduardix2020, 2020-09-13:
Positive,

Here the specifications:

Communication distance: 15KM

Sensitivity: down to -148dBm

Programmable bit rates: up to 300kbps

RSSI dynamic range: 127dB

Wireless frequency: 433MHz

Working voltage: 1.8-3.7v

Working temperature: -40-+80 ℃

IoTThinks, 2020-09-13:
Have you put in the antenna?

Eduardix2020, 2020-09-13:
> Have you put in the antenna?

No, actually the module came with no antenna.

I was following this tutorial where the module was working without antenna. There is no way to send the signal if no antenna? I thought there was a chance to get the signal even with no antenna.

I wanted to rule out some issues and if I go to my electronic store I wanted to get all potential components I'll need, what else shout I get it's not the antenna?

IoTThinks, 2020-09-13:
Get the antennas or you may damage the modules.

If too lazy, you can hook in a copper wire.

For 433Mhz, the antenna length is around 17cm.

Eduardix2020, 2020-09-16:
Thanks it worked,

Also I had to connet the Vin of my nano and not the the 3.3V to my 3.3+ external power , also very important is to have a common ground with the nano and the lora, also they have to be plugged to the external supply.

I was about to give up and buy a pro mini, thanks a lot @IoTThinks!

IoTThinks, 2020-09-16:
The 3.3v of Arduino Nano normally doesn't have enough current (mA) for LoRa modules. Sometimes, LoRa modules may hit up to 100mA while transmitting.
You may need to provide external 3.3v to LoRa modules.

https://github.com/sandeepmistry/arduino-LoRa/issues/230

closed 2020-09-26 by Eduardix2020

rocky910, 2023-12-17:
> Thanks it worked,
> 
> Also I had to connect the Vin of my nano and not the the 3.3V to my 3.3+ external power , also very important is to have a common ground with the nano and the lora, also they have to be plugged to the external supply.
> 
> I was about to give up and buy a pro mini, thanks a lot @IoTThinks!

Hiii,can u please give the pin connections or the diagram of your connections,because lora module have the voltage rating till 3.7v while vin voltage is 5v in Uno,so please help me with the pin connections,I'm having the same problem