sandeepmistry/arduino-LoRa #399: Adding LNA Support for sx1278?

IoTThinks, 2020-09-19:
Hi guys,
It seems that this library does not support LNA gain yet?

[image]

I believe LNA gain can help sx1278 to receive packet near -140dBm RSSI and below -21 SNR.
Therefore, it could help to receive packets farther.

This library has hasGain function for this purpose: https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX1278.cpp#L316

Any supporting idea?

IoTThinks, 2020-09-19:
@morganrallen @sandeepmistry Do you have any idea on this?
Thanks a lot. :D

sandeepmistry, 2020-09-29:
@IoTThinks do you have a proposed API(s) in mind? Would you be able to pull request for this?

NOTE: `LnaBoostHf` is currently set to 0x03 in [begin(...)](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L134).

IoTThinks, 2020-09-29:
@sandeepmistry May be, I will create my very first Pull request :D 
So by default, the LnaBoostHf is set to enable LNA for high frequency 868-923Mhz?

[image]

Do you think we need to set LnaBoostLf for low frequency like 433Mhz?
[image]

IoTThinks, 2020-10-03:
I'm able to make it works.
I will make a function setGain(byte gain).
Gain is from 0 to 6.

If the gain is 0, then AGC will be on.

[image]

I realize that LNA in HF 868/915/923 is better than LF 433 Mhz in default gain 0.
Therefore, RSSI for HF is better than for LF 433Mhz.

With AGC off and gain 6, I can now have RSSI -20 with two 433Mhz nodes in 20cm.

Will try to make it works and create a pull request.

IoTThinks, 2020-11-05:
Added pull request at https://github.com/sandeepmistry/arduino-LoRa/pull/408

closed 2020-11-05 by IoTThinks