sandeepmistry/arduino-LoRa #400: endpacket() never returns due to SX127x crashed, possible workaround

Markusenz, 2020-10-05:
There is a known problem with SX127x where it hangs and never reports TX finished.
As reported by several users - 
https://forum.arduino.cc/index.php?topic=574911.0
when that happens, also a normal reset procedure cannot recover the SX127x logic.
That means, the application will be hanging until power toggle. The root cause could be SPI problems due to a bad pcb layout (that's what Semtech seems to claim), but nobody knows for sure.
There is a workaround (reset procedure consisting of several steps)  which can be found here:
https://github.com/Lora-net/LoRaMac-node/blob/develop/src/radio/sx1276/sx1276.c
after the line "// BEGIN WORKAROUND".
 
I am using ATMega328p with SX1278 at 3.3 Volt in a battery powered sensor application. I can send several thousand messages during several weeks runtime, and suddenly problem occurs. It would be great to update this very good library with the workaround reset procedure so that whenever TXdone times out, the Lora logic could be reset successfully and automatically by the library.

IoTThinks, 2020-10-06:
Try to increase the SPI speed to 8Mhz or above?

IoTThinks, 2020-10-06:
BTW, is your chip SX1278 or SX1276?
You mention ra02 433Mhz, so it is sx1278?
[image]

Markusenz, 2020-10-06:
Increase SPI speed? I thought I would try to decrease to make it more robust...
As said I am using Atmega328P with SX1276 (RFM95 Module)

IoTThinks, 2020-11-16:
@Markusenz Have you tried to put in the fix?

bernii, 2020-12-24:
@Markusenz seems I'm running into the same issue - after some time my TTGO Lora32 is unable to send or receive LoRa traffic (even though it thinks it sends the data successfully). Were you able to resolve it?

Markusenz, 2020-12-28:
> @Markusenz seems I'm running into the same issue - after some time my TTGO Lora32 is unable to send or receive LoRa traffic (even though it thinks it sends the data successfully). Were you able to resolve it?

I tried to implement the fix, however I also made additional improvements in my code, and it never happened so far. I am running about 30 nodes since October without any problems. But note that if the issue arises, the library call for sending the msg will hang until the device is externally reset (the TX done interrupt from the RFM96 module  is not firing so the code hangs at that stage within the library code). 
If in your case the device returns as if the msg was sent out, there is most likely another problem in your case.

maverick061, 2021-04-12:
Possible solution is add delayMicroseconds(1); in function singleTransfer befor digitalWrite(_ss, HIGH);

https://forum.pjrc.com/threads/41878-Probable-race-condition-in-Radiohead-library
https://forum.pjrc.com/attachment.php?attachmentid=10948&d=1499109224

IoTThinks, 2021-06-23:
May be related and solved issue: https://github.com/sandeepmistry/arduino-LoRa/issues/363

Portia-Lin, 2026-01-11:
But my RA-02 modules have SX1278 chip, not SX1276. if I use Radiohead library for RFM95 but specify frequency 433mhz, will it cause problems?

Portia-Lin, 2026-01-12:
Radiolib is too heavy a library for the atmega328 microprocessor. I don't have the option to use another microcontroller with more memory.