sandeepmistry/arduino-LoRa #402: Arduino MKR 1010: Starting LoRa failed!

Commifreak, 2020-10-11:
I read the FAQs and used different libs AND setPins - no luck.

I use an Dragino LoRa shield with an MKR 1010. Should no big deal if the wiring is correct, right?

|Dragino PIN | Arduino PIN |
|--|--|
|RST|D5|
|3V3 (PWR LED is on)|VCC|
|GND|GND|
|13|SCK|
|12|MISO|
|11|MOSI|
|10|D4|
|8|D3|
|7|D2|
|6|D1|
|2|D0|

=> https://wiki.dragino.com/index.php?title=File:LoRa_Shield_Pin_Mapping.png
=> https://content.arduino.cc/assets/Pinout-MKRwifi1010_latest.png

I adapted the pinout from Uno to MKR1010.

I use:

```
  LoRa.setPins(4, 5, 0);
  if (!LoRa.begin(915E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
```

and get The failed message. The SPI wirung should be, right? I dont find the error on the wiring. Any ideas? Thanks in advance!

IoTThinks, 2020-10-12:
According to this image, LoRa should connect to D8, 9 and 10.
Why you connect SCK to D13?
https://content.arduino.cc/assets/Pinout-MKRwifi1010_latest.png

Commifreak, 2020-10-12:
SCK is not connected to D13. The PIN labeled 13 on the Dragino is connected to SCK (D9) on the MKR 1010.

So Dragino 11,12,13 is connected to D8-10. Normally they would be connected to UNO's MISO/MOSI/SCK since it is a shield for it.

IoTThinks, 2020-10-12:
You may try to use external 3.3v power for your LoRa.
And may try to set SS pin to OUTPUT and HIGH before LoRa.begin()

Commifreak, 2020-10-12:
Got it! After trying your tips, nothing worked. Then I tested the Pin lines where they come from and whre they go.

The 3 jumpers (SV2, 3 and 4) were set to ICSP! I had to change them in order to get the data ported to 11, 12, 13. Not the failing did not show up!

Weird - this default setup would never work on an UNO :/

I suggest this to their wiki.

Thanks anyway :)

closed 2020-10-12 by Commifreak

IoTThinks, 2020-10-12:
The picture already mentioned about the pin jumpers.

You miss the basic parts for the junpers!

We can only assume you do the basics right.

Commifreak, 2020-10-12:
I was not sure - this shield is for the UNO - I thought it would be working with default settings, since no text stated this would be important. Sorry about that :(

IoTThinks, 2020-10-12:
Nevermind, it works.
Happy coding :-)