sandeepmistry/arduino-LoRa #403: best library for lora arduino

sajadahmadi14, 2020-10-18:
I'm using two Lora transceivers for sending and receiving data
Lora has capability for sending data to 10KM
but I just can have range for about 200 meters 
what points should I check in my code sketch?
Is there any reliable library that gives maximum range?
I'm using Arduino official library

IoTThinks, 2020-10-18:
10km for open space only.

What is your board and environment?

sajadahmadi14, 2020-10-18:
my result for outdoor was about 200 meter in position that two modules can see each other.
does it deeply depend on antenna!?

sajadahmadi14, 2020-10-18:
> 10km for open space only.
> 
> What is your board and environment?

my result for outdoor was about 200 meter in position that two modules can see each other.
does it deeply depend on antenna!?

IoTThinks, 2020-10-19:
Show us your antenna.
Is your MCU Arduino Nano? Some of the modules can not provide up to 100mA current for LoRa. So it may not transfer too far.

sajadahmadi14, 2020-10-19:
> Show us your antenna.
> Is your MCU Arduino Nano? Some of the modules can not provide up to 100mA current for LoRa. So it may not transfer too far.

this is my antenna, and I'm using Arduino uno which can't provide 120mA for Lora in tx mode, so for Lora power supply I use a lithium polymer battery
here is the antenna
[https://www.amazon.in/INVENTO-antenna-Quad-Band-Female-Connector/dp/B08KGZGD55/ref=sr_1_3?dchild=1&keywords=ufl+antenna&qid=1603108979&sr=8-3](url)

IoTThinks, 2020-10-19:
You may want to try another antenna. This antenna looks like a GSM antenna to me.
And use the ipex to SMA cable with the length as short as possible.

sajadahmadi14, 2020-10-19:
Thanks for your suggestion
Is there any special parameter that I should focus on or change in Lora code sketch?

sajadahmadi14, 2020-10-19:
> You may want to try another antenna. This antenna looks like a GSM antenna to me.
> And use the ipex to SMA cable with the length as short as possible.

Thanks for your suggestion
Is there any special parameter that I should focus on or change in Lora code sketch?

IoTThinks, 2020-10-19:
Even with SF7, BW 500Khz can reach more than 200m in open space.
SF12, BW125Khz will be farther.

sajadahmadi14, 2020-10-20:
> Even with SF7, BW 500Khz can reach more than 200m in open space.
> SF12, BW125Khz will be farther.

I can't find information about SF. would you please explain it for me?
actually I'm new to wireless communication, what should I search about?

IoTThinks, 2020-10-20:
Google is your best friend.

sajadahmadi14, 2020-10-20:
> Google is your best friend.
> LoRa + Spreading Factor
I'll go through it
thanks so much for your useful help

closed 2020-12-24 by IoTThinks