sandeepmistry/arduino-LoRa #404: DIO interrupt capability?

maxholgasson, 2020-10-29:
HI,
How can we add DIO interrupt capability to the sketch LoRaReceiver.ino?
I'm using a menu library which calls a function every 100ms and the LoRa reception is not reliable as I suspect a missing interrupt capabilty when receiving a packet.

IoTThinks, 2020-10-30:
You can use receive callback with DIO0.

Example
https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaReceiverCallback/LoRaReceiverCallback.ino

The sketch uses default DIO0 at https://github.com/sandeepmistry/arduino-LoRa#semtech-sx1276777879-wiring

You can change if you want. Note that, you may not be able to change for Arduino boards.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins

maxholgasson, 2020-11-01:
Thanks @IoTThinks  
I'm using LoRa.setPins(10, 9, 2); and connected DIO0 to pin2. My board works perfect with the CAD mode DI0 sketch. So the connection of the RFM95 should be correct.
However I can't get the LoRaReceiverCallback.ino sketch running.
Do you have any idea how to test and find the mistake?

IoTThinks, 2020-11-02:
What example with CAD did you use?
CAD may use both DIO0 and DIO1.

Which MCU board do you use?
Check if GPIO2 is interruptable.

maxholgasson, 2020-11-02:
I'm using this example https://github.com/chocotov1/arduino-LoRa/blob/cadmode/examples/LoRaReceiverCadDemo328pDIO0/LoRaReceiverCadDemo328pDIO0.ino

>Did you solder LoRa DIO0 to GPIO2 of MCU?

Yes, DIO0 is connected to pin2 (arduino pro mini). I'm using a own designed pcb
It's interruptable because with the other example it wake's up from sleeping which wouldn't be possible if it's non-interruptable.

IoTThinks, 2020-11-02:
Are you using this Sandeep library to test?
The CAD Receiver is a fork from Sandeep's.

Normally, you can only intercept either RXDone or CAD Done from DIO0.

maxholgasson, 2020-11-02:
That was a good question. Thanks. I used the fork's lib as I'm very happy with the lowpower LoRaReceiverCadDemo328pDIO0.ino but for another node I would need to keep the device awake and not to sleep in between. Mh ok, but than it's not an issue of this lib(besides it would be awesome to implement this great fork). I'm adding and removing the libs everytime I need either the one or the other (have no other ide how to do it)

IoTThinks, 2020-11-05:
I use preample 8 and still can use receive callback.

Preamples are just signal to indicate the beginning of a LoRa packets.

closed 2020-11-25 by IoTThinks