sandeepmistry/arduino-LoRa #405: parsePacket() returns zero in duplex mode and from both ends its only transmitting the data not receiving in any side.

AjayJassal55, 2020-11-03:
Are you receiving `Starting LoRa failed` while using the demo code?

PLEASE see the [FAQ #1](https://github.com/sandeepmistry/arduino-LoRa#faq) about using [setPins](https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins) **BEFORE** submitting an issue.

IoTThinks, 2020-11-03:
Option 1: If parsePacket() ==0, skip it. There is not always a LoRa packet coming when you use this command.
Option 2: Use callbackReceiver instead for continuous receiving.

AjayJassal55, 2020-11-03:
> Option 1: If parsePacket() ==0, skip it. There is not always a LoRa packet coming when you use this command.
> Option 2: Use callbackReceiver instead for continuous receiving.

Is callbackreceiver is suitable for duplex mode when i also send the data from reciever. the int data.

IoTThinks, 2020-11-03:
https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDuplexCallback/LoRaDuplexCallback.ino

closed 2020-11-16 by IoTThinks