sandeepmistry/arduino-LoRa #407: Transmit multiple single bytes or an array

maxholgasson, 2020-11-04:
Hi,

I stored a device's settings in an array of bytes that I want to transmit
as an example:
byte value_a=10;
byte value_b=11;

```
LoRa.beginPacket();
LoRa.print(2); //number of follwing bytes
LoRa.print(value_a);
LoRa.print(value_b);
LoRa.endPacket();
```

Receiving as a continious string works
```
while (LoRa.available()) {
        Serial.print((char)LoRa.read());
}
```

but not like 
```
byte numberofbytes;
byte received_a;
byte received_b; 
while (LoRa.available()) {
        numberofbytes =  LoRa.read());
        received_a = LoRa.read());
        received_b = LoRa.read());
}
Serial.print("Received value a");  Serial.println(received_a);
Serial.print("Received value b");  Serial.println(received_b);
```

I didn't find any hint in the example and asking how that could be done?

maxholgasson, 2020-11-05:
Found an example and the API description was helpful

closed 2020-11-05 by maxholgasson

IoTThinks, 2020-11-05:
This is how you use the library API.
Everything is actually based on bytes.