sandeepmistry/arduino-LoRa #410: Compatibility with STM32 boards

maxholgasson, 2020-11-09:
Hi,
I just installed in my Arduino IDE the boards from STM and tried to compile this library's receiver example for a Nucleo 64 board. Compilation was sucessful. Does that mean I could use such a board with your library? https://www.st.com/en/evaluation-tools/nucleo-l152re.html
I'm running out of space, memory and pins of the 328P and the STM32 chips look very decent concerning power consumption and everything else

IoTThinks, 2020-11-10:
For STM32, you can use stm32duino library to program your STM32 board in Arduino IDE.
Then use this library for LoRa as normal.

Sample code for stm32L0 and LoRa SX1278. A bit outdated only.
https://github.com/IoTThinks/EasyLoRaNode_LowPower

maxholgasson, 2020-11-13:
Thanks for your info! Great!

closed 2020-11-16 by IoTThinks