sandeepmistry/arduino-LoRa #411: SPI connection problem

ptkuo0322, 2020-11-15:
Are you receiving `Starting LoRa failed` while using the demo code?

PLEASE see the [FAQ #1](https://github.com/sandeepmistry/arduino-LoRa#faq) about using [setPins](https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins) **BEFORE** submitting an issue.

Initially, I have the same problem for "Starting LoRa failed` while using the demo code". I did all the method that I found on the internet, and still have one problem.

According to the closed issue, https://github.com/sandeepmistry/arduino-LoRa/issues/68, in that issue, there is an approach to  "add a Serial.println(version) " in LoRa.cpp file, and the serial would output some value.

I have two adafruit FRM95W, one of them works, and the other one doesn't work.

I already use the setPins to change the NSS, RESET, DIO0 to different pins and slow down the SPI frequency to 2E6 by LoRa.setSPIFrequency(frequency). But one of them does not work.

As a result, I have a question here,

Even though I use setPins and LoRa.setSPIFrequency to change the default value, the value print out by Serial.println(version) is still 0. I guess, that represents something wrong about my  SPI connection or CS pin. Does anyone know how to fix this problem?

IoTThinks, 2020-11-15:
1. Check issue #199?
Likely wrong pin.

2. Try to use external 3.3 power?

3. Try to set SS to OUTPUT and HIGH before LoRa.begin?

ptkuo0322, 2020-11-15:
> 1. Check issue #199?
>    Likely wrong pin.
> 2. Try to use external 3.3 power?
> 3. Try to set SS to OUTPUT and HIGH before LoRa.begin?

Thanks for quick reply. I try 1 and 3, but still get 0 for the Serial.println(version) value. However, if I connect vin to5V pin on arduino, the value would be 255 which is the same as I get While using STM32.

I have other questions.

1. For the antenna, can I use spring for 915Mhz as the external for Lora to communication?

2. How to do debugging since I can send package at the transmitter side, but at the receiver size, after it finishes initializing, it does not receive anything...... Can you please tell me what can I do to find out the possible reason.

Thank you

IoTThinks, 2020-11-15:
So the LoRa failed issue is fixed by using Vin? 
If yes, then likely due to insuffcient power.

ptkuo0322, 2020-11-15:
> So the LoRa failed issue is fixed by using Vin?
> If yes, then likely due to insuffcient power.

I use 5V on arduino Uno

IoTThinks, 2020-11-15:
You can use spring antenna if the antenna is tuned for 915 Mhz.

Or you can use a bronze wire.
For 915Mhz, it is around 78mm.

299792458 / 915000000 / 4 * 0.95 * 100 = 78mm

ptkuo0322, 2020-11-15:
> You can use spring antenna if the antenna is tuned for 915 Mhz.
> 
> Or you can use a bronze wire.
> For 915Mhz, it is around 78mm.
> 
> 299792458 / 915000000 / 4 * 0.95 * 100 = 78mm

Yes, the spring antenna is specially designed for 915Mhz according to the description on the website. But I still do not get the message on the received end....

Is there any way to figure out what is wrong? Thank you

IoTThinks, 2020-11-15:
Check if the DIO0 is attached to an interruptable GPIO.

Manual set the DIO0 in the code as the default DIO0 may not match the DIO0 in your board.

You can dumpRegister to see the internal state of LoRa. Check examples.
But need to understand the data sheet.

IoTThinks, 2020-11-15:
Btw, which example are you using?

Can try the default exapmple Receiver call back.

ptkuo0322, 2020-11-15:
I use the sender and the receiver example to do the test.

For sender, I use nucleo 32 F401RE as as the MCU, and I set up the SPI for this MCU

the code here is
 `#include <SPI.h>
#include <LoRa.h>

//define the pins used by the transceiver module
#define ss PB12
#define rst PB3
#define dio0 PA10

int counter = 0;

void setup() {
  //initialize Serial Monitor
  Serial.begin(115200);
  while (!Serial);
  Serial.println("LoRa Sender");
  LoRa.setSPIFrequency(2E6);
  pinMode(ss, OUTPUT);
  digitalWrite(ss, HIGH);  
  //setup LoRa transceiver module
  LoRa.setPins(ss, rst, dio0);
  
  //replace the LoRa.begin(---E-) argument with your location's frequency 
  //433E6 for Asia
  //866E6 for Europe
  //915E6 for North America

  //LoRa.setFrequency(915E6);
  while (!LoRa.begin(915E6)) {
    Serial.println("Lora Failed");
    delay(500);
  }
   // Change sync word (0xF3) to match the receiver
  // The sync word assures you don't get LoRa messages from other LoRa transceivers
  // ranges from 0-0xFF
  LoRa.setSyncWord(0xF3);
  Serial.println("LoRa Initializing OK!");
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  //Send LoRa packet to receiver
  
  LoRa.beginPacket();
 // LoRa.write("hello ");
  LoRa.write(counter);
  LoRa.endPacket();

  counter++;

  delay(10000);
}`

For Receiver, I use Arduino Uno As the MCU and  the code is

`#include <SPI.h>
#include <LoRa.h>

//define the pins used by the transceiver module
#define ss 10
#define rst 5
#define dio0 2

void setup() {
  //initialize Serial Monitor
  Serial.begin(115200);
  while (!Serial);
  Serial.println("LoRa Receiver"); 
  LoRa.setSPIFrequency(2E6);
  pinMode(ss, OUTPUT);
  digitalWrite(ss, HIGH); 
  //setup LoRa transceiver module
  LoRa.setPins(ss, rst, dio0);
  
  //replace the LoRa.begin(---E-) argument with your location's frequency 
  //433E6 for Asia
  //866E6 for Europe
  //915E6 for North America
  //LoRa.begin(866E6);
  while (!LoRa.begin(915E6)) {
    Serial.println(".");
    delay(500);
  }
   // Change sync word (0xF3) to match the receiver
  // The sync word assures you don't get LoRa messages from other LoRa transceivers
  // ranges from 0-0xFF
  LoRa.setSyncWord(0xF3);
  Serial.println("LoRa Initializing OK!");
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");
    Serial.print(packetSize);

    // read packet
    while (LoRa.available()) {
      String LoRaData = LoRa.readString();
      Serial.print(LoRaData); 
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}`

For the sender, the serial print is 

Hello 0,
Hellp,1.. and so on

But the receiver, nothing pop out except LoRa Initializing OK!

IoTThinks, 2020-11-15:
Define SPI pins in STM32 code.

The code works for my STM32L0 last time.

https://github.com/IoTThinks/EasyLoRaNode_LowPower/blob/master/EasyLoRaNode_LP/05_lora.ino#L10

ptkuo0322, 2020-11-15:
I have another question here. Since the Uno device is the slave, do I need to set it as the slave by some functions in the SPI.h. Or the Lora.h has already set that up for me?

Thank you.

IoTThinks, 2020-11-16:
You can read the source code LoRa.cpp for more information.

Uno has fixed SPI pins. This library sets the SPI pins for you.

However, stm32 and esp32 can use many gpios for SPI. Need to check if the default settings for SPI fits the board.

huyle250398, 2020-11-16:
I have the same problem, I'm using Arduino nano with lora02 and it displays "Starting Lora failed", have you fixed this problem?

IoTThinks, 2020-11-16:
> I have the same problem, I'm using Arduino nano with lora02 and it displays "Starting Lora failed", have you fixed this problem?
It is unlikely an issue of this library.

Arduino Nano is known for providing insufficient power for LoRa.
Just try again external 3.3v power for LoRa.

huyle250398, 2020-11-16:
@IoTThinks thank you, I'll
 try

ptkuo0322, 2020-11-17:
> Define SPI pins in STM32 sender.
> 
> The code works for my STM32L0 last time.
> 
> https://github.com/IoTThinks/EasyLoRaNode_LowPower/blob/master/EasyLoRaNode_LP/05_lora.ino#L10

I discover a problem today, for the code I post previously, it cannot work on the LoRa module that I used 

> https://www.adafruit.com/product/3072

However, I use the different LoRa module from the other group, the same code works.

> https://www.amazon.com/gp/product/B07KZPQ4GB/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

Does anyone know the reason?

IoTThinks, 2020-11-17:
Can try to set OUTPUT and HIGH for CS pin before LoRa.begin().
Or you can do LoRa.dumpRegisters(Serial) to compare register values of two modules.
Or faulty module?

ptkuo0322, 2020-11-17:
> Can try to set OUTPUT and HIGH for CS pin before LoRa.begin().
> Or you can do LoRa.dumpRegisters(Serial) to compare register values of two modules.
> Or faulty module?

I tried to set CS pin as OUTPUT and high before LoRa.begin(). However, still doesn't work

As for the   LoRa.dumpRegisters(Serial), I need to try tomorrow since I need to borrow the module from other people. if the value is different, what does it mean and what should I do. Thank you for replying

IoTThinks, 2020-11-17:
Buy another one and try? Haha

ptkuo0322, 2020-11-17:
I plan to do so..........although I have spend three days. on it....