sandeepmistry/arduino-LoRa #413: Cannot receive data when interfacing Lora02 with ESP8266

huyle250398, 2020-11-16:
Hi guys, I'm a newbie of Lora, I'm using an Arduino nano as a Lora sender to send temperature and humidity, and Node MCU ESP8266 as a receiver, the sender works fine but as for ESP8266 I did not receive anything except the output " ⸮Hp⸮⸮>dh⸮⸮H<⸮⸮LoRa Receiver LoRa Initializing OK!" 
How can i solve this problem, please help, best regard

IoTThinks, 2020-11-17:
What is the pin mapping?
Remove your code.
Use the default Receiver Callback example to check first.

huyle250398, 2020-11-17:
my pin map is:
DIO0->D2
RST-> D1
NSS-> D8
MISO->D6
MOSI->D7
I tried the callback example but still cannot read String

IoTThinks, 2020-11-17:
Try D5, D6, D7 and D8 instead.

And force to use the above pins for SPI before LoRa.begin.
The default SPI pins may not fit all boards.

SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI);
LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO0);

[image]

huyle250398, 2020-11-17:
thanks bro, I'll try

huyle250398, 2020-11-17:
[image]
I tried with uno but it still cannot receive

IoTThinks, 2020-11-17:
Show your code.

huyle250398, 2020-11-17:
this is for sender:

#include <LoRa.h>
#include <SPI.h>
#include "DHT.h"

#define DHTPIN A0     // dht11
#define rainPin A2 // rain
#define soilPin A1
#define ss 14
#define rst 13
#define dio0 12

#define DHTTYPE DHT11   // DHT 11

String  myStr, myStr2; 
DHT dht(DHTPIN, DHTTYPE);

//Stores temperature value

int counter = 0; 
const int rainMin = 0;     // sensor minimum
const int rainMax = 1024;  // sensor maximum
void setup() {
  pinMode(ss, OUTPUT);
  digitalWrite(ss, HIGH);
  LoRa.setPins(ss, rst, dio0);
  Serial.begin(9600);
  while (!Serial);
  Serial.println("LoRa Sender");
  dht.begin();     //initialize DHT11
  // setting lora frquency
  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);// if lora connection failed, do nothing 
  } 
  Serial.println("LoRa Initializing OK!");
}
void loop() 
{
  int hum = dht.readHumidity();                                              
  int temp = dht.readTemperature();
  int mois=analogRead(soilPin);
  int rain=analogRead(rainPin);
  int range = map(rain, rainMin, rainMax, 0, 3);
  if (isnan(hum) || isnan(temp)) {                                                
    Serial.println("Failed to read from DHT sensor!");
    return;
  }
  Serial.print("Sending packet: ");
  Serial.println(counter);
  Serial.print("\n");
  String msg= String(temp)+String(hum)+ String(mois)+String(rain);
  // send packet
  counter++;
  Serial.print(msg);
  LoRa.beginPacket();
  LoRa.print(msg);
  LoRa.endPacket();
  
  delay(5000);
}

and uno receiver:
#include <SPI.h>
#include <LoRa.h>

#define ss 10

#define rst 9
#define dio0 2

void setup() {
  SPI.begin();
  pinMode(ss, OUTPUT);
  digitalWrite(ss, HIGH);
  LoRa.setPins(ss, rst, dio0);
  
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver Callback");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
 
 
}

void loop() {

  int packetSize= LoRa.parsePacket();
  if(packetSize)
  {
    Serial.print("Received: ");
    delay(100);
    while(LoRa.available())
    {   
        Serial.print((String)LoRa.read());
        
      }
     Serial.print("with rssi:");
      Serial.println(LoRa.packetRssi());
      
    }
}

IoTThinks, 2020-11-17:
1. Use the default sender code and receive callback code.

Remove your custom code.

2. Set SPI pins for esp8266.
Check my previous comment.

IoTThinks, 2020-11-17:
[image]

IoTThinks, 2020-12-24:
ESP8266 + AI-Thinker ra01/02 is confirmed working with D5, 6, 7 and 8 as SPI pins.
And D4 as interrupt.
[image]

https://github.com/sandeepmistry/arduino-LoRa/issues/419#issuecomment-747509424

closed 2020-12-24 by IoTThinks