sandeepmistry/arduino-LoRa #415: Communication with SX1262 module,

TinManAkshay, 2020-11-24:


TinManAkshay, 2020-11-24:
Hi all,

I have been using these libraries in adafruit atmega 32u4 to communicate with other sx1272 module based on Raspberry pi. Thats works fine.
But when I transmit the data to our custom board (having sx1262 chip on it), SX1262 doesnt receive it. 

LoRa configuration on both sides-> 923.3mhz, 500khz, sf=7, cr=6, power=20, syncword = 0x34, preamble length=8, enabled CRC. 

I wanted to know if somebody has some familiarity with SX262 chip communicating to SX1272?

Thanks,
Akshay

IoTThinks, 2020-11-25:
This library may not support sx126x.
For sx126x, you may try RadioLib instead.

And set both sides to have the same settings.

closed 2021-05-25 by IoTThinks