sandeepmistry/arduino-LoRa #416: Sending/receiving issues with TTGO LoRa32 Board (SX1276)

Kasperdelasopa, 2020-11-29:
Hello,

i tried to implement a Bi-directional communication. But at some point in time the communication doesn't seem to be reliable.
The communication i tried to accomplish is a request/respond scenario. It seems that the SX1276 Module stops receiving at some point in time.

It works most of the times but sometimes the two devices just doesn't send or receive.

Are there general communication rules, like "there needs to be a pause between sending and receiving"?

Best regards

IoTThinks, 2020-11-30:
By default, LoRa.endPacket() will wait until a transmission completes.
After transmission, you can call LoRa.receive() again.

orimate, 2021-02-15:
Hi, I have the same issue. I called extra LoRa.receive() after LoRa.endPacket() but does not helped.

IoTThinks, 2021-02-19:
@orimate 
Check the sample code again.
The below code is incorrect.

onReceive(LoRa.parsePacket());

closed 2021-02-19 by IoTThinks