sandeepmistry/arduino-LoRa #418: Failed Receiving.

HamzaHajeir, 2020-12-05:
Hi, 

I'm testing out my LoRa connectivity using duplex example. 

It seems everything up to sending is great! But I can't receive from both modules. 

Both modules run this code : 

```cpp

#include <SPI.h> // include libraries
#include <LoRa.h>

//setTxPower

const int csPin = D1;    // LoRa radio chip select
const int resetPin = D2; // LoRa radio reset
const int irqPin = D0;   // change for your board; must be a hardware interrupt pin

#define dev1

#ifdef dev1
#define LOCAL_ADDR 0xBB
#define DEST_ADDR 0xFF
#else
#define LOCAL_ADDR 0xFF
#define DEST_ADDR 0xBB
#endif
String outgoing;          // outgoing message
byte msgCount = 0;        // count of outgoing messages
byte localAddress = LOCAL_ADDR; // address of this device
byte destination = DEST_ADDR;  // destination to send to
long lastSendTime = 0;    // last send time
int interval = 2000;      // interval between sends

void onReceive(int packetSize);
void sendMessage(String outgoing);

void setup()
{
    Serial.begin(115200); // initialize serial

    Serial.println("LoRa Duplex with callback");

    // override the default CS, reset, and IRQ pins (optional)
    LoRa.setPins(csPin, resetPin, irqPin); // set CS, reset, IRQ pin

    if (!LoRa.begin(433E6))
    { // initialize ratio at 915 MHz
        Serial.println("LoRa init failed. Check your connections.");
        while (true)
            ; // if failed, do nothing
    }

    LoRa.setTxPower(10);

    LoRa.onReceive(onReceive);
    LoRa.receive();
    Serial.println("LoRa init succeeded.");
    LoRa.dumpRegisters(Serial);
}

void loop()
{
    if (millis() - lastSendTime > interval)
    {
        String message = "HeLoRa World!"; // send a message
        sendMessage(message);
        Serial.println("Sending " + message);
        lastSendTime = millis();        // timestamp the message
        interval = random(2000) + 1000; // 2-3 seconds
        LoRa.receive();                 // go back into receive mode
    }
}

void sendMessage(String outgoing)
{
    LoRa.beginPacket();            // start packet
    LoRa.write(destination);       // add destination address
    LoRa.write(localAddress);      // add sender address
    LoRa.write(msgCount);          // add message ID
    LoRa.write(outgoing.length()); // add payload length
    LoRa.print(outgoing);          // add payload
    LoRa.endPacket();              // finish packet and send it
    msgCount++;                    // increment message ID
}

void onReceive(int packetSize)
{
    if (packetSize == 0)
        return; // if there's no packet, return

    // read packet header bytes:
    int recipient = LoRa.read();       // recipient address
    byte sender = LoRa.read();         // sender address
    byte incomingMsgId = LoRa.read();  // incoming msg ID
    byte incomingLength = LoRa.read(); // incoming msg length

    String incoming = ""; // payload of packet

    while (LoRa.available())
    {                                  // can't use readString() in callback, so
        incoming += (char)LoRa.read(); // add bytes one by one
    }

    if (incomingLength != incoming.length())
    { // check length for error
        Serial.println("error: message length does not match length");
        return; // skip rest of function
    }

    // if the recipient isn't this device or broadcast,
    if (recipient != localAddress && recipient != LOCAL_ADDR)
    {
        Serial.println("This message is not for me.");
        return; // skip rest of function
    }

    // if message is for this device, or broadcast, print details:
    Serial.println("Received from: 0x" + String(sender, HEX));
    Serial.println("Sent to: 0x" + String(recipient, HEX));
    Serial.println("Message ID: " + String(incomingMsgId));
    Serial.println("Message length: " + String(incomingLength));
    Serial.println("Message: " + incoming);
    Serial.println("RSSI: " + String(LoRa.packetRssi()));
    Serial.println("Snr: " + String(LoRa.packetSnr()));
    Serial.println();
}

```

Output (mainly for `    LoRa.dumpRegisters(Serial);` : 
dev1 : 

```
LoRa Duplex with callback
LoRa init succeeded.
0x0: 0xFF
0x1: 0x85
0x2: 0x1A
0x3: 0xB
0x4: 0x0
0x5: 0x52
0x6: 0x6C
0x7: 0x40
0x8: 0x0
0x9: 0x88
0xA: 0x9
0xB: 0x2B
0xC: 0x23
0xD: 0x1
0xE: 0x0
0xF: 0x0
0x10: 0x0
0x11: 0x0
0x12: 0x0
0x13: 0x0
0x14: 0x0
0x15: 0x0
0x16: 0x0
0x17: 0x0
0x18: 0x4
0x19: 0x0
0x1A: 0x0
0x1B: 0x2B
0x1C: 0x0
0x1D: 0x72
0x1E: 0x70
0x1F: 0x64
0x20: 0x0
0x21: 0x8
0x22: 0x1
0x23: 0xFF
0x24: 0x0
0x25: 0x0
0x26: 0x4
0x27: 0x0
0x28: 0x0
0x29: 0x0
0x2A: 0x0
0x2B: 0x0
0x2C: 0xE
0x2D: 0x50
0x2E: 0x14
0x2F: 0x45
0x30: 0x55
0x31: 0xC3
0x32: 0x5
0x33: 0x27
0x34: 0x1C
0x35: 0xA
0x36: 0x3
0x37: 0xA
0x38: 0x42
0x39: 0x12
0x3A: 0x52
0x3B: 0x1D
0x3C: 0x0
0x3D: 0xAF
0x3E: 0x0
0x3F: 0x0
0x40: 0x0
0x41: 0x0
0x42: 0x12
0x43: 0x24
0x44: 0x2D
0x45: 0x0
0x46: 0x3
0x47: 0x0
0x48: 0x4
0x49: 0x23
0x4A: 0x0
0x4B: 0x9
0x4C: 0x5
0x4D: 0x84
0x4E: 0x32
0x4F: 0x2B
0x50: 0x14
0x51: 0x0
0x52: 0x0
0x53: 0x10
0x54: 0x0
0x55: 0x0
0x56: 0x0
0x57: 0xF
0x58: 0xE0
0x59: 0x0
0x5A: 0xC
0x5B: 0xE5
0x5C: 0x6
0x5D: 0x0
0x5E: 0x5C
0x5F: 0x78
0x60: 0x0
0x61: 0x1C
0x62: 0xE
0x63: 0x5B
0x64: 0xCC
0x65: 0x0
0x66: 0x1
0x67: 0x50
0x68: 0x0
0x69: 0x0
0x6A: 0x0
0x6B: 0x0
0x6C: 0x0
0x6D: 0x0
0x6E: 0x0
0x6F: 0xB
0x70: 0xD0
0x71: 0x0
0x72: 0x13
0x73: 0x0
0x74: 0x0
0x75: 0x0
0x76: 0x0
0x77: 0x0
0x78: 0x0
0x79: 0x0
0x7A: 0x0
0x7B: 0x0
0x7C: 0x0
0x7D: 0x0
0x7E: 0x0
0x7F: 0x0
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
```

dev2 : 
```
LoRa init succeeded.
0x0: 0xFF
0x1: 0x85
0x2: 0x1A
0x3: 0xB
0x4: 0x0
0x5: 0x52
0x6: 0x6C
0x7: 0x40
0x8: 0x0
0x9: 0x88
0xA: 0x9
0xB: 0x2B
0xC: 0x23
0xD: 0x1
0xE: 0x0
0xF: 0x0
0x10: 0x0
0x11: 0x0
0x12: 0x0
0x13: 0x0
0x14: 0x0
0x15: 0x0
0x16: 0x0
0x17: 0x0
0x18: 0x4
0x19: 0x0
0x1A: 0x0
0x1B: 0x2B
0x1C: 0x0
0x1D: 0x72
0x1E: 0x70
0x1F: 0x64
0x20: 0x0
0x21: 0x8
0x22: 0x1
0x23: 0xFF
0x24: 0x0
0x25: 0x0
0x26: 0x4
0x27: 0x0
0x28: 0x0
0x29: 0x0
0x2A: 0x0
0x2B: 0x0
0x2C: 0xF
0x2D: 0x50
0x2E: 0x14
0x2F: 0x45
0x30: 0x55
0x31: 0xC3
0x32: 0x5
0x33: 0x27
0x34: 0x1C
0x35: 0xA
0x36: 0x3
0x37: 0xA
0x38: 0x42
0x39: 0x12
0x3A: 0x52
0x3B: 0x1D
0x3C: 0x0
0x3D: 0xAF
0x3E: 0x0
0x3F: 0x0
0x40: 0x0
0x41: 0x0
0x42: 0x12
0x43: 0x24
0x44: 0x2D
0x45: 0x0
0x46: 0x3
0x47: 0x0
0x48: 0x4
0x49: 0x23
0x4A: 0x0
0x4B: 0x9
0x4C: 0x5
0x4D: 0x84
0x4E: 0x32
0x4F: 0x2B
0x50: 0x14
0x51: 0x0
0x52: 0x0
0x53: 0xF
0x54: 0x0
0x55: 0x0
0x56: 0x0
0x57: 0xF
0x58: 0xE0
0x59: 0x0
0x5A: 0xC
0x5B: 0xF2
0x5C: 0x6
0x5D: 0x0
0x5E: 0x5C
0x5F: 0x78
0x60: 0x0
0x61: 0x1C
0x62: 0xE
0x63: 0x5B
0x64: 0xCC
0x65: 0x0
0x66: 0x1
0x67: 0x50
0x68: 0x0
0x69: 0x0
0x6A: 0x0
0x6B: 0x0
0x6C: 0x0
0x6D: 0x0
0x6E: 0x0
0x6F: 0xB
0x70: 0xD0
0x71: 0x0
0x72: 0x13
0x73: 0x0
0x74: 0x0
0x75: 0x0
0x76: 0x0
0x77: 0x0
0x78: 0x0
0x79: 0x0
0x7A: 0x0
0x7B: 0x0
0x7C: 0x0
0x7D: 0x0
0x7E: 0x0
0x7F: 0x0
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
Sending HeLoRa World!
```

By comparing, it seems 2 modules differs in 3 registers  : 
- 2C
- 53
- 5B

What might be my problem ? 

could it be the soldered antenna ? 433Mhz antenna. but originally ipx terminal, I've cut it and soldered it to the pin.

Any ideas ?

IoTThinks, 2020-12-07:
Show us the photos of your LoRa modules. Esp. how you soilder the antenna. 
And use the default example below:
- Sender: https://github.com/sandeepmistry/arduino-LoRa/tree/master/examples/LoRaSender
- Receiver: https://github.com/sandeepmistry/arduino-LoRa/tree/master/examples/LoRaReceiverCallback

You need to wire the DIO0.
Tell us your MCU (ESP32, Arduino...) and how you wire the pins.

HamzaHajeir, 2020-12-08:
Hi, Thanks for your input

The LoRa Module is [E19-433M20S2](https://www.aliexpress.com/item/4000538750358.html). 

I've used the default examples, Does reset regularly because of default SS , RST , IRQ  pins, After correction by LoRa.setPins(D1,D2,D0) it worked and passed the while(!LoRa.begin()) loop.
Also changed Frequency to 433E6.

Pins wiring : 
NodeMCU | E19
D01 --> D0
NSS --> D1
RST --> D2
SCK --> D5
MISO--> D6
MOSI --> D7
VCC --> 3V3
Gnf --> Gnd

Well, Your input really helped to make one step further, I got aware of Antenna got 2 wires (Ant and Gnd), And soldered them separately.

Modified sketches : 

SENDER : 
```cpp
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial.println("LoRa Sender");
    LoRa.setPins(csPin, resetPin, irqPin); // set CS, reset, IRQ pin
  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}
```

RECEIVER : 

```cpp
#include <SPI.h>
#include <LoRa.h>

#ifdef ARDUINO_SAMD_MKRWAN1300
#error "This example is not compatible with the Arduino MKR WAN 1300 board!"
#endif
void onReceive(int packetSize);

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial.println("LoRa Receiver Callback");

  LoRa.setPins(csPin, resetPin, irqPin); // set CS, reset, IRQ pin
  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  // Uncomment the next line to disable the default AGC and set LNA gain, values between 1 - 6 are supported
  // LoRa.setGain(6);
  
  // register the receive callback
  LoRa.onReceive(onReceive);

  // put the radio into receive mode
  LoRa.receive();
}

void loop() {
  // do nothing
}

void onReceive(int packetSize) {
  // received a packet
  Serial.print("Received packet '");

  // read packet
  for (int i = 0; i < packetSize; i++) {
    Serial.print((char)LoRa.read());
  }

  // print RSSI of packet
  Serial.print("' with RSSI ");
  Serial.println(LoRa.packetRssi());
}
```

Solder images : 
Sender : 
[image]
[image]

Receiver : 

[image]
[image]

I've confirmed pins are correctly connecting to both sides (NodeMCU and E19) with continuity check. Also there's no short in both sides.

HamzaHajeir, 2020-12-08:
The same issue goes for this module : [Ra-01](https://www.alibaba.com/product-detail/Ai-Thinker-Long-Range-Spread-Spectrum_62379245471.html) 

Same code.

Images : 
Sender : 

[image]
[image]

Receiver : 

[image]

confirmed the connections as described before : 
NodeMCU | Ra-01
D01 --> D0
NSS --> D1
RST --> D2
SCK --> D5
MISO--> D6
MOSI --> D7
VCC --> 3V3
Gnf --> Gnd

IoTThinks, 2020-12-09:
Strange.
You never define csPin, resetPin and irqPin?

LoRa.setPins(csPin, resetPin, irqPin);

Your board is NodeMCU esp8266?

IoTThinks, 2020-12-09:
Seems you use non-default SPI pins for LoRa.

You can try to set the pin for SPI before LoRa.begin for both nodes.

  SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_SS);
  LoRa.setPins(LORA_SS, LORA_RESET, LORA_DIO012);

HamzaHajeir, 2020-12-09:
Ofcourse I've defined them. 
Sorry it was as shared code just before #ifdef preprocessing code block :
```cpp
#ifdef SENDER 
*sender code*
#else 
*receiver code*
#endif 
```

You can find the code [here](https://gist.github.com/HamzaHajeir/3c1993ef16e82ce466c3fde5c297f571).

Yes, It's ESP8266 NodeMCU. I've used this map for reference (Used the right side SPI Pins *except CS*) : 

[image]

I'll do try your suggestion and keep this thread updated.

HamzaHajeir, 2020-12-09:
> Seems you use non-default SPI pins for LoRa.
> 
> You can try to set the pin for SPI before LoRa.begin for both nodes.
> 
> SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_SS);
> LoRa.setPins(LORA_SS, LORA_RESET, LORA_DIO0);

There's no improvements of what I actually got, still not receiving messages

HamzaHajeir, 2020-12-09:
I've tested [RadioLib](https://github.com/jgromes/RadioLib/tree/master/examples/SX127x), And it worked well, with adding DIO1 to the connection.

I don't know what happened here, But it seems software issue.

IoTThinks, 2020-12-09:
Dio1 is not used for sending or receiving.

If got free time this weekend, I will try my esp8266 with ra01.

HamzaHajeir, 2020-12-09:
Great, 

For DIO1, It's just used in that library.

IoTThinks, 2020-12-24:
esp8266 + ra01/02 are confirmed working at https://github.com/sandeepmistry/arduino-LoRa/issues/419#issuecomment-747509424

closed 2020-12-24 by IoTThinks