sandeepmistry/arduino-LoRa #419: loraduplexcallback problem

huyle250398, 2020-12-17:
hi guys, I'm testing Loraduplexcallback between Arduino mega and esp8266. the problem is mega only sending and not receiving. has anyone got the same problem?

huyle250398, 2020-12-17:
[image]
my result on serial monitor

IoTThinks, 2020-12-17:
Try the LoRaSender and LoRaReceiverCallback first.

Then swap the sender and receiver.

huyle250398, 2020-12-17:
@IoTThinks ths bro, I tried that but esp cannot receive, so what probably the problem, mega still sending

IoTThinks, 2020-12-17:
What is your pin mapping for LoRa in esp8266?

You can try to set the pin for SPI before LoRa.begin for esp8266.

SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_SS);
LoRa.setPins(LORA_SS, LORA_RESET, LORA_DIO0);

IoTThinks, 2020-12-17:
May try the D5-D8 of esp8266 for LoRa?

http://wei48221.blogspot.com/2017/09/esp8266-testing-spi-interface-using.html?m=1

IoTThinks, 2020-12-21:
@huyle250398 How is the progress?

huyle250398, 2020-12-24:
@IoTThinks ths bro, i tried with D5-D8 and it worked, 2 interrupt pin I used are D5 and D4

closed 2020-12-24 by IoTThinks