sandeepmistry/arduino-LoRa #420: Ra 01 Problem Recivig

KittyCat00, 2020-12-23:
Hi guys!
I'm using a LoRa 2 modules "**Ra-01**" from Ai-thinker, with 2 Arduinos UNO, I'm using level shifter for SPI communication with 3.3v regulator from Arduino. I uploaded the example of Sender and Receiver to each Arduino.
I have to **433E6 in place of 915E6,** cause the modules just can function at that frecuency.
But the **problem is I cannot see the correct message in the Receiver**, on some occasions I just receive 'trash', in others, I don't receive anything.
In the console of the sender, the **serial monitor shows correctly the message 'Sending Packet: X',** but in the serial monitor of Receiver, I don't receive anything.
I hope you can help me. Thanks in advance.

[image]

IoTThinks, 2020-12-24:
How do you do the wiring?

KittyCat00, 2020-12-26:
I´ve already checked the wiring it´s ok, that´s strange because in some occasions it works, in others not. I've tried also with the answer https://github.com/sandeepmistry/arduino-LoRa/issues/125#issuecomment-379545493, but still does not work properly.

mchacher, 2020-12-26:
My 2 cents. It might a slight frequency shift on the 2 modules ... try to adjust by 500Hz steps

IoTThinks, 2020-12-27:
@mchacher The frequency error could be up to 25% of bandwidth.
So if he uses default bw 125Khz, the error could be up to 31.5kHz.

So I guess not an issue.

IoTThinks, 2020-12-27:
@KittyCat00 Show us a photo of your physical setup.
We may spot errors.

You can try to run without the logic shifter for a while to see if it works.

KittyCat00, 2020-12-27:
@IoTThinks I've tried first without the logic shifter, but there was an error in the Setup.
That problem was fixed with the shifter.

Here the pictures
[image]
[image]
[image]

IoTThinks, 2020-12-28:
> I´ve already checked the wiring it´s ok, that´s strange because in some occasions it works, in others not. I've tried also with the answer [#125 (comment)](https://github.com/sandeepmistry/arduino-LoRa/issues/125#issuecomment-379545493), but still does not work properly.

What is your pin mapping? Espcially DIO0?
What examples you use?

Check if your soildering is correct? Aka the pins are not wrongly connected with each other.

Sometimes you receives trash? Either pin connection is shorted. Or the logic shifter is wrong somewhere.

KittyCat00, 2021-01-13:
Hi.
In the end, I thought that the problem could be the logic shifter so I decided to change the Arduino UNO, now I´m using Arduino Mini Pro (3.3v 8 mHz), I still have the same problem. The sender says that the package is being transmitted but I cannot see it at the serial monitor in the Receiver.
Do you have any other idea?
I´ve alredy check the soildering .
I´m using "LoRa Reciver" and "LoRa Sender".
My DIO0 is mapping on pin 2.
[image]
[image]

IoTThinks, 2021-01-13:
Your antenna looks longer than expected for 433Mhz. 
Likely wrong antenna.
You can keep the two modules as near as possible (in few centimetres) to each other to see if the messages are received.

You can try this for receiver. Change to 433E6.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaReceiverCallback/LoRaReceiverCallback.ino

If still not works. You run this example in the receiver side https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDumpRegisters/LoRaDumpRegisters.ino
And upload the register value.

I seldom use Uno+Logic shifter.
For Arduino Mega 2650, I can send/receive well without logic shifter.

KittyCat00, 2021-01-13:
I change the Reciver Program for "Receiver Callback".
I have this.
[image]
And I have this with the program of "Dump Registers" on Receiver's side
```
15:50:43.051 -> LoRa Dump Registers
15:50:43.084 -> 0x0: 0x5C
15:50:43.084 -> 0x1: 0x80
15:50:43.084 -> 0x2: 0x14
15:50:43.084 -> 0x3: 0x6
15:50:43.118 -> 0x4: 0x0
15:50:43.118 -> 0x5: 0x52
15:50:43.118 -> 0x6: 0x68
15:50:43.151 -> 0x7: 0x68
15:50:43.151 -> 0x8: 0xEB
15:50:43.151 -> 0x9: 0x8F
15:50:43.185 -> 0xA: 0x8
15:50:43.185 -> 0xB: 0x26
15:50:43.185 -> 0xC: 0x24
15:50:43.219 -> 0xD: 0x0
15:50:43.219 -> 0xE: 0x0
15:50:43.219 -> 0xF: 0x0
15:50:43.253 -> 0x10: 0x0
15:50:43.253 -> 0x11: 0x0
15:50:43.253 -> 0x12: 0x0
15:50:43.288 -> 0x13: 0x0
15:50:43.288 -> 0x14: 0x0
15:50:43.288 -> 0x15: 0x0
15:50:43.322 -> 0x16: 0x0
15:50:43.322 -> 0x17: 0x0
15:50:43.322 -> 0x18: 0x20
15:50:43.355 -> 0x19: 0x20
15:50:43.355 -> 0x1A: 0x0
15:50:43.355 -> 0x1B: 0x0
15:50:43.389 -> 0x1C: 0x0
15:50:43.389 -> 0x1D: 0x64
15:50:43.422 -> 0x1E: 0x60
15:50:43.422 -> 0x1F: 0x64
15:50:43.422 -> 0x20: 0x0
15:50:43.456 -> 0x21: 0x8
15:50:43.456 -> 0x22: 0x1
15:50:43.456 -> 0x23: 0xFF
15:50:43.490 -> 0x24: 0x0
15:50:43.490 -> 0x25: 0x0
15:50:43.490 -> 0x26: 0x4
15:50:43.523 -> 0x27: 0x0
15:50:43.523 -> 0x28: 0x0
15:50:43.523 -> 0x29: 0x0
15:50:43.557 -> 0x2A: 0x0
15:50:43.557 -> 0x2B: 0x0
15:50:43.557 -> 0x2C: 0x0
15:50:43.590 -> 0x2D: 0x20
15:50:43.590 -> 0x2E: 0x14
15:50:43.624 -> 0x2F: 0x45
15:50:43.624 -> 0x30: 0x4B
15:50:43.624 -> 0x31: 0xC3
15:50:43.658 -> 0x32: 0x0
15:50:43.658 -> 0x33: 0x27
15:50:43.658 -> 0x34: 0x1C
15:50:43.691 -> 0x35: 0xA
15:50:43.691 -> 0x36: 0x3
15:50:43.691 -> 0x37: 0xA
15:50:43.724 -> 0x38: 0x40
15:50:43.724 -> 0x39: 0x84
15:50:43.724 -> 0x3A: 0x24
15:50:43.758 -> 0x3B: 0x18
15:50:43.758 -> 0x3C: 0x0
15:50:43.791 -> 0x3D: 0xAF
15:50:43.791 -> 0x3E: 0x0
15:50:43.791 -> 0x3F: 0x0
15:50:43.825 -> 0x40: 0x0
15:50:43.825 -> 0x41: 0x0
15:50:43.825 -> 0x42: 0x12
15:50:43.860 -> 0x43: 0x24
15:50:43.860 -> 0x44: 0x2C
15:50:43.860 -> 0x45: 0x0
15:50:43.894 -> 0x46: 0x3
15:50:43.894 -> 0x47: 0x0
15:50:43.894 -> 0x48: 0x4
15:50:43.929 -> 0x49: 0x23
15:50:43.929 -> 0x4A: 0x0
15:50:43.929 -> 0x4B: 0x2
15:50:43.962 -> 0x4C: 0x5
15:50:43.962 -> 0x4D: 0x80
15:50:43.962 -> 0x4E: 0x32
15:50:43.996 -> 0x4F: 0x14
15:50:43.996 -> 0x50: 0x20
15:50:44.029 -> 0x51: 0x0
15:50:44.029 -> 0x52: 0x0
15:50:44.029 -> 0x53: 0xC
15:50:44.063 -> 0x54: 0x0
15:50:44.063 -> 0x55: 0x0
15:50:44.063 -> 0x56: 0x0
15:50:44.098 -> 0x57: 0xF
15:50:44.098 -> 0x58: 0xC0
15:50:44.098 -> 0x59: 0x0
15:50:44.131 -> 0x5A: 0xC
15:50:44.131 -> 0x5B: 0xFA
15:50:44.131 -> 0x5C: 0x7
15:50:44.165 -> 0x5D: 0x0
15:50:44.165 -> 0x5E: 0x5C
15:50:44.165 -> 0x5F: 0x78
15:50:44.198 -> 0x60: 0x0
15:50:44.198 -> 0x61: 0x18
15:50:44.198 -> 0x62: 0xE
15:50:44.231 -> 0x63: 0x5B
15:50:44.231 -> 0x64: 0xCC
15:50:44.266 -> 0x65: 0x0
15:50:44.266 -> 0x66: 0x1
15:50:44.266 -> 0x67: 0x20
15:50:44.299 -> 0x68: 0x0
15:50:44.299 -> 0x69: 0x0
15:50:44.299 -> 0x6A: 0x0
15:50:44.333 -> 0x6B: 0x0
15:50:44.333 -> 0x6C: 0x0
15:50:44.333 -> 0x6D: 0x0
15:50:44.367 -> 0x6E: 0x0
15:50:44.367 -> 0x6F: 0xB
15:50:44.367 -> 0x70: 0xC0
15:50:44.402 -> 0x71: 0x0
15:50:44.402 -> 0x72: 0x14
15:50:44.402 -> 0x73: 0x0
15:50:44.435 -> 0x74: 0x0
15:50:44.435 -> 0x75: 0x0
15:50:44.435 -> 0x76: 0x0
15:50:44.469 -> 0x77: 0x0
15:50:44.469 -> 0x78: 0x0
15:50:44.469 -> 0x79: 0x0
15:50:44.503 -> 0x7A: 0x0
15:50:44.503 -> 0x7B: 0x0
15:50:44.503 -> 0x7C: 0x0
15:50:44.536 -> 0x7D: 0x0
15:50:44.536 -> 0x7E: 0x0
15:50:44.570 -> 0x7F: 0x0
```

IoTThinks, 2021-01-13:
Change to another logic shifter or another antenna.
The packet is corrupted or interfered.

Ra01 433Mhz can not get the RSSI better than -40.

thote1276, 2021-02-02:
Hello, I'm currently facing the same kind of problem. Do you thing power input can be a problem ? I did the same as @KittyCat00 but I'm using a 5V to 3.3V plug on the 5V from the arduino which provide normally a sufficient amount of current.

IoTThinks, 2021-02-03:
@thote1276 Don't assume everything is the same with others' problems. 
1. Show your physical / real boards and wiring.
2. Show your code. To use sample in this library and remove your custom code.
3. Show screenshots of problems as proof.

thote1276, 2021-02-03:
Hello, 
Thanks for your message. Here are some complemetns about my stuff. 
The setup are 2 arduinos uno : 1 receiver and one transmitter.
On each arduino we will find a 8 channel shift converter (5V to 3.3V), a power converter (5V to 3.3V with more than 500mA possible). They are wired like this : MISO, MOSI, NSS and SCK on the correct pin (10 to 13, I check everything a couple of times), DIO0 on 2, RESET on 9 [This is what we can observe on the picture] 
[image]

The code are the one from the librairies (Lora_send, Lora_receiver). I added the line related to GPIO and SPI frequency to be sure.
[image]

Here are my result for the moment
[image]
 
In any case, thanks for the time you are according to my case. 
Best regards, Thomas

IoTThinks, 2021-02-17:
@thote1276 Try again without the logic converter?
In 1m distance, the RSSI for ra01 should be around -40 to -60.
RSSI -2, -164 or 7 are not right.

A stupid question: Do you wire MISO of UNO to MOSI or LoRa?
https://github.com/sandeepmistry/arduino-LoRa#semtech-sx1276777879-wiring

thote1276, 2021-02-18:
Hello, 
I convert my arduino to 3.3V remove the logic converter and it works. It is probably the waveform of inputs which were not correct
Thanks for your time, thanks for your attention
Regards, Thomas

IoTThinks, 2021-02-19:
@thote1276 How you convert your Arduino to 3.3v?

thote1276, 2021-02-20:
I remplace the LDO of 5V by one from 3.3V (ref AMS1117-33) for example. This way by external supply we will have 3.3V on the ATMEGA. For the USB part, I remove the fuse and find the pad next to the power supply of the usb of sold a 1N4001 to the 3.3 output of the LDO. There is a tutorial of adafruit with more detail.

IoTThinks, 2021-02-21:
Seem to be the issue at logic converter.
Thanks for the info.

closed 2021-02-21 by IoTThinks