sandeepmistry/arduino-LoRa #421: Documentation details

ptfila, 2020-12-27:
Thank you for the clean documentation in your API.md file!

It would be great to specify data types for each of the parameters in your API functions, as well as min and max values accepted. For example, to change the transmit power:
LoRa.setTxPower(txPower);
How low of a value can txPower be? 0 dB? Can it be negative? What's the data type of the parameter? 
How high can txPower be?

Another example:
In: LoRa.write(byte); is "byte" a uint8_t or a uint8_t *? (It should be a uint8_t but stating all data types of parameters for consistency would be great!)

Another example:
In: LoRa.setSignalBandwidth(signalBandwidth); what is the accepted range of "signalBandwidth"?

Thanks for all of your hard work on the LoRa API!

IoTThinks, 2020-12-28:
Most of the APIs indicate “Supported Values”.

https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#signal-bandwidth

ptfila, 2020-12-28:
I agree. I'm referring to the ones that don't, such as those mentioned in the comment :)

IoTThinks, 2020-12-28:
Tx power
Supported values are 2 to 20 for PA_OUTPUT_PA_BOOST_PIN, and 0 to 14 for PA_OUTPUT_RFO_PIN.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#tx-power

Write
byte - single byte to write to packet
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#writing

setSignalBandwidth
Supported values are 7.8E3, 10.4E3, 15.6E3, 20.8E3, 31.25E3, 41.7E3, 62.5E3, 125E3, 250E3, and 500E3.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#signal-bandwidth

BTW, you can create a pull request if you love to do so. :)

closed 2021-01-05 by IoTThinks