sandeepmistry/arduino-LoRa #427: Arduino Due fails

Keyxss, 2021-01-24:
I am receiving `Starting LoRa failed` every time I run the program on my Arduino Due, but when I plug in Arduino Uno it works perfectly fine. I use Lo Ra-02 sx1278. I run the basic sender program and it fails. What could be the problem? Does in not work on Arduino Due?

IoTThinks, 2021-01-24:
How do you so the wiring? Any screenshot?

Check if the Due uses 3.3v or 5v logic.
You may need logic converter if the board uses 5v logic.

Likely either wrong SPI wiring or logic level.

Keyxss, 2021-01-24:
[image]

IoTThinks, 2021-01-25:
Double check the SPI pins.
[image]

Different Arduino boards (Nano, Uno, Mega, Due...) may have different SPI pins.

Your wiring seems to be for Uno instead.
[image]

Keyxss, 2021-01-25:
[image]

I changed the pins according to that table, and it still does not work. Also I found here that Arduino UNO use 5v logic. Maybe that it the problem.

IoTThinks, 2021-01-25:
3.3v logic is better than 5v logic.
For 5v logic boards, sometimes you need a logic converter to 3.3v.

Normally, only 3 issues:
1. Wrong SPI pins. Check if the DIO0 is wired to interruptable GPIO.
2. Wrong logic level. Sometimes, a logic converter to 3.3v is needed.
3. Not enough power for LoRa. Up to 100mA.

Keyxss, 2021-01-25:
Somehow it worked. Thank you very much for your help

closed 2021-01-25 by Keyxss

IoTThinks, 2021-01-25:
@Keyxss How did you fix it?

IoTThinks, 2021-01-25:
Ok enjoy.

Keyxss, 2021-01-25:
Again thanks a lot for your time. It was my fault that I did not checked spi ports. I couldn't find anything on the internet. So I asked here.