sandeepmistry/arduino-LoRa #430: onReceive(LoRa.parsePacket()) causes 3x4 keypad column 3 to think its column 1

paulrose71, 2021-02-08:
Hi Sandeep, 
Thanks alot for this, its appreciated. I've tried everything but can't get it to work! You are my last resort!!

I have two Arduino UNO R3's talking to each other via two LoRa's (duinotech, SX1276, IC sensor, model XC-4392) at a range of 70 cm. Each LoRa shield just plugs onto the top of each UNO. I'm supplying both UNO's with 12v. The sender UNO has a simple membrane 3x4 keypad. It also has a i2C 16x2 LCD.  

When I comment out onReceive(LoRa.parsePacket()) the keypad is fully functional. But when I remove the comment, column 3 thinks its column 1? 

I've tried using different:
1. UNO's
2. LoRa shields
3. DIO pins
4. Tx power levels
5. SPI frequencies
6. keypads

#include <Keypad.h>
#include <SPI.h>                                            
#include <LoRa.h>
#include <Wire.h>

const byte ROWS = 4; 
const byte COLS = 3; 

char hexaKeys[ROWS][COLS] = {
  {'1', '2', '3'},
  {'4', '5', '6'},
  {'7', '8', '9'},
  {'*', '0', '#'}
};

byte rowPins[ROWS] = {8, 7, 6, 5}; 
byte colPins[COLS] = {4, 3, 2}; 

byte localAddress = 0xBB;     
byte destination = 0xFF;      
String incoming = "";         
String outgoing; 

Keypad customKeypad = Keypad(makeKeymap(hexaKeys), rowPins, colPins, ROWS, COLS); 

void setup() {    
  Serial.begin(9600);
  while (!Serial);
  LoRa.setSPIFrequency(8E6);                        
  if (!LoRa.begin(915E6)) {                         
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  Serial.println("LoRa init succeeded.");
  Serial.println();
  LoRa.setSyncWord(0x34);                            
  LoRa.setTxPower(20);                              
  LoRa.setSpreadingFactor(7);                       
  LoRa.setSignalBandwidth(125E3);                   
  LoRa.setCodingRate4(5);                            
  LoRa.receive();                                   
}

void loop() {                                              
  char customKey = customKeypad.getKey();
  if (customKey){
    outgoing = customKey;
    Serial.println(customKey);                                    
    LoRa.beginPacket();
    LoRa.write(destination);                               
    LoRa.write(localAddress);                              
    LoRa.write(outgoing.length());                         
    LoRa.print(customKey);
    LoRa.endPacket();                                          
    LoRa.receive();     
  }
  onReceive(LoRa.parsePacket());                           
}

void onReceive(int packetSize) {                                       

}

IoTThinks, 2021-02-17:
I do not think you can use onReceive(LoRa.parsePacket())
When you use continuous RX mode (RX callback), the chip will trigger onReceive to read the packets from registers.
LoRa.parsePacket() will try to intercept another LoRa packet in Single RX mode.

You can read the sample code again to use the library properly.
Look for Receiver (Single RX mode) or ReceiverCallback (Continuous RX mode)

paulrose71, 2021-02-18:
OK, thanks for that. Appreciated. I'll read the sample code again. Paul

closed 2021-02-19 by IoTThinks