sandeepmistry/arduino-LoRa #432: Range vs RSSI value

budnikasr, 2021-02-14:
Hello Everyone. I am hoping someone can help me. I am working with an engineer overseas and we are getting the exact same results with the exact same semtech sx1276 chip. We are struggling getting the range anywhere near what is documented. I'm in Canada so using the 915 version. Also the antenna (Spiral) on this is the one that came with the chip.

Here are the settings

TX - LoRa.setTxPower(20);
       LoRa.setSpreadingFactor(8);
       LoRa.setSignalBandwidth(125E3);
       LoRa.setGain(6);

RX - LoRa.setSpreadingFactor(8);
        LoRa.setSignalBandwidth(125E3);
        LoRa.setGain(6);

Prior to adding  LoRa.setGain(6) I was getting an rssi of -49 @ 1m
After adding  LoRa.setGain(6) I am getting an rssi of -38 @ 1m

If I test this in my house, 2 floors, my rssi goes from -38 @ 1m to -70 @ 5m and it is directly above, with the floor as the obstacle.
If I take this outside my rssi is at -118 @ 200m. If I go beyond this, I loose the connection completely.

These tests are all at ground level and various obstacles. It doesn't matter if there is line of sight or not.

Hope there is an easy explanation. The only thing I can think of is the antenna being the issue.

Thanks in advance for your help.

Robert

IoTThinks, 2021-02-15:
Wrongly tune antenna or insufficient power for LoRa.

Show us your physical MCU board/antenna/LoRa.

budnikasr, 2021-02-15:
Everything is attached to NodeMcu. The sx1276 is attached to 3.3v and the temp sensor and screen are attached to 5v.

[image]

[image]

IoTThinks, 2021-02-15:
Your rssi within 5m looks fine to me.
You can print the SNR within 5m.
It should be around 10 or more.

If not, then likely antenna issue. The 915 Mhz antenna length looks a bit long to me.

You can change your frequency to 916,917... to see if it is better.
If better, then there is interference issue.

budnikasr, 2021-02-15:
Hi, I have added the print SNR statement in my code and with the two devices 1m apart, I am getting an RSSI value of 33 and an SNR of 12.25. At 5m RSSI is now 60 and the SNR is 10.75. 

I've also changed the Frequency to 916 and 917 tested and got the exact same results.

IoTThinks, 2021-02-16:
The SNR is quite ok at 5v.
1. Do not suround your wires around the antenna.
2. You can try HopeRF 915Mhz antenna.
3. Try to supply the LoRa with external 3.3v to have more power. Some modules doesnt supply enough current.
4. Is this RFM module from HopeRF? What is the code name, RFM95w, 96w...? And for 868 or 915Mhz?

IoTThinks, 2021-02-16:
To use the sample code.
Dont use your custom code to avoid errors from custom code.

budnikasr, 2021-02-16:
1.) I agree about the wires. Unfortunately it was an after thought that I should have put the pins on the bottom. I have another set and will modify.
2.) I will order better antennas
3.) I will add external power to it.
4.) Here is the link from Aliexpress for the modules. https://www.aliexpress.com/item/32984655636.html?spm=a2g0s.9042311.0.0.30ea4c4dCteeFt

Regarding your other message, I simply used the examples from the library. First to send and receive and had the same rssi. I only added the oled screen and temp/humidity sensor after and still was getting the same results.

I'll make changes over the next couple of days and report back.

Thanks

IoTThinks, 2021-02-16:
I use RFM95W 915Mhz.
[image]

My 915Mhz antenna is quite short.
I check the VSWR of the antenna. Quite ok.

[image]

budnikasr, 2021-02-16:
I confirmed with the supplier before shipping and they only ship the 915mhz unit to North America even though their link shows an 868mhz module.  Regarding the antenna length, I googled the length for 915mhz and it indicates a 78mm antenna is required. Mine is 39mm.

budnikasr, 2021-02-17:
So just an update, everything now has gone south on me. Since attaching external power to both devices and separating out the 5v and 3v, the rx unit starts, runs for about 3-4 minutes at an rssi of -71 and snr of 5.75 @ 5m and then just stops. When I put everything back, it's fine.

IoTThinks, 2021-02-18:
Snr 5.75 @5m is not so good.  It means the power has lost a lot to reach the receiver compared to noise power.
Is it ok for more than 200m?

budnikasr, 2021-02-18:
I brought the units into a commercial building environment and it's a disaster. not sure how it just stops working after 5m when it has it's own power and I am not talking about a 9v battery. They are powered by separate external power supplies. When I reconnect them as they were to 2 laptops, it's better but nowhere near what is claimed for distance.

IoTThinks, 2021-02-19:
Buy a pair of RFM modules and correct 915Mhz antenna.

budnikasr, 2021-02-19:
Easier said than done. Seems to be a worldwide shortage and I don't have weeks to wait on Aliexpress or Alibaba.

DedePele, 2021-03-05:
Hi, I am beginer at LoRa  trying to get information to set my first point to point simple configuration. Your post is being very helpful. I really appreciate you could help.

1) How can I get SNR information from link? Must I include some special line of code as RSSI?

2)  I have 2 SX176 (915) and one Arduino Mega and one Wemos D1 board. I am not sure but I think Wemos logical level is 3,3v and Mega 5v. Browsing on SX1276 data sheet I can not see if it is 5v tolerant. There are confusion information about that. Some youtuber video show there no problems using 5 volts directly... I have none logical converter voltage here and I want to test my setup. I hope if you could help.

IoTThinks, 2021-03-05:
@DedePele Next time, please create a new issue.
1. Read this and check example code https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#packet-snr
2. You can connect LoRa to 3.3v of Mega. Most of the time, it will be ok.

DedePele, 2021-03-05:
Hello,

Very good day.

Sorry I am beginer from LoRa and also from Git Hub. In fact that is my
first post and I really don't know when I must create a new issue. I was
afreid to make something wrongly..kkk

Also this reply to my question was very fast! I don't know if it was
generated automatically or it came from sandeepmistry.

If this reply came to me from you (sandeepmistry) I am very pround. You
sandeepmistry made a great thing for me and your help for all is brilhant.
Thanks a lot and congratulations for your great job.

I will so try to set my first Lora connection directly from Mega and Wemos
without logical converter.

I hope you have a great day.

Um grande abraço a você aqui do Brasil.

BR,

André

IoTThinks, 2021-03-05:
@DedePele When the post is not relevant to the original topic, then you should create new post.

Ok, happy coding.

closed 2021-05-25 by IoTThinks