sandeepmistry/arduino-LoRa #434: Sending a message containing double quotes with Lora.print

jerome-angeloni, 2021-02-17:
Hi,

I have a working setup with 2 TTGO LORA boards. On the TX side, I'm using this library for sending a message. 
I can send this message {Air_temp: 25, Water_temp: 22}, and it is correctly received.
But actually I need to send {"Air_temp": 25, "Water_temp": 22}. For this I'm using the backslash escape character in front of the double quotes, but the print method does not detect the escape character and prints : {\"Air_temp\": 25, \"Water_temp\": 22}.

I've looked at the library code but can't find where the Lora.print function is defined. Can someone help? thanks!

IoTThinks, 2021-02-17:
The library sends data as bytes.
It doesn't care about backslash. Backslash is just a byte. Then you cast it as a char or number depending on how you want to see.

IoTThinks, 2021-02-17:
To send it as {"Air_temp": 25, "Water_temp": 22}
`LoRa.print("{\"Air_temp\": 25, \"Water_temp\": 22}");`

jerome-angeloni, 2021-02-17:
One more information: on the RX side, I run OpenMQTTGateway and I check the result in the MQTT broker logs:
1 - LoRa.print(0x22);
gives in MQTT logs:
{
  "rssi" : -47,
  "snr" : 9.75,
  "pferror" : 9068,
  "packetSize" : 2,
  "message" : "34"
}

2 - LoRa.print(char(0x22));
gives in MQTT logs:
{
  "rssi" : -64,
  "snr" : 8.5,
  "pferror" : 9017,
  "packetSize" : 1,
  "message" : "\""
}

3 - LoRa.print("\"");
gives in MQTT logs:
{
  "rssi" : -45,
  "snr" : 9.25,
  "pferror" : 9068,
  "packetSize" : 1,
  "message" : "\""
}

 I can't find a way to get something like: "message" : """  (of course this is not the final objective, I'm just simplifying to the max for debug purpose)

IoTThinks, 2021-02-17:
It is not relevant to the library anyway.
Everything is built from bytes.

It is you or your application to define how you view the bytes.

jerome-angeloni, 2021-02-18:
Thanks for looking into this. Finally I've found a workaround so I don't have to handle the double quotes (I tried but couldn't find something working because of C strings and String object in Arduino. There must be a way, but I found way of doing which brings the same results in HA):
On the TX side: 
```
 LoRa.beginPacket();
  LoRa.print(Air_Temp);
  LoRa.print("-");
  LoRa.print(Water_Temp);
LoRa.endPacket();

```
On the RX side:
```
    String packet;
    String air_temp;
    String water_temp;
    packet = "";
    air_temp="";
    water_temp="";
    for (int i = 0; i < packetSize; i++) {
      packet += (char)LoRa.read();
    }
    air_temp = packet.substring(0, 5);
    water_temp = packet.substring(6, 12);  //always the same size anyway
    LORAdata.set("air_temp", (char*)air_temp.c_str());
    LORAdata.set("water_temp", (char*)water_temp.c_str());
```

In HA configuration.yaml:
sensor:
```
  - platform: mqtt
    name: Température Eau Piscine
    state_topic: "Lora/OpenMQTTGateway/LORAtoMQTT"
    unit_of_measurement: "°C"
    value_template: "{{ value_json.water_temp }}"
  - platform: mqtt
    name: Température Extérieure Piscine
    state_topic: "Lora/OpenMQTTGateway/LORAtoMQTT"
    unit_of_measurement: "°C"
    value_template: "{{ value_json.air_temp }}"
```

closed 2021-02-18 by jerome-angeloni

IoTThinks, 2021-02-18:
Glad that it works for you.
For me, I still send json string with double quotes with no issue.
RX side needs to use ArduinoJson to parse the json.

Btw, keep the LoRa messages as short as possible.