sandeepmistry/arduino-LoRa #435: Library conflict?

Dr-Bundolo, 2021-02-18:
I have come upon a snag I can't seem to solve.   
Recently I wanted to make a couple more copies of my Nano   as a backup. (I have one backup and it works fine when( I plug it in).  I uploaded the unaltered sketch to the Nano and found that program froze.  
There may have been  a lot of causes for not working.  The main culprit might have been my computer.  My laptop  died and I had to get a new computer.  When installing all the programs I updated the IDE from an earlier version 1.8.7.0 to the current one.  Also many of the libraries updated.  
I reinstalled the old version...   no change.    In order to troubleshoot, I boiled down the transmitting program to it's basic elements.  The distance measuring code (simple) the display, and the radio.
Here's the basic ultrasonic program WITHOUT any radio elements.

`   #include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
 
#define trigPin 7
#define echoPin 8
 
#define OLED_RESET 4
Adafruit_SSD1306 display(OLED_RESET);
 
void setup() {
Serial.begin (9600);
pinMode(trigPin, OUTPUT);
pinMode(echoPin, INPUT);
display.begin(SSD1306_SWITCHCAPVCC, 0x3C); //initialize with the I2C addr 0x3C (128x64)
display.clearDisplay();
 
}
 
void loop() {
float duration;
float distance_in;
 
digitalWrite(trigPin, LOW); //PULSE ___|---|___
delayMicroseconds(2);
digitalWrite(trigPin, HIGH);
delayMicroseconds(10);
digitalWrite(trigPin, LOW);
 
duration = pulseIn(echoPin, HIGH);
 
distance_in = (duration/2) / 73.914;
 
display.setCursor(5,0); //oled display
display.setTextSize(2);
display.setTextColor(WHITE);
display.println("Water Down");
 
display.setCursor(05,18); //oled display
display.setTextSize(2);
display.setTextColor(WHITE);
display.println(distance_in);
display.setCursor(75,18);
display.setTextSize(2);
display.println("ins.");
display.display();
 
delay(500);
display.clearDisplay();
 
Serial.println(distance_in);
}
  `
This works and puts a working display on the OLED 

However when the radio info is added to the setup, the program freezes.  Here's the program with the libraries and setup added.
`#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
 #include<LoRa.h>
#include <SPI.h>
#define OLED_RESET 4

#define trigPin 7
#define echoPin 8
 

Adafruit_SSD1306 display(OLED_RESET);
 
void setup() {
Serial.begin (9600);

 while (!Serial);
  Serial.println("LoRa Receiver");
  if (!LoRa.begin(433E6))
  {
   Serial.println("Starting LoRa failed!");
    while (1);
  }

display.begin(SSD1306_SWITCHCAPVCC, 0x3C); //initialize with the I2C addr 0x3C (128x64)
display.clearDisplay();

pinMode(trigPin, OUTPUT);
pinMode(echoPin, INPUT);
}

void loop() {
float duration;
float distance_in;
 
digitalWrite(trigPin, LOW); //PULSE ___|---|___
delayMicroseconds(2);
digitalWrite(trigPin, HIGH);
delayMicroseconds(10);
digitalWrite(trigPin, LOW);
 
duration = pulseIn(echoPin, HIGH);
 
distance_in = (duration/2) / 73.914;
 
display.setCursor(5,0); //oled display
display.setTextSize(2);
display.setTextColor(WHITE);
display.println("Water Down");
 
display.setCursor(05,18); //oled display
display.setTextSize(2);
display.setTextColor(WHITE);
display.println(distance_in);
display.setCursor(75,18);
display.setTextSize(2);
display.println("ins.");
display.display();
 
delay(500);
display.clearDisplay();
 
Serial.println(distance_in);
}`

As soon as the Lora info is added in setup the program freezes....  Didn't happen before.  

I've tried different SSD1306 libraries with the same result.   As I mentioned earlier, if I plug the backup Nano into the circuit board is fine.  It appears they are the same Nanos (USB Nano V3.0 ATmega328 16M 5V Micro controller CH340G board) although with Chinese arduinos you never know, but the Nano seems to work fine with other programs, so it's probably okay.  I don't have any other Nanos - but I've ordered some.  

So what's changed?   Why doesn't the program work when it worded fine a few weeks ago on my old computer?
Any help will be most gratefully accepted.
Jeff

IoTThinks, 2021-02-19:
Arduino Nano is well-known to provide insufficient power to LoRa.

To check, you provide external 3.3v power to LoRa.

Dr-Bundolo, 2021-02-19:
It’s a Printed circuit board, that has worked for a year.  The board works with same Nanos that I programmed a month ago.  But now doesn’t work since I changed my computer and had to reinstall all programs.  Will try wiring up  a circuit using an UNO to see if it works with that.  

Any other ideas about conflicts?

Jeff

IoTThinks, 2021-02-20:
1. Different power between PCs
2. Different library version between PCs. You can copy old libraries to new PCs.
3. Comment out code to debug.

Very unlikely due to this library.

Dr-Bundolo, 2021-02-20:
Thank you for your suggestions.  I’ll let you know when I get to the route of the problem.

Jeff