sandeepmistry/arduino-LoRa #436: SpreadingFactor stuck at 7 on Ra-02 board

YD0NXX, 2021-02-19:
I am testing a setup where a TTGO T-Beam is acting as a trasmitter with a Ra-02 on Arduino UNO board as a receiver. While changing the SpreadingFactor from 7 to 12 I noticed that Ra-02 board stuck at SF of 7. I kept the transmitter locked at SF 7 and vary the SF of Ra-02 and the result is the Ra-02 board receives all packets fine on all other SF values.

If I change the SF to 8 to 12 on the T-Beam transmitter site and match on the Ra-02 receive side then nothing is received.

So is this a bug on the implementation of SF on Ra-02 board ?

thanks, Yono

IoTThinks, 2021-02-19:
I have used SF7 with Ra01 and Ra02 for a few years. And never see such problem.

You can try the default sample Sender and Receiver callback to confirm the issue?

YD0NXX, 2021-02-19:
SF7 works fine on Ra-02. It is the other SF values that are not honored by the board.

thanks, Yono

IoTThinks, 2021-02-20:
Try with the default sample code first.
Change to SF8...
To delay 5 seconds between sending.

Unless there is new flaw, ra02 works for me from sf6 to sf12.

YD0NXX, 2021-02-20:
Thanks for the hints, so I start with default sample code and add one line to set the other SF values (and have the same number on the transmitter) and this Ra-02 receiver works fine.

So the library works fine, it is operator error somewhere in my code. Thanks for the help and I'll close this issue.

closed 2021-02-20 by YD0NXX