sandeepmistry/arduino-LoRa #437: Documentation inconguent: is BAND 868E6 or 866E6 ?

clabnet, 2021-02-20:
On doc page :

https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md

there are write two bands : **868E6** and **866E6**. 

What is the correct band ?

IoTThinks, 2021-02-20:
By right, should be 868E6 as 868Mhz is more common.

However, 868Mhz chip can send packets at 866, 868, 899, 878.556 or 915 Mhz with no issue.

SX1278 can use frequencies from 135-950Mhz.
433, 868 or 915 Mhz modules are for frequency optimization, cost optimization and marketing.

Should have 2 types of modules 135-525Mhz and 525-950Mhz.

sandeepmistry, 2021-02-20:
@clabnet thanks for letting us know, would you be up for making a pull request to correct this?

IoTThinks, 2021-02-22:
> @clabnet thanks for letting us know, would you be up for making a pull request to correct this?

Done. Likely the shortest pull request ever.
https://github.com/sandeepmistry/arduino-LoRa/pull/438

sandeepmistry, 2021-02-27:
Closed via #438. Thanks @IoTThinks !

closed 2021-02-27 by sandeepmistry