sandeepmistry/arduino-LoRa #439: Doesn`t work with S12 and BW7.8

Portia-Lin, 2021-02-22:
I use RA-02 module and your library. I have set the following settings on the transmitter and receiver:
```
LoRa.setSpreadingFactor(12);
LoRa.setSignalBandwidth(7.8E3);
```
I can't receive the signal from the receiver when SF=12 or 11. But if you install SF10 with this BW, then everything seems to work.
What am I doing wrong? In the datasheet on SX1278 it is specified that using settings, I receive sensitivity of a signal on a receiver input in -148dBm.
I use standart sender/receiver example, but set the settings:
```
LoRa.setSpreadingFactor(11);
LoRa.setSignalBandwidth(62.5E3);
LoRa.setPreambleLength(8);
LoRa.setSyncWord(0x12);
LoRa.setCodingRate4(4);
LoRa.enableCrc();
```
And I got a range of 320m on an open street. What should be the settings for the longest range?

IoTThinks, 2021-02-23:
This library works well for my ra02 at Sf12. 62.5 or 125Khz.

Show a photo of your kit with antenna.
Show the RSSI and SNR at 5 meters.

Likely wrongly tuned antenna. Espcially Chinese ipex antenna.

Portia-Lin, 2021-02-23:
Yes, Chinese antenna... Now I don't have the equipment to check them. Maybe I can check at the university next week. Images are hidden under the spoiler
My transmitter and receiver configuration:
<details>
  <summary>TX and RX images</summary>

  [image]
</details>
The location of the transmitter antenna near the window:
<details>
  <summary>Location</summary>

  [image]
</details>
And at 5 meters RSSI: -63, SNR:11
<details>
  <summary>RSSI andSNR</summary>

  [image]
</details>

IoTThinks, 2021-02-23:
Is your MCU Arduino Nano? If yes, you may try to use external 3.3v for the LoRa.

The RSSI and SNR looks fine for me at 5 meters.

Portia-Lin, 2021-02-23:
Yes, I use Arduino Nano. But I use AMS1117-3.3 for Lora and LCD screen power at TX and RX. The RX doesn`t receive the signal when LCD screen is connected. I solved this problem by adding two capacitors.
<details>
  <summary>AMS1117</summary>

  [image]
</details>

On YouYube I saw better results of range and SNR. What settings do you use for RA-02 and what antenna? What distance were you able to reach?

IoTThinks, 2021-02-23:
My best guess is insufficient power for LoRa and wrongly tuned antenna.

> What settings do you use for RA-02 and what antenna? What distance were you able to reach?
1. SF12, BW 125Khz, 433Mhz, CR 4/5, CRC
In open space for rural area with 1 wall between, can reach 1 km.

I check the VSWR for the antenna, quite good.
[image]

2. Don't go under 62.5Khz as the crystal of the cheap ra02 may not good enough.
3. Generally, the 923Mhz can reach farther than 433Mhz as it has better LNA gain.
Can try to enable gain(6), but it only a plus.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#lna-gain

Portia-Lin, 2021-02-23:
I set the LoRa.setGain(6), as you said, but the receiver could not receive the signal. I experimented with different values, it doesn't work, maybe I did something wrong. I did not use this in the next experiment.
This time I fixed the transmitter outside at a height of about 6 meters. The best result I got 1.2 km in case 2.
In case 1 I could receive a signal in motion (approximately 5km/h) most of the way.
In the case of 3 direct visibility interfered with two high-rise buildings (5 floors).
<details>
  <summary>CASE 1 | RSSI: -101, SNR: -21.25 </summary>

  [image]
  [image]

</details>
<details>
  <summary>CASE 2 | RSSI: -101, SNR: -18.50 </summary>

  [image]
  [image]

</details>
<details>
  <summary>CASE 3 | RSSI: -102, SNR: -19.75 </summary>

  [image]
  [image]
</details>

Portia-Lin, 2021-02-23:
Ipex to SMA has a length of 20 cm, I don't have a shorter. So I twisted it like this:
<details>
  <summary>Ipex to SMA</summary>

  [image]
  [image]

</details>

Will this create communication problems?

IoTThinks, 2021-02-23:
> This time I fixed the transmitter outside at a height of about 6 meters. The best result I got 1.2 km in case 2
So is this result accepted? Yes, higher position will have better light of sight and less interference (from the ground)
From the picture, I think it is quite far already for non-tuned antenna / device.

slavino, 2021-02-23:
With 433MHz LoRa I got around 4km without any issues (forest zone). Clean line of sight might mean few hundred kilometers. Google for world record attempts with LoRa ;)

IoTThinks, 2021-02-23:
@slavino May I know briefly about your kit?

Portia-Lin, 2021-02-23:
When I started studying lora, I expected better results with a lower antenna height on TX. I would like to get 2km without lifting the antenna so high. It's sad that I can't reduce the BW to 7.8kHz, although it worked with a BW of 30kHz. The next step is to measure the performance of my antennas. Possibly creating my own GP antennas. Thanks for your answers!!!

Portia-Lin, 2021-02-23:
@slavino I also wanted to know about your lora kit, antennas and experimental conditions

slavino, 2021-02-23:
I have few boards from LilyGo (TTGO) LoRa Series. I chose 433MHz because variety of amateurs radio antennas are highly available for these frequencies and 433MHz is OK to be used in my country.

https://www.aliexpress.com/item/4001178678568.html
https://www.aliexpress.com/item/32915894264.html
https://www.aliexpress.com/item/33047631119.html

Portia-Lin, 2021-02-23:
This TTGO was a bit expensive for me. I wonder what kind of lora module there is. Looks like RFM95, but with a protective shield

Portia-Lin, 2021-02-25:
@IoTThinks Unfortunately, my university has equipment for measuring VSWR from only 1.8 GHz.  There was an idea to create a VSWR meter on an Arduino, I saw some examples of this on the Internet.  Also the teacher told me that it is possible to measure the power of the received signal using the SDR receiver.  Then I can try to transmit at different frequencies within the range and find the best frequency that is suitable for these antennas.  But CSWR of my antennas I never know.

IoTThinks, 2021-02-25:
Yes, your teacher is right.
If follow that path, you need a noise source and a SDR. You may know which frequency to transmit the most power.

Btw, you can get a bronze wire around 1mm diameter.
For 433Mhz, the length should be 16.445cm
299792458 / (433000000 * 4) * 100 * 0.95  = 16.445cm

Or 16.445cm / 2 for a shorter one.

closed 2021-03-01 by IoTThinks