sandeepmistry/arduino-LoRa #441: Feature Request LoraVisionShield

hpssjellis, 2021-02-25:
Could the definitions for the [Lora Vision Shield](https://store.arduino.cc/usa/portenta-vision-shield-lora) that connects to the [Arduino Portenta](https://store.arduino.cc/usa/portenta-h7) be added to the [LoRa.h](https://github.com/sandeepmistry/arduino-LoRa/blob/c1bb134755b538f0e195d5d6784b949cd3382167/src/LoRa.h#L10-L15) file.

The following is already there
```
#if defined(ARDUINO_SAMD_MKRWAN1300)
#define LORA_DEFAULT_SPI           SPI1
#define LORA_DEFAULT_SPI_FREQUENCY 200000
#define LORA_DEFAULT_SS_PIN        LORA_IRQ_DUMB
#define LORA_DEFAULT_RESET_PIN     -1
#define LORA_DEFAULT_DIO0_PIN      -1

```
Curious about the SPI frequency should it be the same.

Also here is the [pins info for the PortentaH7](https://github.com/arduino/ArduinoCore-mbed/blob/66f1f03f27dbf97a79a906e823d1f81a9cc881a7/variants/PORTENTA_H7_M7/pins_arduino.h#L117-L120)   

```
#define SerialLoRa		Serial3
#define LORA_BOOT0      (PG_7)
#define LORA_RESET      (PC_7)
#define LORA_IRQ_DUMB   (PJ_11)

```

IoTThinks, 2021-02-26:
Can just use this instead?
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins

hpssjellis, 2021-02-27:
@IoTThinks I have been messing around with that. I think since the Portenta uses Serial3 that the commands 
```
LoRa.setSPI(spi);

LoRa.setSPIFrequency(frequency);
```

need to be working, but they don't for me and I can't even see the code for those functions in either the .h or .cpp file. I think they don't work anymore. Another weird issue is I can't even get the LoraVisionShields working on TTN, so either I am doing something wrong (most likely) or my vision shields have issues (not likely). I know people have tested the Vision Shields in Europe, but have not heard from anyone in North America yet.

Thanks for the suggestion?

IoTThinks, 2021-02-28:
This library is for SPI LoRa. Serial3 is not relevant here.

It is LoRa library. For TTN, you need LoRaWAN library.

hpssjellis, 2021-02-28:
@IoTThinks Thank you so much, that makes so much sense. I will try again and see if I can get it working as if it is the same as the MKRWAN boards.

hpssjellis, 2021-03-01:
Back to the lab again. Things still are not working for the Portenta LoRa Vision Shield. Any suggestions would be appreciated. I would really like to get this Library working on the shields.

IoTThinks, 2021-03-01:
I believe this board has its own MCU with LoRa firmware.
The SPI of LoRa is not exposed. Hence, you can not use this library.
You must use the UART interface provided with the board. Check the documentation of the board.

https://content.arduino.cc/assets/SL-ASX00026.PDF

[image]

hpssjellis, 2021-03-01:
@IoTThinks Thank you, nice to finally understand why it isn't working. I was really hoping to get point-to-point LoRa working on these shields. I am looking into mbed versions, do you have a suggestions?

IoTThinks, 2021-03-01:
You have two choices:
1. Simple: To use UART LoRa and use the vendor's API. E.g UART E32 LoRa.
2. Flexible: To use this library and choose SPI LoRa. E.g HopeRF RFM85/96 or Ai-Thinker Ra01, 02...

hpssjellis, 2021-03-01:
I posted a similar issue on the [MKRWAN library](https://github.com/arduino-libraries/MKRWAN/issues/87) github 

That library has code from this library but it does not work.

@IoTThinks I am trying to understand your 2 solutions. The Portenta has a possible 160 pins so accessing anything low level is a bit confusing at the moment.

closed 2021-03-03 by IoTThinks