sandeepmistry/arduino-LoRa #444: Deep sleep current 77μA, can be less

jbrepogmailcom, 2021-03-02:
Hello. I use LoRa Radio Node V1.0. Cool thing, with following code it has 77μA in deep sleep. I wonder if anyone experienced lower deep sleep current and with what hardware.

```
    LoRa.end();
    LowPower.powerDown(SLEEP_8S,ADC_OFF, BOD_OFF);
```

IoTThinks, 2021-03-03:
My custom board with esp32 reaches 13uA deepsleep. :D
77uA is quite good for standard board already.

jbrepogmailcom, 2021-03-03:
13μA even with LoRa chip? That is good, could you share schematics please? Is it possible to buy the board?

IoTThinks, 2021-03-03:
I off the power for the whole LoRa module to reach 13uA.
With LoRa, 77uA is very good already.

Other people here can reach 35uA with LoRa.  You can check for hints https://github.com/sandeepmistry/arduino-LoRa/issues/431#issuecomment-788961131

closed 2021-03-03 by IoTThinks