sandeepmistry/arduino-LoRa #447: Receiver hangs up after random time interval

romigr, 2021-03-04:
[image]

IoTThinks, 2021-03-04:
I have no problem receiving packets for months.
You can try to use the receiver at https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaReceiverCallback/LoRaReceiverCallback.ino

You need to set the pins, too.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins

peringek, 2021-05-03:
Did you solve this problem ?
I have the same issue.

romigr, 2021-05-03:
Thanks! I will try that.
Regards,
Richard Alan (Al) Romig
2619 Forest Oaks Drive
College Station, TX
77845
979-485-1148 home
713-201-4413 cell

robas, 2021-06-18:
Hey @romigr any updates about this issue? Apparently, I have been facing the same problem and trying to understand what is going on.

Thank you.

romigr, 2021-06-18:
Hi Rodrigo,
No, I don't have any more news. I have been working on other projects, but
I will get back to LoRa before long..

Richard Alan (Al) Romig
2619 Forest Oaks Drive
College Station, TX
77845
979-485-1148 home
713-201-4413 cell

bharadwajtke, 2021-06-23:
Hi,
I am facing the same problem, if anybody has any brekthrough on this plz let me know. For time being i am polling for data reception of lora, after putting delay lets say 3ms it reduced hanging considerably, if it stps receiving i am sending data like heart beat again it resumes...its little peculiar for me...plz let me know any brwakthrough by any one or any other work around

IoTThinks, 2021-06-23:
Can try to use callback listener instead of parsePacket.

IoTThinks, 2021-06-23:
May be related and solved issue: https://github.com/sandeepmistry/arduino-LoRa/issues/363

doidgy, 2021-07-18:
I am afraid I have exactly the same issue, I followed the #363 issue but that was more for transmitter problems, my issue is exactle as the OP

krysatpat, 2021-08-29:
I'm also having the same issue, I thought reinitializing the can solve  the issue (which is same as [61](https://github.com/HelTecAutomation/Heltec_ESP32/issues/61)) but still it will hang randomly for some time, too.

kulaksazov, 2021-09-10:
Hi,
To solve the problem with the HELTEC LoRa receiver ALWAYS! after accepting the package, include the order:

// put the radio into receive mode
        LoRa.receive();

You will not have a problem anymore!
Great!

Inrebusillis, 2022-04-19:
> Hi, I had the same problem with my heltec esp 32 LoRa - at any moment it just stopped taking! There is no such thing with a transmitter. The problem is solved very easily! In the loop section, or where the LoRa read loop is, after the end of the loop just put:
> 
> // put the radio into receive mode LoRa.receive();
> 
> just to "remind" each time the device that you need to keep taking! It hasn't stopped yet! At least for me the problem is solved :) Heltec library version: 1.1.0

Unfortunatly not working fr me ( Heltec V2 with heltec library)

Inrebusillis, 2022-04-19:
Same issue:
Heltec Lora OLED v 2 ( two pairs)  with Heltec library/ arduino library, lora libraries....
tryed onreceive method, different library, rewrite code, add Lora receive in the loop-....
still randomly hanging....
anyone has a clue of what is happening?

Inrebusillis, 2022-07-07:
> Same issue: Heltec Lora OLED v 2 ( two pairs) with Heltec library/ arduino library, lora libraries.... tryed onreceive method, different library, rewrite code, add Lora receive in the loop-.... still randomly hanging.... anyone has a clue of what is happening?

Do not ask me why:
I put a line 
Serial.println(currentTime);
at the beginning and at the end of the loop and everything is now working.....
A bug between Serial and Lora communication?
( no, I did not test setting the Serial Enable to "false", I was using the Heltec library.
I was just happy that finally works.

//Serial.begin(115200);
  Heltec.begin(true /*DisplayEnable Enable*/, true /*Heltec.Heltec.Heltec.LoRa Disable*/, true /*Serial Enable*/, true /*PABOOST Enable*/, BAND /*long BAND*/);

fizga, 2023-01-15:
hello to all, I am facing the same problem, the arduino MEGA freezes randomly, I use an RFM95 module. after several tests the problem occurs when I use the LoRa library.
I tried the solutions above but the problem persists.
has anyone found a solution?

livioenrico, 2024-07-20:
I found a solution !!!

Many people have experienced this but until now no one had found a solution:
https://arduino.stackexchange.com/questions/94389/lora-sx1278-stops-receiving-after-a-short-time
https://github.com/sandeepmistry/arduino-LoRa/issues/447
https://stackoverflow.com/questions/63861160/arduino-nano-with-ra-02sx1278-freezes-after-receiving

This is a good workaround, but it wouldn't hurt for SEMTECH to understand why it happens
and also that he would give us the official solution, or fix the defect in his chips.

```
// ============================================================================
//  Each time the OnReceive callback is called there is a small error chance
//  this could be approximately every 1000, 3000 or sometimes 10000 calls. 
// ----------------------------------------------------------------------------
//  When this happens the DIO0 interrupt pin remains stuck high
//  with a voltage fixed to 3.3V and interrupt receiving no longer works.
// ----------------------------------------------------------------------------
//  To correct this problem we test in the loop() if the DIO0_PIN is HIGH
//  and in this case we redo all the initializations.
// ============================================================================

// ============================================================================
//  RESTART RECEIVER INTERRUPT IF LOCKED
//  This function must be called periodically from the loop()
// ============================================================================
void RestartReceiverInterruptIfLocked()
{
  if (digitalRead(DIO0_PIN) == HIGH)
  {
    RTX_Init();
  }
}

// ============================================================================
//  RTX INIT
// ============================================================================
void RTX_Init()
{
  // ---------------------------------------------- BEGIN
  while (!LoRa.begin(RTX_FREQUENCY))
  { 
    Serial.println("Error initializing LoRa module");
    delay(100);
  }
  // ---------------------------------------------- SETTINGS
  LoRa.setTxPower(17); // do here your additional settings           
  // ---------------------------------------------- REGISTER CALLBACK
  LoRa.onReceive(RTX_OnReceive); 
  // ---------------------------------------------- START RECEIVE 
  LoRa.receive();
}
```